Xiashangning/BigSurface #1: Proposal to fix power management during sleep

sauregurke, 2021-04-01:
Hi Xiashangning,

I believe I have come across another issue that may need to be solved with this project. I believe that the SP7 is incapable of entering into an S1 state during sleep in macOS. The CPU never ceases during sleep, which draws tons of power - not ideal for the battery size in these machines. I have tried diagnosing this on my own for three weeks, and in that time several experienced people on the Hackintosh discord tried and failed to find the root cause.

Have you gotten sleep to work, as in the CPU turns off? If not, do you think it is something this project needs to patch as well?

sauregurke, 2021-04-01:
I am experiencing 67% battery drain after 12 hours of sleep; iStat Menus logs cpu usage hovering at about 0.6% constantly during sleep as well. (iStat Menus is not the cause; the battery drain test was done without it installed and with no other programs running.)

Xiashangning, 2021-04-02:
@sauregurke Yes, I believe this is because that the sleeping type is AOAC on SP7, see https://github.com/daliansky/OC-little/tree/master/01-%E5%85%B3%E4%BA%8EAOAC for reference
There is only the Chinese version but i think you can use google translate since the phrases are not complicated.

sauregurke, 2021-04-05:
Hey, totally forgot to mention, thank you! This did enable deep sleep on my device - perhaps I should add a request to add this to the guide. The only thing is at the moment, it seems that battery drain during sleep is somehow worse. Do you have any battery drain issues? I’m investigating...

But regardless, may I ask for your experiences with SSD speeds in macOS? I’ve run a few tests and it seems to give read speeds between 300 and 600MB/s, which is much less than the 2000MB/s in windows. I am doing further testing on this as well, but it might be something else to consider in BigSurface, maybe.

sauregurke, 2021-04-28:
Hey, I have good news - one of my friends on discord who has some experience with writing things like this is working on this project. He elected to start anew, so no clone - not sure why, perhaps just wanted a clean slate? Regardless, once there is some progress and he uploads to Github, I will link it here!

jackjack1-su, 2021-05-04:
> Hey, totally forgot to mention, thank you! This did enable deep sleep on my device - perhaps I should add a request to add this to the guide. The only thing is at the moment, it seems that battery [...]

Hmm...
How did you make the DeepIdle work?
Could you put your ACPI Settings and SSDTs here?
Really Thx!

sauregurke, 2021-06-12:
Hey! SO so sorry about my lack of contribution to the project as of late, I just graduated after a mad dash to get everything done on time. COVID-19 seemed to hit me even harder this past semester, but thankfully that's over. In all honesty, I completely forgot to submit my patches until I got the email from the other issue that was just opened - apologies!

Anyways, here is a folder with the contents of my ACPI folder, plus my config file. Please note that I have a fake battery patch because it helps with power management, and that my config file was lightly modified to boot the latest macOS beta - Monterey. The main difference for this version is removing the bluetooth path from USBMap.kext, as currently there is no way for bluetooth to work in this beta - enabling it will hang the OS at boot. Also, since AirportItlwm.kext only works on the macOS versions it is compiled for, I switched to itlwm.kext+Heliport for Monterey. That is what the max/minkernel values are for, so I can keep Big Sur flawless while having access to Monterey. Other than these two things, and the -lilubetaall flag to enable Lilu patches on untested versions, you don't need any other modifications to boot Monterey - that is the power of OpenCore!

My friend on the Hackintosh discord says that his estimate for a beta for a battery readout kext is mid-July. This issue had practically every experienced programmer on there stumped, so frankly I'm surprised with how soon that estimate is.

As to the ACPI patches, Deep Idle should work with just adding the patches you don't yet have. (Apologies once again for taking this long; I don't blame you if you already made them yourself.)
[Deep Idle Patches and Extras.zip](https://github.com/Xiashangning/BigSurface/files/6642774/Deep.Idle.Patches.and.Extras.zip)

jackjack1-su, 2021-06-13:
> Hey! SO so sorry about my lack of contribution to the project as of late, I just graduated after a mad dash to get everything done on time. COVID-19 seemed to hit me even harder this past semester, [...]

However, I've got black screen or blue screen when waking from sleep.
Have you ever met this problem?

therealjbenam, 2021-08-06:
Any news from your friend @sauregurke ? Battery read-out is the last big thing I properly miss on the SP6 when hackintoshed.

I also gave a look at your ACPI folder, it seems you really have a LOT of patches in there! Are all of them really needed? I have a Surface Pro 6 here, from what I can see here you have a Surface Pro 7? (I can see PNLF-CFL...)

sauregurke, 2021-08-06:
Yeah, I have a SP7. As far as I remember, all of these patches are necessary to get deep idle to work, which is basically new age sleep. Allows it to be connected to Wi-Fi even when “sleeping,” yada yada.

sauregurke, 2021-08-06:
> However, I've got black screen or blue screen when waking from sleep.
> Have you ever met this problem?

No, not me. The intermittent 4k60 flickering issue continues, and is worse in macOS 11.3 and above - I’m assuming some kind of race condition. Really irritating, but that seems to be the nature of 11.3 and above - race conditions all over the place.

jackjack1-su, 2021-08-08:
OK really thx

sauregurke, 2021-11-10:
> Any news from your friend @sauregurke ? Battery read-out is the last big thing I properly miss on the SP6 when hackintoshed.
>
> I also gave a look at your ACPI folder, it seems you really have a LOT [...]

Just re-read this; are you sure you had battery readouts working on your SP6, or perhaps did you have a SP4? Since the SP5, Microsoft has been using a special UART module for battery callouts and two engineers have been unable (as of yet) to write a kext to get SMCBatteryManager to read them. Did you perhaps find a solution online?

therealjbenam, 2021-11-11:
> Just re-read this; are you sure you had battery readouts working on your SP6, or perhaps did you have a SP4? Since the SP5, Microsoft has been using a special UART module for battery callouts and [...]

Hi, maybe I wrote something in a confusing way, but I never claimed I had SP6 battery readouts working - I wish I did 😄  The entire UART-thing puts it confortably outside the reach of my hackintoshing skills.

techlogik, 2021-11-11:
Have a SF6 with Catalina 10.15.7 working pretty well. Only issue using three types of surface docks are not being able to get dual monitors working. It is just blank the 2nd monitor, but MacOS sees it in the display, or I get flickering. Have the dock with USB C and dual mini display ports. Of course, use the side port fine to mirror and at least have 1 monitor larger while the lid is closed. Shows Intel UHD 620 and two other monitors in the system info. Stumped how to resolve. Sorry, not to clutter the thread, but not many active people working on SF devices. Any guidance appreciate, might give Big Sur a go if I can figure out Catalina dual monitors. Thx

closed 2022-06-25 by Xiashangning

balopez83, 2024-02-06:
> Hey! SO so sorry about my lack of contribution to the project as of late, I just graduated after a mad dash to get everything done on time. COVID-19 seemed to hit me even harder this past semester, [...]

Hi @sauregurke  I am attempting to keep the SP7 EFI updated and relevant with the latest versions on my Github but have not been able to successfully resolve the sleep issues. I did attempt to use the deep idle patches though it always freezes before hibernating. Can you send me you latest EFI so I can see what you did to get sleep functional? Also, did you have to set any special commands in macOS directly (e.g. Hibernatemode=25)?