Xiashangning/BigSurface #3: Hardware buttons

billabongbruno, 2021-06-12:
Hey there.
First of all, I would like to congratulate you on your project and the endeavour you are about to take.

I know that Hackintoshing a Surface can sometimes be very tricky.

That being said:

I came across your project while Googling for fixes for the cameras on the Surface Pro under a Hackintosh setup.

I just wanted to lend out my help and let you know that I fixed the Hardware button issue (Power button and the two volume buttons) by editing my dumped DSDT.aml and then patching it with the corresponding code.

**First I injected this in my DSDT.aml using MaciASL's latest version (https://github.com/acidanthera/MaciASL/releases):**

#Maintained by: RehabMan for: ACPIKeyboard.kext
#patch.txt
 
#This patch serves to provide a catalyst to load ACPIKeyboard.kext
#and it also provides a sink to send notifications to in
#order to send messages to the kext to simulate keystrokes.
 
into device label RMKB remove_entry;
into definitionblock code_regex . insert
begin
Device (RMKB)\n
{\n
    Name(_HID, "RMKB0000")\n
}\n
end;

**Then, after saving it, I re-opened it and saved this to it:**

into method label _Q52 replace_content
begin
// Volume-up pressed\n
Notify(\RMKB, 0x1148)\n
end;
 
into method label _Q53 replace_content
begin
// Volume-up released\n
Notify(\RMKB, 0x1248)\n
end;
 
into method label _Q54 replace_content
begin
// Volume-down pressed\n
Notify(\RMKB, 0x1149)\n
end;
 
into method label _Q55 replace_content
begin
// Volume-down released\n
Notify(\RMKB, 0x1249)\n
end;
 
into method label _Q60 replace_content
begin
//Power button pressed\n
Notify(\RMKB, 0x117f)\n
end;
 
into method label _Q61 replace_content
begin
// Power button released\n
Notify(\RMKB, 0x127f)\n
end;

I then proceeded to get the ACPIKeyboard.kext from https://bitbucket.org/RehabMan/os-x-acpi-keyboard/downloads/RehabMan-ACPIKeyboard-2015-1230.zip

Then I placed it under Kexts on my OC folder (using OpenCore 0.7.0) and then edited the config.plist to include it.

My source for this was:
https://www.programmersought.com/article/9099709695/

Hope this helps with your project and saves you some trouble.

Xiashangning, 2021-06-12:
Thank you! This is exactly what i needed--the magic numbers to dispatch corresponding events. I will borrow some code from RehabMan's os-x-acpi-keyboard source code and integrate it into my project.
Maybe I will create a virtual device for surface using ssdt, it is much safer to do so.
Right now I don't have too much time, final exams are coming :( But as soon as I am free, maybe in early June, I will update my repo and let you know
Besides, cameras seem impossible to drive, they are i2c and there is no source code of Mac virtual camera project to let me learn the code structure.

billabongbruno, 2021-06-12:
> Thank you! This is exactly what i needed--the magic numbers to dispatch corresponding events. I will borrow some code from RehabMan's os-x-acpi-keyboard source code and integrate it into my project. [...]

You're welcome, happy to have helped, even though it was not a lot.

Glad to know you're committed to this project.

I know what that's like (the exams). Thankfully I'm already done with my masters.

I wish you all the best and good luck on your exams.
I'm assuming it's computer engineering or something related. If it really is, then you have your work cut out for you and do need the time to study.

Thank you for your message.

Also, a shame to know that we won't be getting either touchscreen nor cameras anytime soon.
(I should point out that Touch-base drivers exist, but they are paid. I did try it out for a day - because they have a 30-day trial - but I was unable to get them working, probably because of the patches that I have already applied that probably are blocking the Touch-base drivers from working).

If you want me to drop a link to these drivers that I'm talking about, I can, but if you do a quick Google search for "Touch base drivers Surface Pro 4" you should be able to find them. There are drivers for almost every OS, MacOS included.

Anyway, best of luck once again to your exams.

Best regards,
Bruno.

billabongbruno, 2021-06-14:
@Xiashangning , I found something regarding multitouch drivers using "mei".

Came across this:
https://newlineinteractive.freshdesk.com/support/solutions/articles/8000032202-mac-multi-touch-driver-

Just wanted to leave this info here, so we can debate it after you're done with your exams.

Good luck.

Xiashangning, 2021-06-29:
@billabongbruno Sorry to be late, but could you publish your patched DSDT here? I cannot find _Q* functions in my DSDT, so did you just add them ?

plsh2me, 2021-06-29:
In fact even I don't find any _Q* functions on my DSDT as well. I found the _L*, but they are not useful if I don't know the corresponding buttons. 
[image]

closed 2021-10-21 by Xiashangning

sauregurke, 2021-11-08:
I'd like to reopen this as a suggestion to add the power button functionality into BigSurface. Perhaps we could have it put the computer to sleep with a press of it? Not sure if this could be possible, as putting the computer to sleep via keys requires a key combination, not just a simple key press like volume up or down. Could be nice to make macOS feel more natural on the surface.

sauregurke, 2021-11-08:
Edit: nvm, I read your note on the power button not working after DeepIdle. Though I'm pretty sure for me it never quite worked, will test now.

billabongbruno, 2021-11-08:
I just got all these notifications.
Do you still need my patched DSDT?

Also, I have found a way to use updated Voodoo2IC kexts with an argument on boot-args.

I am, however, struggling with getting gestures (and the trackpad to be recognized as a trackpad, even though it works) under Monterey.

Would you be willing to help?

sauregurke, 2021-11-08:
It's likely, at least in my experience, because you didn't remove the old Voodoo____ kexts when you added those integrated into the BigSurface kext. Also, there is a slight modification you need to make to the info.plist for BigSurface for Monterey trackpad support, let me go find that - I'll probably submit a pull request to add that to the main page.

sauregurke, 2021-11-08:
Also, power button does work before sleep - thank you! I'll try to figure out why it doesn't work after but chances are I won't be of much help there lol.

billabongbruno, 2021-11-08:
My power button works before and after sleep.

Do you want my DSDT?

sauregurke, 2021-11-08:
For Monterey multitouch support, you need to edit the info.plist of BigSurface. Under Root > IOKitPersonalities > Surface Type Cover Driver, change IOProbeScore to 9300 (this is tested to work for Monterey).

sauregurke, 2021-11-08:
Do you have DeepIdle enabled with SSDT patches? If so, I'd like to know how you patched it

billabongbruno, 2021-11-08:
I did remove all Voodoo kexts before I updated. I am not yet using BigSurface integrated kext, I am using the old method. But trackpad doesn't show up as a trackpad, but works as a mouse. No gestures, though.

billabongbruno, 2021-11-08:
I didn't manually patch the DSDT for use with DeepIdle. I got the right mix of files from a ton of sources and then just added the power and volume buttons patch. But it works out of the blue. I can share my 0.7.4 EFI folder (with serials removed), if you want. 0.7.5 is still work in progress.

sauregurke, 2021-11-08:
I would be down to have a look, but power button functionality isn't a strictly necessary thing, so if I can figure out what you did I'd be happy.

billabongbruno, 2021-11-08:
I'll upload it to my GoogleDrive and post the links here, then.

billabongbruno, 2021-11-08:
https://drive.google.com/file/d/1bcU9XWXG8pVpJPyjH88mWXjQripIfmZh/view?usp=sharing

Here it is.

Password is:
123

sauregurke, 2021-11-08:
It doesn't seem like you have DeepIdle enabled; therefore it isn't really sleeping (at least in my experience prior to figuring out how to enable DeepIdle). Let me know if you get the power button working both before and after deep idle!

billabongbruno, 2021-11-08:
I'll have a look, then.

Xiashangning, 2021-11-09:
@billabongbruno No need any more, the buttons work now.

billabongbruno, 2021-11-09:
I'm glad it all works, then.

One stupid question, though.

With your project is there still need to use kexts like "USBMap", "GenericUSBXHCI", "USBPorts", "HibernationFixUp", "NVMeFix" and thing of such?

Or do we only need to use the kexts like "Lilu", WhateverGreen", etc?

Or not even those?

Sorry if it is a dumb question, but I'm trying to understand.

Xiashangning, 2021-11-09:
USBPorts NVMeFix Yes, others no
+ Lilu & weg etc..

billabongbruno, 2021-11-09:
Thank you for the clarification.

Having taken that into consideration, I now have what I believe to be the "perfect" mix of kexts and DSDT files.

SD Card is fully working as well.

I would like to share my work with you here.

I only removed the serials for you to insert your own, but apart from that, everything is good to go.

I have added all the kexts necessary to work with the hub I got, which includes Ethernet, found here:
https://www.aliexpress.com/item/1005001936076326.html?spm=a2g0s.9042311.0.0.27424c4dgQrvL7

I have also included the kexts that are installed with Chris1111's OC Wireless Adapter installer, so you only need to install the app itself. It can be found here:
https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter

That all being said, my work has been uploaded to GoogleDrive.
It is using OpenCore 0.7.5 with the latest kexts and is designed for use with MacOS Monterey 12.01
I have made it available here, again with the **password being "123", without quotes**:

https://drive.google.com/file/d/1bZC8zbH_owa33mpPb1e-PpMHmtfI3wtr/view?usp=sharing

Thank you for all your help.
Hopefully, in time, we will be able to get Wifi, Bluetooth and the touchscreen working. Who knows?

Best regards,
Bruno.

plsh2me, 2021-11-10:
@billabongbruno I checked your efi folder. I don't see any wifi and bluetooth kexts. But both are fully working on this device. Or you have any other method to make this working?

Xiashangning, 2021-11-10:
@plsh2me I believe that is because Monterey breaks support for wifi and bluetooth, this is the common problem which has not been solved yet.

plsh2me, 2021-11-10:
@billabongbruno No, both are still working. For WiFi download the latest kext for monterey. And for bluetooth you have to  delete the injector kext, instead of this you need to use blue tool kext from broadcom kext resources.

billabongbruno, 2021-11-10:
There is no working kext for Wifi and Bluetooth because I have a Surface Pro 4, not a Pro 6 or 7, which means I have a Marvell Avastar board, which has no support for it.

For Intel boards, on Monterey, you need the "BlueToolFixUp.kext" and the "IntelBluetoothFirmware.kext", loaded in this order.

I did not include those in mine because my work is dedicated to whatever equipment I have, a Surface Pro 4, making it pointless for me to include more kexts then I need, increasing the boot time.

The work has been done, no need to spoon-feed everything. ;)

billabongbruno, 2021-11-10:
@Xiashangning , I am wondering if you would be able to help me on an unrelated issue and separate machine.

I have a fully working setup on an ACER Nitro 5, with everything working (except dGPU) and I have an issue that you might be able to know your way around.

So, everything works after I have booted into any OS (I am using rEFInd as my bootloader picker), but after I boot INTO Monterey, the next reboot / cold boot breaks my keyboard during the boot process.

I believe I have narrowed it down to the VoodooPS2 Keyboard Controller kext, but I have no idea on how to fix this issue.

Do you have any idea?

A related link to this very issue is described here and I jumped in on that train:

https://www.reddit.com/r/hackintosh/comments/qn82qi/keyboard_problem_after_reboot_from_macos/

Xiashangning, 2021-11-10:
Well, don't know if i understand this issue correctly, your keyboard uses EC to send key event so you do need VoodooPS2 kext right? Without this kext you cannot use keyboard in macOS but won't have such problem.
So I suggest you first look at VoodooPS2 and see if there has any code related to configuring EC or something else. (I don't really use this kext so I have not read its source code) and then try comment them on places like stop() releaseResources() and have a try.

billabongbruno, 2021-11-10:
That's the weird part.
Inside MacOS (or any other OS I use - Kali, Ubuntu, Windows 11 and ChromeOS), keyboard works perfectly.

Even inside MacOS.

It just stops working after a reboot originated from within MacOS.

To clarify:

Reboot from / cold boot after booting:

Kali - Works
Ubuntu - Works
Windows - Works
ChromeOS - Works
MacOS - doesn't work.

It works inside MacOS with no issue. After a reboot or cold boot, the keyboard is not usable ONLY during boot process (like BIOS and bootloader).
It starts working fine again after I boot into any OS.

Xiashangning, 2021-11-10:
Yes, and I guess that is because it has not be shutdown correctly, maybe you also have to look in your DSDT to find some clue

billabongbruno, 2021-11-10:
I suppose.
BigSur 11.6.1 did not behave like this.
Monterey sure did break a lot of things, especially regarding Hackintoshes.

Oh well.

Thank you for the help.

dkoluris, 2022-08-01:
> Thank you for the clarification.
>
> Having taken that into consideration, I now have what I believe to be the "perfect" mix of kexts and DSDT files.
>
> SD Card is fully working as well.
>
> I would like to [...]

Is there a chance you could repost this EFI please?