Xiashangning/BigSurface #4: Compiled .kext in releases

MrHomebrew, 2021-06-21:
Would it be possible to get a compiled .kext in the release section?

Xiashangning, 2021-06-22:
Got it, I will update my repo and release a beta version of it, (surface typecover keyboard & touchpad full support, power and volume buttons maybe)

plsh2me, 2021-06-28:
@Xiashangning I will wait to give a try on my surface pro 7. Thank you for your efforts.

MrHomebrew, 2021-06-28:
Sounds great 👍

Xiashangning, 2021-06-29:
@plsh2me @MrHomebrew Done, please try it and let me know if it works properly.

plsh2me, 2021-06-29:
@Xiashangning I am pleased to confirm you that the kext works as expected with your recommended config settings on my Surface pro 7 (with Surface pro type cover -model 1725). 
Hope you will proceed to add more functions in future version.
Thank you very much.

jackjack1-su, 2021-06-30:
Emmm...
I suggest to close this issue because the kext is released.

closed 2021-07-01 by Xiashangning