Xiashangning/BigSurface #5: macOS Monterey Support?

jackjack1-su, 2021-07-01:
With the BigSurface, the touchpad seems only partially working on Monterey(Can't use gestures on Touchpad)
But on BigSur it works perfect on my Surface
(However, some people say it doesn't work)

Xiashangning, 2021-07-01:
maybe you should first use the fake battery and configure the touchpad in system preference, as for the potential incompatibility, please upload ioreg and corresponding explanation. i will try to fix that

MrHomebrew, 2021-07-18:
I also noticed with Monterey that voodooi2cHID does not seem to be working. Gestures like swiping up and down in a browser just do not work. I checked in Hackintool and for some reason same setup on Monterey that I had on Big Sur but no gestures. Big Sure I have gestures working.

plsh2me, 2021-10-31:
> maybe you should first use the fake battery and configure the touchpad in system preference, as for the potential incompatibility, please upload ioreg and corresponding explanation. i will try to fix that

Hi, I upgraded the system to Monterey. The system is working just like Big Sur. In new release you added the key support also works on Monterey. But on Big Sur the trackpad and gesture was perfect with fake battery patch- now on Monterey the trackpad not found, so the gestures are not woking (trackpad working with basic functions). I also tried the multitouch commands from badstrom to get the gesture working. Unfortunately nothing works (with or without sudo).
I attached the ioreg, please let me know if you need more information from me.

[MacBook Pro.zip](https://github.com/Xiashangning/BigSurface/files/7448637/MacBook.Pro.zip)
.

plsh2me, 2021-11-01:
@Xiashangning I found the previous ioreg which may be can help you to find out the cause of incompatibility. 
Please don't confuse about the system version, due to wrong cpu model I changed it to MacBook Pro. 
[image]
[MacBook Air-ioreg.zip](https://github.com/Xiashangning/BigSurface/files/7456617/MacBook.Air-ioreg.zip)

fenglinnet, 2021-11-02:
My problem is similar to yours. It was fixed after I modified the driver IOProbeScore. I guess it's a driver loading priority issue.

plsh2me, 2021-11-02:
@fenglinnet Can you please share the kext or tell me what IOProbeScore you applied? 
And I am using the driver priority as per instructions, I you have the different order please share me.
[image]

Thank you

fenglinnet, 2021-11-02:
@plsh2me I just changed 300 to 9300 in BigSurface.kext and it works :)
``` xml
			<key>IOClass</key>
			<string>SurfaceTypeCoverDriver</string>
			<key>IOProbeScore</key>
			<integer>9300</integer>
			<key>IOProviderClass</key>
			<string>IOHIDInterface</string>
			<key>ProcessBluetoothMouseStopsTrackpad</key>
			<false/>
			<key>ProcessUSBMouseStopsTrackpad</key>
			<false/>
			<key>QuietTimeAfterTyping</key>
			<integer>200</integer>
			<key>RM,deliverNotifications</key>
			<true/>
```

plsh2me, 2021-11-02:
@fenglinnet Yes, its also worked for me.
Thank you very much

Xiashangning, 2021-11-09:
@fenglinnet @plsh2me I am wondering if you also have VoodooI2C.kext loaded ? Why is there such priority issue on Monterey...

sauregurke, 2021-11-09:
> @fenglinnet @plsh2me I am wondering if you also have VoodooI2C.kext loaded ? Why is there such priority issue on Monterey...

Not them, however I have tested this fix in Monterey. I have VoodooI2CServices loaded as it is integrated into BigSurface and VoodooI2C is not loaded.

Helloearther, 2021-11-09:
> @plsh2me I just changed 300 to 9300 in BigSurface.kext and it works :)
> 
> ```
> 			<key>IOClass</key>
> 			<string>SurfaceTypeCoverDriver</string>
> 			<key>IOProbeScore</key>
> 			<integer>9300</integer>
> 			<key>IOProviderClass</key>
> 			<string>IOHIDInterface</string>
> 			<key>ProcessBluetoothMouseStopsTrackpad</key>
> 			<false/
> 			<key>ProcessUSBMouseStopsTrackpad</key>
> 			<false/>
> 			<key>QuietTimeAfterTyping</key>
> 			<integer>200</integer>
> 			<key>RM,deliverNotifications</key>
> 			<true/>
> ```

太好了，改了果然正常！THANK YOU！
surface pro 6 , Monterey 12.0.1，300-〉9300后触摸板正确识别，手势正常了！

PS，bigsurface.kext 让音量键正常了，感谢！

Xiashangning, 2021-11-10:
Solved in next release.

closed 2021-11-10 by Xiashangning

qingqingshan, 2021-11-11:
> [image]
> 
> @fenglinnet Yes, its also worked for me. Thank you very much

大佬，能否分享一下你的EFI，我的也是surface Pro 7，但是总是无法驱动触摸板和电源，我的邮箱：qingqingshan@hotmail.com，在此表示感谢。