Xiashangning/BigSurface #6: Support for Surface Laptops in future (keyboard/mouse)?

JChamberlain28, 2021-07-01:
This repo is interesting as attempts to port the surface aggregator module to macOS have not happened before. I am unsure how the SP 2017 type cover works, or if it utilizes the surface aggregator module, as the SL does. Is there scope for testing on and enabling support for the SL 1/2/3?

Xiashangning, 2021-07-01:
Unfortunately no, Surface treats type cover as a regular USB connected keyboard, but with a specialized connector. So macOS can recognize it without any problem. But for Surface Laptop, that is another story, it use SAM to control its keyboard so right now there is no way to drive it.

JChamberlain28, 2021-07-02:
Thanks for the extra info, seems MS made it hard so users wouldn't use anything other than windows :)

JChamberlain28, 2021-08-08:
@Xiashangning i see your readme mentions deleting surface laptop and surface book parts of the SAM code when developing battery readout support for SP6/7. Will you consider developing support for the surface laptop and surface book code depending on the success with getting battery readout done.

Xiashangning, 2021-08-09:
@JChamberlain28 Sure, but i would suggest you hold no holp for that. The ONLY thing stops us from getting battery or keyboard of SL work is the F**King stupid SAM, it is in nature an UART device and so far, we haven't got any documents about UART drivers on macOS yet. Its driver is perquisite for a proper SAM driver. To accomplish that driver needs some expert in this domain. Of course not me, sadly. I hear that someone is working on this driver. Until he or she has made some progress, i guess we have to wait

closed 2021-08-09 by Xiashangning