Xiashangning/BigSurface #15: [Discussion] Surface Ambient Light Sensor

Xiashangning, 2021-11-11:
Right now the `ALS` branch implements the functionality of the Ambient Light Sensor(ACSD) on Surface series. However it is not usable yet. Its readout is only valid when hot reboot from Windows(200-400 depends on light conditions). Cold boot will result in a readout of only 0-3 lux.
This issue is opened for the discussion of exploring the right initialization sequence of it. It is an APDS9960 device and the datasheet it here. https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf
You can consult to my driver's source code and try to modify it, mainly the initialization part happens in `initDevice()`
Anyone who wants to experience the light sensor functionality can switch to ALS branch, copy `Lilu` and `VirtualSMC`'s debug version under `Dependencies` folder and compile it.

liuson945, 2021-11-15:
夏大神，surface pro4 可以使用吗?需要修改什么？在哪里修改？

Xiashangning, 2021-11-15:
你要看一下你的光线传感器型号，比如上传一份ioreg和dsdt给我，但是这个光线传感器目前基本用不了，因为只有热启动才能正常读出光线强度，冷启动差点东西，应该是哪个寄存器需要enable吧
不过你可以用这个驱动来更好地支持你的typecover以及添加机身三个按钮功能

plsh2me, 2021-11-15:
I just wanted to share you the link -[ ](https://www.bilibili.com/read/cv10042865) , he claims the battery driver already developed and the touch development will take a few months.

llccd, 2021-11-19:
This seems to work on cold boot
```C
IOReturn SurfaceAmbientLightSensorDriver::initDevice() {
    IOReturn ret;
    UInt8 id;
    ENSURE(readRegister(APDS9960_ID, &id, 1))
    IOLog("%s::Device id is %x\n", getName(), id);
    
    ENSURE(writeRegister(APDS9960_ENABLE, 0x00))
    ENSURE(writeRegister(APDS9960_WTIME, TIME_TO_VALUE(400)))
    ENSURE(configDevice(ENABLE_WAIT, true))
    // set ADC integration time to 100 ms
    ENSURE(writeRegister(APDS9960_ATIME, TIME_TO_VALUE(100)))
    ENSURE(writeRegister(APDS9960_CONFIG1, 0x60))
    ENSURE(writeRegister(APDS9960_CONTROL, ALS_GAIN_16X))
    ENSURE(writeRegister(APDS9960_CONFIG2, 0x03))
    ENSURE(writeRegister(APDS9960_CONFIG3, 0x14))
    ENSURE(configDevice(ENABLE_POWER, true))
    configDevice(ENABLE_ALS, true);

    return kIOReturnSuccess;
}
```

Xiashangning, 2021-11-19:
Cool :)
That does work on my SP7.
BTW, do you know what each term means ? (CONFIG 1-3)

llccd, 2021-11-20:
I'm not sure, setting CONFIG1 to 0x60 and a higher WTIME may reduce power consumption.

About CONFIG2 and CONFIG3, Windows driver set then to 0x03 and 0x14 separately. Without one of them we will get a lower readout, and without both of them we can only get a readout of 0-3. I don't know what actually happens, since this strange behavior cannot be described by the datasheet.

Xiashangning, 2021-11-20:
Done and released :)

closed 2021-11-20 by Xiashangning

plsh2me, 2021-11-20:
@Xiashangning Hi, On my SP7 I installed the new release of the kext. It seems the kext loaded, but it not functioning properly. I put the brightness on maximum, then put the light in front of the sensor nothing happens. Although it's being selected on automatically adjust brightness. Do I need any additional settings for that? 
As we have the same matchine, can you please share your EFI folder.
[image]

Xiashangning, 2021-11-20:
try sudo dmesg | grep Light and tell me your output

plsh2me, 2021-11-20:
@Xiashangning Please check the output file.
[Ambient light.zip](https://github.com/Xiashangning/BigSurface/files/7574393/Ambient.light.zip)

Xiashangning, 2021-11-20:
it functions correctly
open the screen system preference window, adjusting the brightness low, use your phone’s flash light and watch the brightness bar, it should respond to bright light

Xiashangning, 2021-11-20:
oh, do you have a fake ali ssdt?

plsh2me, 2021-11-20:
@Xiashangning Yes, you were right. There was a fake dsdt als0 for it. I removed it, but still not working.The kext loaded (I can see on ioreg) . Please see the dsdt & kext list, if any problem with these.

[image]

plsh2me, 2021-11-21:
Hi, I found it's working. But it's not regular. As we know when the brightness is lower if we put the light in front of the sensor it will be go brighter. In my case it's not respond like this. It adjusts in a certain time, when it wants. 
But I am happy you are doing hard to make everything working. 
Thank you very much for your contribution.

sauregurke, 2021-11-30:
I'm having the same issue as @plsh2me - SurfaceAmbientLightSensorDriver is loaded and reading values but macOS is not responding to ALS changes, apparently because it isn't even showing the option in System Preferences to adjust the brightness automatically. I don't have SSDT-ALS0 or anything to do with the ambient light sensor; I tried booting with SMCLightSensor.kext but that did nothing either. It seems it boots with brightness turned all the way down until I increase it manually. IORegistryExplorer shows it is attached as an I2C device. Any ideas?