Xiashangning/BigSurface #16: SD Card Reader working, but showing as USB 3.0 - Possible fix?

billabongbruno, 2021-11-13:
Hey there again.
How is it going?

So, the SD card is fully working as the title indicated.
However, it shows up as a USB 3.0 device rather than Internal SD Card Reader (once you check under System Report).

It would be a nice touch to edit the BigSurface kext to address this purely "cosmetic" issue.

Someone has already come across this:
https://www.tonymacx86.com/threads/realtek-sd-card-reader.257141/page-2#post-2040364

Any thoughts?

Xiashangning, 2021-11-14:
Our card reader is connected via USB not PCI

closed 2021-11-14 by Xiashangning

therealjbenam, 2021-11-17:
> Hey there again. How is it going?
>
> So, the SD card is fully working as the title indicated. However, it shows up as a USB 3.0 device rather than Internal SD Card Reader (once you check under System [...]

Try looking at https://github.com/0xFireWolf/RealtekCardReaderFriend

It should do what you seek. I have already installed https://github.com/0xFireWolf/RealtekCardReader to great success and now finally my card reader on my SP6 is working.

I have tried RealtekCardReaderFriend but it stopped my OpenCore from booting. Probably was something I need to look at more in the config.

Xiashangning, 2021-11-18:
@therealjbenam So without this kext your card reader does not work ?

therealjbenam, 2021-11-18:
> @therealjbenam So without this kext your card reader does not work ?

Pretty sure it didn't work. Will try ASAP to disable the kext and I will try again.