Xiashangning/BigSurface #19: Request to add Keyboard Support for Surface laptop

plsh2me, 2021-12-06:
Hi, Can you please  add the keyboard and trackpad support for surface laptop 3. Device configuration is nearly same as surface pro 7(compare ioreg), macOS runs very well on it. May be other surface laptop's have the same issue - everything works except keyboard and trackpad.
Both keyboard & trackpad are connected with i2c >HID_device.  

Trackpad:
HID\Target_KIP&Category_HID&Col01
HID\VID_045E&PID_09AF&Col01&UP:0001_U:0002
HID\VID_045E&UP:0001_U:0002
HID_DEVICE_SYSTEM_MOUSE
HID_DEVICE_UP:0001_U:0002
HID_DEVICE

Keyboard:
HID\Target_KIP&Category_HID&Col01
HID\VID_045E&PID_09AE&Col01&UP:0001_U:0006
HID\VID_045E&UP:0001_U:0006
HID_DEVICE_SYSTEM_KEYBOARD
HID_DEVICE_UP:0001_U:0006
HID_DEVICE

I also found that someone has prepared 2 keyboard commits to add, but finally not added -https://github.com/VoodooI2C/VoodooI2CHID/pull/3 . 
At least if you can add this 2 commits and give a try if they works (You will find me here, if you need to test it).
Thank you
[ioreg surface laptop3.zip](https://github.com/Xiashangning/BigSurface/files/7663154/ioreg.surface.laptop3.zip)

Xiashangning, 2021-12-10:
Nope, this is a virtual device created by MS, it is not connected by i2c
Microsoft intentionally makes it that it communicates to the system by SAM which is also the case for battery, so a SAM driver which I am currently working on is needed for keyboard as well as trackpad support for SL series.
It is on the verge of releasing and still need testing. Maybe in this month or January it will be updated here.

plsh2me, 2021-12-10:
WOW!!! You make me happy. I will eagerly wait for that with a coffee on hand for you. 
Thank you very much.

plsh2me, 2022-01-03:
@Xiashangning Hi, You should know I was waiting for update from you. I installed the latest released kext on my surface laptop 3. As expected the battery bar is working and showing the correct percentage. 
About keyboard & trackpad- To be honest, this is beyond my knowledge to make a new register for these. So once again I request you to make a kext with the keyboard & trackpad. I am always here for testing and to provide any information you need.
I appreciate your effort.
Thank you

[MacBook Pro.zip](https://github.com/Xiashangning/BigSurface/files/7801953/MacBook.Pro.zip)

Xiashangning, 2022-01-03:
@plsh2me Well, that depends on whether I will have extra free time during winter vocation :) 
I will send you precompiled kext here if so.

plsh2me, 2022-01-03:
@Xiashangning Thank you so much for your assurance. I will wait for your reply. Please let me know if you need any additional information from me.

plsh2me, 2022-01-25:
@Xiashangning Hi, Hope you remember to add the support for keyboard & trackpad. If you have time, please try to add the support for next release or leave a compiled kext for testing purpose.
Thank you

plsh2me, 2022-02-13:
> @plsh2me Well, that depends on whether I will have extra free time during winter vocation :) I will send you precompiled kext here if so.

Wish this vocation you will manage time to send me at least one compiled kext.
Thank you

plsh2me, 2022-05-15:
> @plsh2me Well, that depends on whether I will have extra free time during winter vocation :) I will send you precompiled kext here if so.

Hi  @Xiashangning, I am happy to hear you are back. You are my only hope for make keyboard and trackpad working. Will you please try to give me the precompiled kext adding the keyboard & trackpad support. I was following the thread and waiting for you.

Hope you will give ma a good news.
Thank you

Xiashangning, 2022-05-16:
OK, the first thing I would like to make sure is that right now `BIgSurface` works normally on your SL3, and that battery shows properly

plsh2me, 2022-05-16:
@Xiashangning Yes, I can confirm you that the kext works on both macos BigSur and Monterey, shows battery properly.
Now the device has MacOS Monterey 12.3.1 installed.
Thank you

Xiashangning, 2022-05-16:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8699673/BigSurface.kext.zip)

Try this one. After booting, press some key on your keyboard and use external keyboard&mouse to go to Terminal then type `sudo dmesg | grep "SurfaceLaptop"`
Tell me your output

plsh2me, 2022-05-16:
Hi, Battery is sill working. Keyboard & Mouse are not responding. 
Here is the output of the terminal - Last login: Mon May 16 13:08:19 on ttys000
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceLaptop"
Password:
1777]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 3
[   89.751793]: SurfaceLaptopHIDDriver::Here
[  234.585274]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  234.585318]: SurfaceLaptopHIDDriver::Here
[  234.681385]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  234.681440]: SurfaceLaptopHIDDriver::Here
[  234.694396]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  234.694424]: SurfaceLaptopHIDDriver::Here
[  234.806160]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  234.806170]: SurfaceLaptopHIDDriver::Here
[  234.902075]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  234.902085]: SurfaceLaptopHIDDriver::Here
[  234.972510]: Sandbox apply: VTDecoderXPCServ[674] <bytes>SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.003790]: SurfaceLaptopHIDDriver::Here
[  235.116628]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.116655]: SurfaceLaptopHIDDriver::Here
[  235.211129]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.211168]: SurfaceLaptopHIDDriver::Here
[  235.313319]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.313340]: SurfaceLaptopHIDDriver::Here
[  235.423775]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.423797]: SurfaceLaptopHIDDriver::Here
[  235.520528]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.520559]: SurfaceLaptopHIDDriver::Here
[  235.622453]: SurfaceLaptopHIDDriver::Got report! iid: 1, cid: 0, data length: 12
[  235.622559]: SurfaceLaptopHIDDriver::Here

Xiashangning, 2022-05-16:
Good, these outputs correspond to your key press
Later I will give you another build

Xiashangning, 2022-05-16:
And are there any differences in the output if you touch your trackpad?

ajdarius, 2022-05-16:
Hope you can work this out. It will be really appreciated. You're doing a great job!

plsh2me, 2022-05-16:
> And are there any differences in the output if you touch your trackpad?

If I touch the trackpad only the keyboard back lit comes on(before was the same) . Rather than I don't see any visual difference.

Xiashangning, 2022-05-16:
I mean of course the `dmesg` output...

plsh2me, 2022-05-16:
This is the last output after reboot - pressing and touching the trackpad.
Password:
[ 1383.200692]: SurfaceLaptopHIDDriver::Here
[ 1383.215257]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.215312]: SurfaceLaptopHIDDriver::Here
[ 1383.222632]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.222688]: SurfaceLaptopHIDDriver::Here
[ 1383.229999]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.230056]: SurfaceLaptopHIDDriver::Here
[ 1383.237382]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.237421]: SurfaceLaptopHIDDriver::Here
[ 1383.244752]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.244808]: SurfaceLaptopHIDDriver::Here
[ 1383.252134]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.252168]: SurfaceLaptopHIDDriver::Here
[ 1383.259480]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.259535]: SurfaceLaptopHIDDriver::Here
[ 1383.268433]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.268472]: SurfaceLaptopHIDDriver::Here
[ 1383.275266]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.275302]: SurfaceLaptopHIDDriver::Here
[ 1383.282124]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.282170]: SurfaceLaptopHIDDriver::Here
[ 1383.290072]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.290128]: SurfaceLaptopHIDDriver::Here
[ 1383.298849]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.298905]: SurfaceLaptopHIDDriver::Here
[ 1383.306186]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.306218]: SurfaceLaptopHIDDriver::Here
[ 1383.313628]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.313661]: SurfaceLaptopHIDDriver::Here
[ 1383.320963]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.320996]: SurfaceLaptopHIDDriver::Here
[ 1383.328303]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.328338]: SurfaceLaptopHIDDriver::Here
[ 1383.335715]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.335770]: SurfaceLaptopHIDDriver::Here
[ 1383.343107]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.343163]: SurfaceLaptopHIDDriver::Here
[ 1383.350497]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.350553]: SurfaceLaptopHIDDriver::Here
[ 1383.357687]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.357720]: SurfaceLaptopHIDDriver::Here
[ 1383.365064]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.365120]: SurfaceLaptopHIDDriver::Here
[ 1383.372415]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.372472]: SurfaceLaptopHIDDriver::Here
[ 1383.379603]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.379636]: SurfaceLaptopHIDDriver::Here
[ 1383.386897]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.386954]: SurfaceLaptopHIDDriver::Here
[ 1383.394272]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.394306]: SurfaceLaptopHIDDriver::Here
[ 1383.401603]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.401647]: SurfaceLaptopHIDDriver::Here
[ 1383.408989]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.409021]: SurfaceLaptopHIDDriver::Here
[ 1383.416326]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.416358]: SurfaceLaptopHIDDriver::Here
[ 1383.423710]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.423766]: SurfaceLaptopHIDDriver::Here
[ 1383.431061]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.431117]: SurfaceLaptopHIDDriver::Here
[ 1383.438399]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.438458]: SurfaceLaptopHIDDriver::Here
[ 1383.448201]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.448237]: SurfaceLaptopHIDDriver::Here
[ 1383.453209]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1383.453257]: SurfaceLaptopHIDDriver::Here
[ 1383.462955]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[... 1225 lines cut ...]
[ 1389.223312]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.223368]: SurfaceLaptopHIDDriver::Here
[ 1389.230686]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.230743]: SurfaceLaptopHIDDriver::Here
[ 1389.238038]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.238070]: SurfaceLaptopHIDDriver::Here
[ 1389.245384]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.245441]: SurfaceLaptopHIDDriver::Here
[ 1389.252758]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.252790]: SurfaceLaptopHIDDriver::Here
[ 1389.260141]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.260173]: SurfaceLaptopHIDDriver::Here
[ 1389.267486]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.267544]: SurfaceLaptopHIDDriver::Here
[ 1389.274871]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.274903]: SurfaceLaptopHIDDriver::Here
[... cut ...] report! iid: 3, cid: 0, data length: 6
[ 1389.282282]: SurfaceLaptopHIDDriver::Here
[ 1389.289388]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.289428]: SurfaceLaptopHIDDriver::Here
[ 1389.296482]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.296515]: SurfaceLaptopHIDDriver::Here
[ 1389.303837]: SurfaceLaptopHIDDriver::Got report! iid: 3, cid: 0, data length: 6
[ 1389.303894]: SurfaceLaptopHIDDriver::Here
mdpalashmiah@MDs-MacBook-Pro ~ %

Xiashangning, 2022-05-16:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8700280/BigSurface.kext.zip)
try this one
press and release `a` and `b` then touch once on touchpad, save the output of the terminal and give it to me

plsh2me, 2022-05-16:
No output.

Last login: Mon May 16 14:22:31 on console
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceLaptop"
Password:
mdpalashmiah@MDs-MacBook-Pro ~ %

Xiashangning, 2022-05-16:
> Update: Something happened inside, I left the laptop right on this page. When I return I found the laptop freezed instead of sleep and then I tried to wake up but after a while rebooted.

Hmm, there might be some bug, I experienced that too. Right now don't mind that

plsh2me, 2022-05-16:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8700958/BigSurface.kext.zip)
> This one then
Last login: Mon May 16 15:51:16 on console
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceLaptop"
Password:
[   45.056676]: 6 duplicate reports for Sandbox: SafariBookmarksS(507) deny(1) user-preference-read com.apple.CloudKitSandbox: swcd(510) deny(1) system-fsctl _IO('h', 47)SurfaceLaptopHIDDriver::Got input with iid 1
[   45.057416]: SurfaceLaptopHIDDriver::01 00 04 00 00 00 00 00 00 00 00 00
[   45.161904]: SurfaceLaptopHIDDriver::Got input with iid 1
[   45.161927]: SurfaceLaptopHIDDriver::01 00 04 00 00 00 00 00 00 00 00 00
[   45.273812]: SurfaceLaptopHIDDriver::Got input with iid 1
[   45.273821]: SurfaceLaptopHIDDriver::01 00 04 00 00 00 00 00 00 00 00 00
................
Minimum 50 entry of same thing.
And after pressing the keyboard & touching the trackpad lost the battery percentage.
[image]

Xiashangning, 2022-05-16:
can you reboot and try this `press and release a and b then touch once on touchpad`?

plsh2me, 2022-05-16:
I am hopeful. 
[ioreg.zip](https://github.com/Xiashangning/BigSurface/files/8703846/ioreg.zip)

[image]

Xiashangning, 2022-05-17:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8705296/BigSurface.kext.zip)
This time, I don't need Terminal output, instead, send me your ioreg.

Xiashangning, 2022-05-17:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8705808/BigSurface.kext.zip)
Hopefully it works, then this build would possibly make SL3 keyboard functions properly

plsh2me, 2022-05-17:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8705296/BigSurface.kext.zip)
> send me your ioreg and output of `sudo dmesg | grep "SurfaceLaptop"`

[   80.243684]: SurfaceLaptopHIDDriver::Got Surface Serial Hub! SurfaceSerialHubDriver
[   80.343728]: SurfaceLaptopKeyboard:0x1000007f4 start
[   80.352609]: SurfaceLaptopKeyboard::report descriptor len 128
[   80.366093]: SurfaceLaptopKeyboard:0x1000007f4 failed to parse report descriptor
[   80.366095]: SurfaceLaptopKeyboard:0x1000007f4 start failed
[   80.366097]: SurfaceLaptopKeyboard:0x1000007f4 stop
[   80.366112]: SurfaceLaptopHIDDriver::Failed to init Surface Laptop Keyboard Device, exiting
[   82.343910]: SurfaceLaptopHIDDriver::Got Surface Serial Hub! SurfaceSerialHubDriver
[   82.444184]: SurfaceLaptopKeyboard:0x1000007f9 start
[   82.453370]: SurfaceLaptopKeyboard::report descriptor len 128
[   82.467159]: SurfaceLaptopKeyboard:0x1000007f9 failed to parse report descriptor
[   82.467161]: SurfaceLaptopKeyboard:0x1000007f9 start failed
[   82.467163]: SurfaceLaptopKeyboard:0x1000007f9 stop
[   82.467178]: SurfaceLaptopHID
Driver::Failed to init Surface Laptop Keyboard Device, exiting

[ioreg2.zip](https://github.com/Xiashangning/BigSurface/files/8705885/ioreg2.zip)

plsh2me, 2022-05-17:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8705808/BigSurface.kext.zip)
> Hopefully it works, then this build would possibly make SL3 keyboard functions properly

No output with command.

[ioreg3.zip](https://github.com/Xiashangning/BigSurface/files/8705900/ioreg3.zip)

Xiashangning, 2022-05-17:
Okay, that's too bad
Ignore the last build, use the previous build and try `sudo dmesg | grep "Surface"`
There might be a lot of output, you can save it to a file

Xiashangning, 2022-05-17:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8705994/BigSurface.kext.zip)
I add some code, use this one, I need terminal output and ioreg

Xiashangning, 2022-05-17:
I need `sudo dmesg | grep "Surface"`'s output

Xiashangning, 2022-05-17:
Does your keyboard works for now? There should be some output, have you load `DebugEnhancer` correctly? Because I don't see it loaded, that might be the reason why you won't get any output

plsh2me, 2022-05-17:
No, Keyboard is not working. 

Last login: Tue May 17 13:55:47 on console
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "Surface"
Password:
[   55.312402]: SurfaceSerialHubDriver::Warning, buffer size(4) and rx_data_len(12) mismatched!
mdpalashmiah@MDs-MacBook-Pro ~ %

plsh2me, 2022-05-17:
Sorry, I don't know this. What I have to do for that?

Xiashangning, 2022-05-17:
After you pressed some keys on keyboard, is there any new outputs ?

plsh2me, 2022-05-17:
> After you pressed some keys on keyboard, is there any new outputs ?

Yes, all the previous output and new -
[  420.397318]: SurfaceSerialHubDriver::Warning, buffer size(4) and rx_data_len(3) mismatched!

Xiashangning, 2022-05-17:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8709577/BigSurface.kext.zip)
ioreg & `sudo dmesg | grep "SurfaceLaptop"` & `sudo dmesg | grep "SurfaceSerial"`

Xiashangning, 2022-05-17:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8709603/BigSurface.kext.zip)
Use this new one sry

Xiashangning, 2022-05-18:
Try this one, before you boot macOS, first boot once Windows, then reboot to macOS
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8713063/BigSurface.kext.zip)

plsh2me, 2022-05-18:
Ok, rebooted the device- cleaned nvram. Booted to windows and then booted to Mac.
Output - 
Last login: Wed May 18 04:59:14 on console
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceLaptop"
Password:
[    0.672928]: SurfaceLaptopHIDDriver::Got Surface Serial Hub! SurfaceSerialHubDriver
[    0.776433]: SurfaceLaptopHIDDriver::Publishing keyboard nub
[    0.776450]: SurfaceLaptopKeyboardDriver:0x1000002e6 start
[    0.781418]: SurfaceLaptopHIDDriver::Got data with offset: 0, length: 9, finished: 1
[    0.788866]: SurfaceLaptopHIDDriver::Got data with offset: 0, length: 32, finished: 1
[    0.788873]: SurfaceLaptopKeyboard:0x1000002e8 start
[    0.803741]: SurfaceLaptopHIDDriver::Got data with offset: 0, length: 118, finished: 0
[    0.808689]: SurfaceLaptopHIDDriver::Got data with offset: 118, length: 10, finished: 1
[    0.810719]: SurfaceLaptopKeyboard:0x1000002e8 creating interfaces
[    0.815450]: SurfaceLaptopHIDDriver::Woke up
[    6.334942]: SurfaceLaptopKeyboard:0x1000002e8 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[    6.495687]: SurfaceLaptopKeyboard:0x1000002e8 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[    8.398396]: Sandbox apply: distnoted[307] <bytes>SurfaceLaptopKeyboard:0x1000002e8 message: 0xe0000101 from: 0x1000002e6 1
[   31.733916]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   31.791674]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   31.848602]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   31.931229]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.043897]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.127101]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.237866]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.331436]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.342070]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.453828]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.548627]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.661385]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.755781]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.867360]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   32.960625]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   33.073546]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   33.168492]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   33.270754]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   33.382931]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.085708]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.137841]: Sandbox apply: parsec-fbf[469] <bytes>SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.180123]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.292233]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.351302]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.399846]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.408961]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.510651]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.558726]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.582278]: Sandbox: CommCenter(399) deny(1) mach-lookup com.apple.lsd.mapdbSystem Policy: CommCenter(399) deny(1) file-read-data /private/var/folders/tb/r9q2t8ss0h37snxqdczvjm_r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSystem Policy: CommCenter(399) deny(1) file-read-metadata /private/var/folders/tb/r9q2t8ss0h37snxqdczvjm_r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.616840]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.619083]: 2 duplicate reports for System Policy: CommCenter(399) deny(1) file-read-metadata /private/var/folders/tb/r9q2t8ss0h37snxqdczvjm_r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSandbox: CommCenter(399) deny(1) mach-lookup com.apple.lsd.modifydbSurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.773906]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.799039]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.825337]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.839160]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.904958]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   43.962079]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   44.006497]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[   44.022570]: SurfaceLaptopKeyboardDriver::Forwarding report to keyboard device
[... cut ...]
[    0.568938]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.571542]: SurfaceSerialHubDriver::SAM version 14.502.139
[    0.571641]: SurfaceBatteryDriver::Got Surface Serial Hub! SurfaceSerialHubDriver
[    0.672928]: SurfaceLaptopHIDDriver::Got Surface Serial Hub! SurfaceSerialHubDriver
[    0.808683]: SurfaceSerialHubDriver::Warning, buffer size(128) and rx_data_len(20) mismatched!
mdpalashmiah@MDs-MacBook-Pro ~ % 

[ioreg6.zip](https://github.com/Xiashangning/BigSurface/files/8713171/ioreg6.zip)

Xiashangning, 2022-05-18:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8713412/BigSurface.kext.zip)
Try this one

plsh2me, 2022-05-18:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8713412/BigSurface.kext.zip)
> Try this one

WOW!!!!!!! It works. 
So the keyboard is done.
I tested it for a while, Looks all the buttons are fine. I have the British keyboard and have some keys different from original apple's keys. Anyway that's not a problem. But the keyboard now works. 

[output.zip](https://github.com/Xiashangning/BigSurface/files/8714416/output.zip)
[ioreg7.zip](https://github.com/Xiashangning/BigSurface/files/8714417/ioreg7.zip)

Please tell me if you need anything else to test. 
And looking forward for the trackpad.

Thank you. I really appreciate your effort.

Xiashangning, 2022-05-18:
> I have the British keyboard and have some keys different from original apple's keys.

This can be solved later

Xiashangning, 2022-05-18:
@plsh2me  Since it works for now and the code is basically all mixed together, I need to reorganise the project a little, later I will send you another build to test if it still works.

plsh2me, 2022-05-18:
Ok, That's fine. I will be with you anytime.

Xiashangning, 2022-05-18:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8714751/BigSurface.kext.zip)
Is this build working ?
PS: I also changed the country code to 0x20->UK, if this one works, is there still any key wrong?

plsh2me, 2022-05-18:
This one is better. Fn keys works now (specially brightness and volume keys... ). Only four keys are alternated to each others (see the photo). 
[image]

plsh2me, 2022-05-18:
Update: After sleep the keyboard not working. After rebooting it works again.

Xiashangning, 2022-05-18:
Emm, if you go to Keyboard in System Preference, there is a key map of your currently keyboard, does that corresponding to you keyboard layout or the actually layout (the one that macOS thinks)

Xiashangning, 2022-05-18:
> Update: After sleep the keyboard not working. After rebooting it works again.

Noted, will fix that after refactoring the code

plsh2me, 2022-05-18:
> And about the keyboard setup assistant, does it shows up again ?

Yes, it shows up again and I set it up.
About keyboard map you are right. I changed it to British PC, now OK.

Xiashangning, 2022-05-18:
Good, so now the only problem is that it fails after sleeping, right ?

plsh2me, 2022-05-18:
> Good, so now the only problem is that it fails after sleeping, right ?

Yes.

plsh2me, 2022-05-18:
> Good, so now the only problem is that it fails after sleeping, right ?

Now it seems waking up normally. I tested manual sleep and normally keep it to sleep. Both time worked, lets hope for the good.

Xiashangning, 2022-05-18:
Nope, the reason why it fails is that the waking up sequence is not coordinated with SurfaceSerialHubDriver, need just a little code

rubel-ahammad, 2022-05-19:
Hi @Xiashangning , it is great that you finally managed some time for this part of this project and finally added keyboard support. I appreciate the work that you are doing. FYI, I was really interested in this part of this project. I also planned to start working on this. But, I did not know where to start.

Xiashangning, 2022-05-20:
@rubel-ahammad Well, that is too bad for you, I've done the code. :)
As for where to start, I have already made it clear in README, just check the linux codes for `surface_hid` and write an IOService driver similar to `SurfaceBattery` driver and that will be enough.
How to communicate to SSH -> `SurfaceBattery`
How to send HID event -> `SurfaceButton`
That's it. 
In fact it only took me 2 days to write a workable (but not clean and elegant) code.

Xiashangning, 2022-05-20:
@plsh2me Please test this build. You have to change the `KBDCountryCode` in Info.plist to 32 which corresponds to British keyboard layout
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8735797/BigSurface.kext.zip)

plsh2me, 2022-05-20:
> @plsh2me Please test this build. You have to change the `KBDCountryCode` in Info.plist to 32 which corresponds to British keyboard layout
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8735797/BigSurface.kext.zip)

Ok, Keyboard works as expected. I haven't add any entry to the plist for the country code. You should remember I already have the keyboard layout British PC. And seems keyboard keys corresponds with this correctly and all keys are works.
About trackpad nothing, think you still not added it.

Output - 
Last login: Fri May 20 07:22:35 on console
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceLaptop"
Password:
[    0.579936]: SurfaceLaptop3Nub::Woke up
[    0.606962]: SurfaceLaptopKeyboard:0x100000359 start
[    0.622682]: SurfaceLaptop3HIDDriver::Woke up
[    0.622694]: SurfaceLaptopKeyboard:0x100000359 message: 0xe0000230 from: 0x100000358 0
[    0.622849]: SurfaceLaptopKeyboard:0x100000359 creating interfaces
[    5.597900]: SurfaceLaptopKeyboard:0x100000359 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[    5.789450]: SurfaceLaptopKeyboard:0x100000359 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[  105.964599]: SurfaceLaptopKeyboard:0x100000359 new user client
mdpalashmiah@MDs-MacBook-Pro ~ % sudo dmesg | grep "SurfaceSerial"
[    0.429626]: SurfaceSerialHubDriver::Found valid UART Bus!
[    0.470800]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.571882]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.574389]: SurfaceSerialHubDriver::SAM version 14.502.139
[    0.574422]: SurfaceSerialHubDriver::Surface Battery nub published!
[    0.579874]: SurfaceSerialHubDriver::Surface Laptop3 nub published!
[    0.597191]: SurfaceSerialHubDriver::Woke up
mdpalashmiah@MDs-MacBook-Pro ~ % 

[ioreg8.zip](https://github.com/Xiashangning/BigSurface/files/8736968/ioreg8.zip)

Xiashangning, 2022-05-20:
And does it still work after a sleep-wake cycle ?

[18 more replies]

closed 2022-05-20 by Xiashangning