Xiashangning/BigSurface #21: Battery Support not working on SP6

therealjbenam, 2022-01-04:
Hi,

I have just installed the latest Kext along with the supplied SSDT.

It seems that it's not working as it's stuck in "charging" mode.

Please let me know if you need anything from me to help you support the SP6.

Thanks,
Best Regards

Xiashangning, 2022-01-04:
Good, it should work on SP6 :)

therealjbenam, 2022-01-04:
> Good, it should work on SP6 :)

Sadly it's not working :( After disabling SMCBatteryManager (BigSurface couldn't assign a battery class since it had the same name) I just don't get any battery.

Xiashangning, 2022-01-04:
I will need `sudo dmesg | grep UART` output and ioreg

therealjbenam, 2022-01-04:
> I will need `sudo dmesg | grep UART` output and ioreg Also, what version of macOS you are running? Big Sur or Monterey?

`sudo dmesg | grep UART` returns empty.

Monterey - I have found something interesting in the kernel logs:

```
2022-01-04 17:21:27.327958+0100  localhost kernel[0]: (VoodooI2CServices) <compose failure [UUID]>
2022-01-04 17:21:27.327965+0100  localhost kernel[0]: (VoodooI2CServices) <compose failure [UUID]>
2022-01-04 17:21:27.327969+0100  localhost kernel[0]: (VoodooI2CServices) <compose failure [UUID]>
2022-01-04 17:21:27.327971+0100  localhost kernel[0]: (VoodooI2CServices) <compose failure [UUID]>
2022-01-04 17:21:27.327992+0100  localhost kernel[0]: IOCommandGate::disable() called when not gated
Backtrace 0x0 0x0 0xffffff8000a2ec2a 0xffffff8000a2e795 0xffffff8000a2d790 0xffffff8000a30839 0xffffff800022718e
2022-01-04 17:21:27.327995+0100  localhost kernel[0]:       Kernel Extensions in backtrace:
2022-01-04 17:21:27.328003+0100  localhost kernel[0]:          com.xavier.BigSurface(1.0.4)[79A27E32-31E7-38FF-B421-86DFD70E2246]@0x0->0x2cfff
2022-01-04 17:21:27.328010+0100  localhost kernel[0]:             dependency: as.vit9696.Lilu(1.5.7)[E42A54DC-9082-35EC-93BC-08AD09861180]@0x0->0x2afff
2022-01-04 17:21:27.328015+0100  localhost kernel[0]:             dependency: as.vit9696.VirtualSMC(1.2.7)[9DAADEEB-D155-3EEC-AFDF-2F05E221EA01]@0x0->0x16fff
2022-01-04 17:21:27.328021+0100  localhost kernel[0]:             dependency: com.alexandred.VoodooI2CServices(1)[8DAB890A-4FBF-3727-96FB-076E476E3AF6]@0x0->0x7fff
2022-01-04 17:21:27.328027+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOACPIFamily(1.4)[B750EC9D-D224-309C-8C32-3F32C6257F1E]@0xffffff800293f000->0xffffff8002940fff
2022-01-04 17:21:27.328034+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOHIDFamily(2.0.0)[4773BB64-73AE-3283-8DE4-D10459C90231]@0xffffff8002a7d000->0xffffff8002adafff
2022-01-04 17:21:27.328040+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOSMBusFamily(1.1)[FBCF7261-2104-3F79-ADCD-C3DDD5B63986]@0xffffff8002eb0000->0xffffff8002eb1fff
2022-01-04 17:21:27.328058+0100  localhost kernel[0]:             dependency: com.xavier.VoodooSerial(1.0.0)[42FF04CF-69AE-3219-9FE6-5647DEAB01C8]@0x0->0x2ffff
2022-01-04 17:21:27.328067+0100  localhost kernel[0]:             dependency: org.coolstar.VoodooGPIO(1.2.0)[F3ED6FA0-F8A5-3384-8367-C3D3E1BE802D]@0x0->0x17fff
2022-01-04 17:21:27.352168+0100  localhost kernel[0]: (Sandbox) <Sandbox`is_release_type> release type: User
2022-01-04 17:21:27.352631+0100  localhost kernel[0]: (ALF) <ALF`sockwall_cntl_updaterules> ALF, old data swfs_pid_entry <private>, updaterules_msg <private>, updaterules_state <private>
```

Looks like it's crashing.

Xiashangning, 2022-01-04:
Well, you need `keepsyms=1` to get the location of error

therealjbenam, 2022-01-04:
> Well, you need `keepsyms=1` to get the location of error

I do have it already in my bootargs :(

Xiashangning, 2022-01-05:
UART driver seems to load fine but logs have to be checked. SurfaceSerialHub failed to load and that is the problem.
I'll need you to enable the verbose mode `-v` and take some screen shots when booting. Specifically, the early booting prints contains `UART` and `Surface` at the very beginning of macOS booting sequence.
This is due to Monterey breaks early boot logs so you cannot find them in `dmesg`
PS: I hope you have a Big Sur operating system so it will be a lot easier. If so, then `sudo dmesg | grep UART` and `sudo dmesg | grep Surface` will return sth.

therealjbenam, 2022-01-05:
> UART driver seems to load fine but logs have to be checked. SurfaceSerialHub failed to load and that is the problem. I'll need you to enable the verbose mode `-v` and take some screen shots when [...]

Sadly I don't, this is my only SP6 system. I guess I can just load a Big Sur installer from USB and use the terminal from there?

Xiashangning, 2022-01-05:
Yes, you can, and send me output of `sudo dmesg | grep UART` and `sudo dmesg | grep Surface`

therealjbenam, 2022-01-05:
> Yes, you can, and send me output of `sudo dmesg | grep UART` and `sudo dmesg | grep Surface`

Sorry for the late reply, creating the installer for Big Sur took far longer than expected.

sudo dmesg | grep UART
```
Jan  5 16:01:32 localhost kernel[0]: [    4.010070]: SurfaceSerialHubDriver::Found valid UART Bus!
Jan  5 16:01:32 localhost kernel[0]: [    5.613421]: VoodooUARTController::UA00 Set PCI power state D0
Jan  5 16:01:32 localhost kernel[0]: [    5.613458]: VoodooUARTController::UA00 Designware UART version 3.14
Jan  5 16:01:32 localhost kernel[0]: [    5.613714]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
Jan  5 16:01:32 localhost kernel[0]: [    5.714334]: VoodooUARTController::UA00 Warning: baudrate should be 4000000, 3000000
```

sudo dmesg | grep Surface
```
Jan  5 16:01:32 localhost kernel[0]: [    4.010070]: SurfaceSerialHubDriver::Found valid UART Bus!
Jan  5 16:01:32 localhost kernel[0]: [    4.010075]: SurfaceSerialHubDriver::Surface Serial Hub found!
Jan  5 16:01:32 localhost kernel[0]: [    4.013046]: SurfaceButtonDriver::Found valid GPIO interrupts
Jan  5 16:01:32 localhost kernel[0]: [    4.013059]: SurfaceButtonDriver::Surface ACPI button device found!
Jan  5 16:01:32 localhost kernel[0]: [    4.048911]: SurfaceButtonDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
Jan  5 16:01:32 localhost kernel[0]: [    4.106503]: IOSurfaceRoot::installMemoryRegions()
Jan  5 16:01:32 localhost kernel[0]: [    4.106515]: IOSurface allowing global lookups (with warnings)
Jan  5 16:01:32 localhost kernel[0]: [    4.150176]: SurfaceButtonHIDDevice:0x1000002f0 start
Jan  5 16:01:32 localhost kernel[0]: [    4.150212]: SurfaceButtonHIDDevice:0x1000002f0 message: 0xe0000230 from: 0x1000002b5 0
Jan  5 16:01:32 localhost kernel[0]: [    4.151619]: SurfaceButtonDriver::started
Jan  5 16:01:32 localhost kernel[0]: [    4.151814]: SurfaceButtonHIDDevice:0x1000002f0 creating interfaces
Jan  5 16:01:32 localhost kernel[0]: [    5.613714]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
Jan  5 16:01:36 MacBook-Pro kernelmanagerd[55]: Error occurred while handling request LoadNotification(arguments: Optional(["CFBundleIdentifier": com.xavier.BigSurface])): System policy error: System policy is unavailable.
Jan  5 16:01:36 MacBook-Pro kernel[0]: [   46.083013]: SurfaceButtonHIDDevice:0x1000002f0 open by AppleUserHIDEventDriver 0x1000003f9 (0x0)
Jan  5 16:01:36 MacBook-Pro kernel[0]: [   46.083498]: [AppleUserHIDEventDriver.cpp:93][0x1000003f9] Surface Button HID Device usagePage: 12 usage: 1 vid: 1452 pid: 630
Jan  5 16:01:37 MacBook-Pro kernel[0]: [   46.104012]: AppleUserUSBHostHIDDevice:0x10000040c open by SurfaceTypeCoverDriver 0x10000044d (0x0)
Jan  5 16:01:37 MacBook-Pro kernel[0]: [   46.104119]: SurfaceTypeCoverDriver::Surface Type Cover Putting device into Precision Touchpad Mode
Jan  5 16:01:37 MacBook-Pro kernel[0]: [   46.127447]: SurfaceTypeCoverDriver::Magic Trackpad 2 Could not parse multitouch elements
Jan  5 16:01:38 MacBook-Pro kernel[0]: [   47.721419]: SurfaceButtonHIDDevice:0x1000002f0 message: 0xe0000101 from: 0x1000002b5 1
```

Thanks again, I hope this will help you in finding the issue.

Xiashangning, 2022-01-05:
well, as far as i can tell, there are two problems here. We should first handle the UART one(the last output of grep UART)
And that is very tricky, in fact the baudrate is calculated from uart clk, and it seems that it has a different clk than the one on SP7, you have to install linux-surface kernel and modify it a little to find out the proper clk for UART
Check the linux-surface repo, install ubuntu and compile the kernel yourself. I will tell you where to modify tomorrow(it is already midnight here), and you can first do the preliminary things today

therealjbenam, 2022-01-05:
Sure, I will gladly help. Need to sort out triple-booting first (that should be easy) and why Ubuntu hangs and crashes on startup...

Xiashangning, 2022-01-06:
Indeed weird, but Opencore can recognise Linux distro and directly boot them. You can try that
You can use any Linux distribution as you like, as long as you can compile and install custom kernel.

Xiashangning, 2022-01-06:
Once you succeed, go to `drivers/tty/serial/8250/8250_port.c` and in `serial8250_do_set_divisor` add these codes
```
dev_err(port->dev, "----mydebug uartclk: %u\n", port->uartclk);
dev_err(port->dev, "----mydebug baudrate: %u\n", baud);
dev_err(port->dev, "----mydebug quot: %u\n", quot);
dev_err(port->dev, "----mydebug 0x200: %lu\n", readl(port->membase+0x200));
dev_err(port->dev, "----mydebug 0x204: %lu\n", readl(port->membase+0x204));
dev_err(port->dev, "----mydebug 0x208: %lu\n", readl(port->membase+0x208));
dev_err(port->dev, "----mydebug 0x800: %lu\n", readl(port->membase+0x800));
dev_err(port->dev, "----mydebug 0x804: %lu\n", readl(port->membase+0x804));
```
compile and boot this kernel
in terminal `sudo dmesg | grep mydebug` and tell me the output

therealjbenam, 2022-01-07:
Hi @Xiashangning 

Since I knew I had done everything right last time, I have just reinstalled Ubuntu and it worked this time. Go figure.

I have edited the source and added the required comment, but it seems that it stops here

```
...other stuff
  BTF [M] sound/x86/snd-hdmi-lpe-audio.ko
  BTF [M] sound/xen/snd_xen_front.ko
make KERNELRELEASE=5.13.19-linux-surface ARCH=x86 	KBUILD_BUILD_VERSION=4 -f ./Makefile intdeb-pkg
sh ./scripts/package/builddeb
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/aegis128-aesni.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/aesni-intel.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blake2s-x86_64.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blowfish-x86_64.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast5-avx-x86_64.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast6-avx-x86_64.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/chacha-x86_64.ko
  INSTALL debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/crc32-pclmul.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blake2s-x86_64.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blowfish-x86_64.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast5-avx-x86_64.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/aegis128-aesni.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/chacha-x86_64.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/crc32-pclmul.ko
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast6-avx-x86_64.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast5-avx-x86_64.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast5-avx-x86_64.ko'
make[6]: *** Waiting for unfinished jobs....
  SIGN    debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/aesni-intel.ko
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blake2s-x86_64.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blake2s-x86_64.ko'
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blowfish-x86_64.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/blowfish-x86_64.ko'
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/chacha-x86_64.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/chacha-x86_64.ko'
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: : No such file or directory
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/crc32-pclmul.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/crc32-pclmul.ko'
make[6]: *** [scripts/Makefile.modinst:83: debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast6-avx-x86_64.ko] Error 1
make[6]: *** Deleting file 'debian/linux-image/lib/modules/5.13.19-linux-surface/kernel/arch/x86/crypto/cast6-avx-x86_64.ko'
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
[... cut ...]
make[5]: *** [Makefile:1793: modules_install] Error 2
make[4]: *** [scripts/Makefile.package:87: intdeb-pkg] Error 2
make[3]: *** [Makefile:1585: intdeb-pkg] Error 2
make[2]: *** [debian/rules:13: binary-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:83: bindeb-pkg] Error 2
make: *** [Makefile:1585: bindeb-pkg] Error 2
```

I know it hasn't got anything strictly to do with BigSurface, but I wonder if you had the same issues and how you solved them?

Thanks

Xiashangning, 2022-01-07:
Well, i just followed the guide on linux
-surface repo (how to compile the kernel)
and haven’t met your problem
I am not a expert for linux so i suggest that you can try follow that

therealjbenam, 2022-01-07:
> Well, i just followed the guide on linux -surface repo (how to compile the kernel) and haven’t met your problem I am not a expert for linux so i suggest that you can try follow that

Yeah I am following that :( Maybe it has something to do with the kernel version? I am using 5.13 - which one did you use for testing?

Xiashangning, 2022-01-23:
@therealjbenam Have you succeeded in installing Ubuntu or any Linux distro and compile a custom kernel?

therealjbenam, 2022-01-26:
> @therealjbenam Have you succeeded in installing Ubuntu or any Linux distro and compile a custom kernel?

I have correctly installer Ubuntu but I can't compile surface-kernel as I get a boatload of errors when compiling crypto. If you still have your build, could you please tell me which kernel did you use and which config file did you point the build at? I think there is something going wrong with the merge_config.sh.

Never had issues when compiling standard kernels in the past.

therealjbenam, 2022-02-11:
> @therealjbenam I have reinstalled Ubuntu, modified the kernel and compile a workable kernel for you, try this. Using `sudo dpkg -i *.deb` [surface-ubuntu.zip](https://github.com/Xiashangning/BigSurface/files/8040960/surface-ubuntu.zip) Once you have installed, boot into this kernel and in terminal input `sudo dmesg | grep mydebug` and tell me the output

Thank you very much for this! Installed it successfully and got the data you need.

Please let me know if you need anything else.

```
[    2.428617] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 153600
[    2.428781] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 9600
[    2.428931] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.429070] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2352283657
[    2.429208] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.429367] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.429482] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.429600] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.429727] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 153600
[    2.429859] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 9600
[    2.429984] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.430101] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2352283657
[    2.430231] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.430352] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.430478] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.430597] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.430755] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.430889] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.431024] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.431143] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.431276] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.431419] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.431539] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.431662] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.431789] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.431922] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.432055] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.432200] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.432354] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.432482] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.432604] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.432723] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.461564] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.461732] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.461881] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.461997] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.462130] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.462249] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.462368] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.462485] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.462614] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.462745] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.462874] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.462989] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.463145] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.463262] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.463378] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.463494] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.463642] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.464111] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.464423] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.466344] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.469770] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.472336] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.477806] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    2.486818] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    2.560585] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    2.560600] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    2.560604] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    2.560609] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    2.560614] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    2.560618] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    2.560622] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[... 281 lines cut ...]
[    4.256494] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    4.256548] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    4.256593] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    4.256633] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    4.256675] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    4.256715] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    4.256755] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    4.256795] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    4.256841] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    4.256883] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    4.256926] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    4.256963] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[... cut ...]
[    4.257210] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    4.257256] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    4.257300] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    4.257340] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    4.257385] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    4.257424] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    4.257464] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    4.257505] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
```

Xiashangning, 2022-02-11:
@therealjbenam Try this build
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8048482/BigSurface.kext.zip)

therealjbenam, 2022-02-11:
> @therealjbenam Try this build [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8048482/BigSurface.kext.zip)

Still crashes. Will look for my Big Sur installer USB later.

```
2022-02-11 14:03:01.566077+0100  localhost kernel[0]: IOCommandGate::disable() called when not gated
Backtrace 0x0 0x0 0xffffff8000a2ec2a 0xffffff8000a2e795 0xffffff8000a2d790 0xffffff8000a30839 0xffffff800022718e
2022-02-11 14:03:01.566081+0100  localhost kernel[0]:       Kernel Extensions in backtrace:
2022-02-11 14:03:01.566088+0100  localhost kernel[0]:          com.xavier.BigSurface(1.0.4)[B5442712-01D2-3B0A-B627-4640C8EEDD5A]@0x0->0x39fff
2022-02-11 14:03:01.566095+0100  localhost kernel[0]:             dependency: as.vit9696.Lilu(1.5.7)[E42A54DC-9082-35EC-93BC-08AD09861180]@0x0->0x2afff
2022-02-11 14:03:01.566100+0100  localhost kernel[0]:             dependency: as.vit9696.VirtualSMC(1.2.7)[9DAADEEB-D155-3EEC-AFDF-2F05E221EA01]@0x0->0x16fff
2022-02-11 14:03:01.566106+0100  localhost kernel[0]:             dependency: com.alexandred.VoodooI2CServices(1)[AE0A651D-3034-3B31-8175-69D0E253FE2A]@0x0->0xafff
2022-02-11 14:03:01.566112+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOACPIFamily(1.4)[B750EC9D-D224-309C-8C32-3F32C6257F1E]@0xffffff800293f000->0xffffff8002940fff
2022-02-11 14:03:01.566119+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOHIDFamily(2.0.0)[4773BB64-73AE-3283-8DE4-D10459C90231]@0xffffff8002a7d000->0xffffff8002adafff
2022-02-11 14:03:01.566126+0100  localhost kernel[0]:             dependency: com.apple.iokit.IOSMBusFamily(1.1)[FBCF7261-2104-3F79-ADCD-C3DDD5B63986]@0xffffff8002eb0000->0xffffff8002eb1fff
2022-02-11 14:03:01.566146+0100  localhost kernel[0]:             dependency: com.xavier.VoodooSerial(1.0.0)[93D13E56-7627-3D62-ABB8-E23178070992]@0x0->0x3ffff
2022-02-11 14:03:01.566151+0100  localhost kernel[0]:             dependency: org.coolstar.VoodooGPIO(1.2.0)[F3ED6FA0-F8A5-3384-8367-C3D3E1BE802D]@0x0->0x1bfff
2022-02-11 14:03:01.585516+0100  localhost kernel[0]: (Sandbox) <Sandbox`is_release_type> release type: User
```

Xiashangning, 2022-02-11:
Try Big Sur instead then, really don't know why crashes...

therealjbenam, 2022-02-12:
> @therealjbenam So do you see the battery status in Big Sur? The warning can be ignored.

Nope, it doesn't appear... My device shows up as a non-battery powered one. Maybe an ACPI problem, at this point?

Xiashangning, 2022-02-12:
@therealjbenam what is the output of sudo dmesg | grep SurfaceSerialHub ?

therealjbenam, 2022-02-12:
> @therealjbenam what is the output of sudo dmesg | grep SurfaceSerialHub in Big Sur?

The grep Surface includes it since it matches everything with Surface in it.

It's just these few lines

```
Feb 11 17:38:31 localhost kernel[0]: [    3.999433]: SurfaceSerialHubDriver::Found valid UART Bus!
Feb 11 17:38:31 localhost kernel[0]: [    3.999438]: SurfaceSerialHubDriver::Surface Serial Hub found!
Feb 11 17:38:31 localhost kernel[0]: [    5.591032]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
```

Xiashangning, 2022-05-15:
Hi, if you still want to solve this problem, I am back from my school work...
Try this build on Monterey, it might fix the kernel panic. If not, then its probably you config problem since it runs smoothly on my SP7 with the latest Monterey 12.3.1. Anyway, Big Sur should work so please send me the output of 'sudo dmesg | grep SurfaceSerial', if it works in Monterey, you might need the latest version of `DebugEnhancer` to be able to see the early output.

[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8695311/BigSurface.kext.zip)

Xiashangning, 2022-05-17:
Well, this shows that the UART is not driven properly, I will need you to install my custom Ubuntu kernel for Surface and send me the corresponding output, see 

> @therealjbenam I have reinstalled Ubuntu, modified the kernel and compile a workable kernel for you, try this. Using `sudo dpkg -i *.deb` [...]

Xiashangning, 2022-05-17:
Also in macOS `sudo dmesg | grep UART`

fenglinnet, 2022-05-18:
The outputs as follows 
```
[    9.978623] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 153600
[    9.978741] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 9600
[    9.978845] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    9.978932] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2352283657
[    9.979033] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    9.979120] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    9.979185] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    9.979231] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    9.979278] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    9.979415] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 9600
[    9.979543] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 313
[    9.979680] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2352283657
[    9.979823] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    9.979938] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    9.980051] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    9.980168] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    9.980297] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 153600
[    9.980422] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 9600
[    9.980515] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    9.980563] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2352283657
[    9.980624] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    9.980682] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    9.980730] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    9.980778] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    9.980876] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    9.980981] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    9.981085] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    9.981191] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    9.981301] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    9.981408] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    9.981498] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    9.981584] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[    9.981691] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[    9.981788] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[    9.981890] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[    9.981946] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[    9.982009] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[    9.982054] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[    9.982101] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[    9.982166] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[   10.006878] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.007004] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.007108] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.007207] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.007363] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.007495] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[   10.007635] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[   10.007774] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[   10.007941] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.008120] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.008276] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.008410] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.008522] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.008627] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[   10.008725] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[   10.008826] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[   10.009006] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.009143] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.009273] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.009414] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.011004] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.012428] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[   10.013394] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[   10.014347] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[   10.071454] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.072533] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.073817] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.075204] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[... 188 lines cut ...]
[   10.925466] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.925471] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.925474] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.925477] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.925480] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.925483] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[   10.925486] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[   10.925489] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
[   10.925499] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.925501] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.925503] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.925505] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.925508] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.925512] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[... cut ...]
[   10.925554] dw-apb-uart dw-apb-uart.4: ----mydebug uartclk: 48000000
[   10.925555] dw-apb-uart dw-apb-uart.4: ----mydebug baudrate: 3000000
[   10.925557] dw-apb-uart dw-apb-uart.4: ----mydebug quot: 1
[   10.925560] dw-apb-uart dw-apb-uart.4: ----mydebug 0x200: 2147811333
[   10.925562] dw-apb-uart dw-apb-uart.4: ----mydebug 0x204: 7
[   10.925566] dw-apb-uart dw-apb-uart.4: ----mydebug 0x208: 0
[   10.925568] dw-apb-uart dw-apb-uart.4: ----mydebug 0x800: 0
[   10.925571] dw-apb-uart dw-apb-uart.4: ----mydebug 0x804: 0
```

fenglinnet, 2022-05-18:
macos sudo dmesg | grep UART

```
[    3.919845]: SurfaceSerialHubDriver::Found valid UART Bus!
[    5.676158]: VoodooUARTController::UA00 device id: 9d27
[    5.676338]: VoodooUARTController::UA00 Set PCI power state D0
[    5.690526]: VoodooUARTController::UA00 Designware UART version 3.14
[    5.691579]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    6.294366]: SurfaceSerialHubDriver::Failed to get SAM version from UART!
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   35.092494]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   35.096698]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```

therealjbenam, 2022-05-19:
Hi, I haven't forgot about this - it's just that I now have a proper Mac for work so the Surface Pro 6 got reset to factory settings. I will install Big Sur on it, if it will help with the debugging process.

Xiashangning, 2022-05-20:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8740928/BigSurface.kext.zip)
Sorry for the delay, I was busy on another issue.
From your log it seems still UART is not configured properly, yet the clock should right.
Please test this build and send me the following output
`sudo dmesg | grep "UART"` and `sudo dmesg | grep "SurfaceSerial"`

fenglinnet, 2022-05-22:
Sorry for the late reply. It output
```
[    5.617180]: SurfaceSerialHubDriver::Found valid UART Bus!
[    6.696066]: VoodooUARTController::UA00 device id: 9d27
[   10.609001]: VoodooUARTController::UA00 Set PCI power state D0
[   10.609054]: VoodooUARTController::UA00 Designware UART version 3.14
[   10.609061]: VoodooUARTController::UA00 UART capabilities:
[   10.609065]: VoodooUARTController::UA00     ABP_DATA_WIDTH      : 32
[   10.609071]: VoodooUARTController::UA00     AFCE_MODE           : YES
[   10.609075]: VoodooUARTController::UA00     THRE_MODE           : YES
[   10.609079]: VoodooUARTController::UA00     SIR_MODE            : NO
[   10.609084]: VoodooUARTController::UA00     SIR_LP_MODE         : NO
[   10.609088]: VoodooUARTController::UA00     ADDITIONAL_FEATURES : YES
[   10.609092]: VoodooUARTController::UA00     FIFO_ACCESS         : YES
[   10.609096]: VoodooUARTController::UA00     FIFO_STAT           : YES
[   10.609099]: VoodooUARTController::UA00     SHADOW              : YES
[   10.609103]: VoodooUARTController::UA00     ENCODED_PARMS       : YES
[   10.609107]: VoodooUARTController::UA00     DMA_EXTRA           : YES
[   10.609111]: VoodooUARTController::UA00     FIFO_SIZE           : 64
[   10.609330]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[   11.211681]: SurfaceSerialHubDriver::Failed to get SAM version from UART!
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   40.464839]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   40.470027]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```

sudo demsg |grep "SurfaceSerial"
```
[    5.617180]: SurfaceSerialHubDriver::Found valid UART Bus!
[    5.704000]: SurfaceSerialHubDriver::resource_consumer 1
[    5.704010]: SurfaceSerialHubDriver::device_initiated 0
[    5.704014]: SurfaceSerialHubDriver::flow_control 0
[    5.704019]: SurfaceSerialHubDriver::stop_bits 1
[    5.704023]: SurfaceSerialHubDriver::data_bits 3
[    5.704027]: SurfaceSerialHubDriver::big_endian 0
[    5.704031]: SurfaceSerialHubDriver::baudrate 3000000
[    5.704036]: SurfaceSerialHubDriver::rx_fifo 32
[    5.704040]: SurfaceSerialHubDriver::tx_fifo 32
[    5.704044]: SurfaceSerialHubDriver::parity 0
[    5.704049]: SurfaceSerialHubDriver::dtd_enabled 0
[    5.704053]: SurfaceSerialHubDriver::ri_enabled 0
[    5.704057]: SurfaceSerialHubDriver::dsr_enabled 0
[    5.704061]: SurfaceSerialHubDriver::dtr_enabled 0
[    5.704065]: SurfaceSerialHubDriver::cts_enabled 0
[    5.704069]: SurfaceSerialHubDriver::rts_enabled 0
[   10.609330]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[   10.710408]: SurfaceSerialHubDriver::Surface Serial Hub found!
[   11.211666]: SurfaceSerialHubDriver::Timeout waiting for response
[   11.211675]: SurfaceSerialHubDriver::Droping waiting request anyway
[   11.211681]: SurfaceSerialHubDriver::Failed to get SAM version from UART!
```

Xiashangning, 2022-05-22:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8749742/BigSurface.kext.zip)
Same as before

fenglinnet, 2022-05-23:
Perfect!It works well.I'll try detach the display later

Xiashangning, 2022-05-23:
Ok, I would like to see the output to know what should be added and what can be deleted.

fenglinnet, 2022-05-23:
Here,
[uart.log](https://github.com/Xiashangning/BigSurface/files/8751349/uart.log)
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8751346/sserial.log)

Xiashangning, 2022-05-23:
Perfect ! :)

Xiashangning, 2022-05-23:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8751661/BigSurface.kext.zip)
Please test this build, this will be the final version.
And the logs as before

fenglinnet, 2022-05-23:
The last build suddenly stopped working after the restart.I will try the new build

sudo dmesg |grep "SurfaceSerial"
```
[    5.101312]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    5.108385]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
[   15.108827]: SurfaceSerialHubDriver::Could not find UART controller, exiting
```
 sudo dmesg |grep "UART"
```
[    5.101312]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    6.935012]: VoodooUARTController::UA00 device id: 9d27
[    8.110308]: VoodooUARTController::UA00 Set PCI power state D0
[   15.108827]: SurfaceSerialHubDriver::Could not find UART controller, exiting
[   24.054159]: VoodooUARTController::UA00 Designware UART version 3.14
[   24.054181]: VoodooUARTController::UA00 Too bad that we can only use timer
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   35.828410]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   35.833843]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```

fenglinnet, 2022-05-23:
After the new build, the problem is still the same. After rebuilding the Kext cache, it worked fine again.
Because my device can't go into sleep mode, I can't test what happens when it wake up.
sudo dmesg |grep "UART"
```
[    1.596407]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.697170]: VoodooUARTController::UA00 device id: 9d27
[    1.699565]: VoodooUARTController::UA00 Designware UART version 3.14
[    1.699605]: VoodooUARTController::UA00 Too bad that we can only use timer
[    1.700978]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   17.078808]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   17.083961]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```
sudo dmesg |grep "SurfaceSerial"
```
[    1.596407]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.596425]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
[    1.700978]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    1.802013]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    1.804579]: SurfaceSerialHubDriver::Warning, NAK received! Resending all pending messages!
[    1.808109]: SurfaceSerialHubDriver::Warning, no pending command found for seq_id 0
[    1.808160]: SurfaceSerialHubDriver::SAM version 182.2304.139
[    1.808239]: SurfaceSerialHubDriver::Surface Battery nub published!
[    1.917349]: SurfaceSerialHubDriver::data length error!
[    1.917361]: SurfaceSerialHubDriver::msg dump: length: 54
[    1.917368]: SurfaceSerialHubDriver::aa 55 80 7f 00 3f 7d 6a 80 02 00 01 01 27 00 02
[    1.917378]: SurfaceSerialHubDriver::00 00 00 00 00 50 46 00 00 1a 45 00 00 01 00 00
[    1.917387]: SurfaceSerialHubDriver::00 4c 1d 00 00 00 00 00 00 00 00 00 00 11 00 00
[    1.917396]: SurfaceSerialHubDriver::00 e8 03 00 00 ff 00 00 00 00 00 00 00 00 00 00
[    1.922400]: SurfaceSerialHubDriver::data length error!
[    1.922413]: SurfaceSerialHubDriver::msg dump: length: 55
[    1.922419]: SurfaceSerialHubDriver::aa 55 80 7f 00 3f 7d 6a 80 02 00 01 01 27 00 02
[    1.922429]: SurfaceSerialHubDriver::00 00 00 00 00 50 46 00 00 1a 45 00 00 01 00 00
[    1.922439]: SurfaceSerialHubDriver::00 4c 1d 00 00 00 00 00 00 00 00 00 00 11 00 00
[    1.922447]: SurfaceSerialHubDriver::00 e8 03 00 00 ff ff 00 00 00 00 00 00 00 00 00
[    1.925032]: SurfaceSerialHubDriver::Warning, received data with unknown tc 2, cid d
[    2.308968]: SurfaceSerialHubDriver::Timeout waiting for response
[    2.308980]: SurfaceSerialHubDriver::Droping waiting request anyway
[    2.308995]: SurfaceSerialHubDriver::Failed to attach Surface Laptop3 nub! Ignore this if you are not a Laptop3 device
[    2.810229]: SurfaceSerialHubDriver::Timeout waiting for response
[    2.810245]: SurfaceSerialHubDriver::Droping waiting request anyway
[    2.810253]: SurfaceSerialHubDriver::Unexpected response from D0-entry notification
[    2.812687]: SurfaceSerialHubDriver::Warning, NAK received! Resending all pending messages!
[    2.816309]: SurfaceSerialHubDriver::Warning, no pending command found for seq_id 12
[    2.816375]: SurfaceSerialHubDriver::Woke up
[    5.232237]: SurfaceSerialHubDriver::Warning, received data with unknown tc 2, cid 3
```

Xiashangning, 2022-05-23:
> The last build suddenly stopped working after the restart

As you can see from the log, that is because UART driver spends too much time starting. There is nothing to do with Kext cache

Xiashangning, 2022-05-23:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8752014/BigSurface.kext.zip)
Try this one

fenglinnet, 2022-05-23:
The battery icon is not displayed when log in, but it will be displayed after a while
[uart.log](https://github.com/Xiashangning/BigSurface/files/8752118/uart.log)
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8752119/sserial.log)

fenglinnet, 2022-05-23:
The battery was lost after the restart

[uart.log](https://github.com/Xiashangning/BigSurface/files/8752157/uart.log)
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8752158/sserial.log)

Xiashangning, 2022-05-23:
Can you please boot to windows, then clear NVRAM and boot to macOS to get the log again ?
I don't understand why there are so many transmission errors.

fenglinnet, 2022-05-23:
I have reset NVRAM and it still has a lot of output

[uart.log](https://github.com/Xiashangning/BigSurface/files/8753570/uart.log)
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8753571/sserial.log)

Xiashangning, 2022-05-23:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8756818/BigSurface.kext.zip)
Same

fenglinnet, 2022-05-27:
Sorry for the late response.Replace the last build, it not working
sudo dmesg |grep "SurfaceSerial"
```
[    5.108510]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    5.112800]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
[    7.114355]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    7.215425]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    8.217889]: SurfaceSerialHubDriver::SAM version 182.2304.139
[    8.718970]: SurfaceSerialHubDriver::Timeout waiting for response
[    8.718978]: SurfaceSerialHubDriver::Droping waiting request anyway
```
sudo dmesg |grep "UART"
```
[    5.108510]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    7.006280]: VoodooUARTController::UA00 device id: 9d27
[    7.006529]: VoodooUARTController::UA00 interrupt pin: 1
[    7.006539]: VoodooUARTController::UA00 interrupt line: 255
[    7.006567]: VoodooUARTController::UA00 Designware UART version 3.14
[    7.006601]: VoodooUARTController::UA00 Too bad that we can only use timer
[    7.114355]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   31.201017]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   31.230440]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```

Xiashangning, 2022-05-27:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8785466/BigSurface.kext.zip)
same

fenglinnet, 2022-05-27:
still not working

sudo dmesg |grep "SurfaceSerial"
```
[    1.560485]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.560496]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
[    1.665204]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    1.766281]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    2.769487]: SurfaceSerialHubDriver::SAM version 182.2304.139
[    3.269541]: SurfaceSerialHubDriver::Timeout waiting for response
[    3.269556]: SurfaceSerialHubDriver::Droping waiting request anyway
[    3.269564]: SurfaceSerialHubDriver::Unexpected response from D0-entry notification
```
sudo dmesg |grep "UART"
```
[    1.560485]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.664168]: VoodooUARTController::UA00 device id: 9d27
[    1.664335]: VoodooUARTController::UA00 Designware UART version 3.14
[    1.665204]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    1.766441]: VoodooUARTController::UA00 Too bad that we can only use timer
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   15.341144]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   15.347408]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
[ 2796.003072]: VoodooUARTController::UA00 Prepare to sleep
[ 2796.003080]: VoodooUARTController::UA00 Going to sleep
[ 2796.954030]: VoodooUARTController::UA00 Too bad that we can only use timer
[ 2797.954922]: VoodooUARTController::UA00 Woke up
[ 2845.340268]: VoodooUARTController::UA00 Prepare to sleep
[ 2845.341058]: VoodooUARTController::UA00 Going to sleep
[ 2845.506570]: VoodooUARTController::UA00 Too bad that we can only use timer
[ 2846.508884]: VoodooUARTController::UA00 Woke up
[ 2893.488459]: VoodooUARTController::UA00 Prepare to sleep
[ 2893.488473]: VoodooUARTController::UA00 Going to sleep
[ 2893.650968]: VoodooUARTController::UA00 Too bad that we can only use timer
```

Xiashangning, 2022-05-31:
@fenglinnet Sorry for the delay, I was busy these days
I don't really understand why it behaves so worriedly....
With no delay there are a lot of errors, with 100ms delay as you can see the first request (version) completes without any error, but the second request (d0-entry) fails...
Try this three, same as before
[BigSurface no delay.zip](https://github.com/Xiashangning/BigSurface/files/8805186/BigSurface.no.delay.zip)
[BigSurface delay 50ms.zip](https://github.com/Xiashangning/BigSurface/files/8805188/BigSurface.delay.50ms.zip)
[BigSurface delay 100ms.zip](https://github.com/Xiashangning/BigSurface/files/8805189/BigSurface.delay.100ms.zip)

fenglinnet, 2022-06-01:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8752014/BigSurface.kext.zip) Try this one

I've tried all builds and the output is the same. Also, I found that all builds after the above version didn't work.

sudo dmesg |grep "SurfaceSerial"
```
[    1.452058]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.456498]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOSunrisePointLP
[    1.531711]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    1.632738]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    1.736469]: SurfaceSerialHubDriver::SAM version 182.2304.139
[    2.237531]: SurfaceSerialHubDriver::Timeout waiting for response
[    2.237545]: SurfaceSerialHubDriver::Droping waiting request anyway
[    2.237552]: SurfaceSerialHubDriver::Unexpected response from D0-entry notification
```
sudo dmesg |grep "UART"
```
[    1.337978]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    1.417283]: VoodooUARTController::UA00 device id: 9d27
[    1.417536]: VoodooUARTController::UA00 Designware UART version 3.14
[    1.417562]: VoodooUARTController::UA00 Too bad that we can only use timer
[    1.420764]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   14.068271]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   14.315130]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
```

Xiashangning, 2022-06-01:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8812420/BigSurface.kext.zip)

Xiashangning, 2022-06-01:
Please boot into Windows once, then try this build, there is no need to clear Nvram

Xiashangning, 2022-06-01:
Emm, I have rechecked the output that worked before and I have found the issue...
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8815237/BigSurface.kext.zip)
This one will work, definitely.

fenglinnet, 2022-06-02:
It's working again! The output is as follows
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8819794/sserial.log)
[uart.log](https://github.com/Xiashangning/BigSurface/files/8819795/uart.log)

Xiashangning, 2022-06-02:
Can you also please send me `sudo dmesg | grep SurfaceBattery` ?

fenglinnet, 2022-06-02:
I connected to the Surface Hub.The output is as follows
[battery.log](https://github.com/Xiashangning/BigSurface/files/8820398/battery.log)
[uart.log](https://github.com/Xiashangning/BigSurface/files/8820399/uart.log)
[sserial.log](https://github.com/Xiashangning/BigSurface/files/8820400/sserial.log)

[9 more replies]

closed 2022-06-02 by Xiashangning