Xiashangning/BigSurface #22: Internal card reader support on SP7

badstorm, 2022-01-06:
Any ideas on how support the internal card reader?

davidkathy, 2022-01-10:
maybe you can adjust the kext order to fix the issue.
I don't have the surface pro7, but I have one pro6. 
When I reload the kext driver, the tf card was missing after the driver sequence is confused .

sauregurke, 2022-01-14:
macOS historically has had trouble with certain card readers because they were placed on odd communication lines; since the SP7 (at least, all of those I have seen) have the SD card reader connected internally via USB, it should work right out of the box with no kexts or modifications. Please verify that yours is working in Windows and that you have USB ports mapped.

badstorm, 2022-01-21:
Was a wrong usb configuration. Now card reader works otb

closed 2022-01-21 by badstorm

badstorm, 2022-01-21:
I think it's better to call it "MacSurface" instead of "BigSurface" as it now works on other versions of macOS as well. :)

Xiashangning, 2022-01-23:
Got it, maybe in next release :)
And how about your problem, has it been solved? (BigSurface causes kp)

badstorm, 2022-01-23:
With my last release all works, but the KP problem still remain if i try to run macOS from an SD card (with an usb adapter). I think there is something relate to the SD that don't like to macOS.

plsh2me, 2022-01-23:
> I think it's better to call it "MacSurface" instead of "BigSurface" as it now works on other versions of macOS as well. :)

> Got it, maybe in next release :) And how about your problem, has it been solved? (BigSurface causes kp)

Yes, I also agree with @badstorm. That would be appropriate name of the kext, because the kext works on my surface laptop 3 & surface pro 7 (may be on other devices as well) without any problems.
Thank you @Xiashangning for your big effort.