Xiashangning/BigSurface #23: Potential incompatibility with certain Surface Pro 7 units

sauregurke, 2022-01-08:
Hi Ziashangning, I would like to thank you for all of the hard work you have done on this project, in particular with the addition of battery support. However, after running many tests with my config, I can't seem to get anything to work with version 4 of the kext. 

As a matter of fact, on version 3, my type cover, trackpad and ALS were working (see attached files for IOReg and dimes ALS driver readings), although macOS did not seem to respond to the output of the ALS driver. I assumed I had something wrong with my config, and as finals were coming up, I let it be for then.

However, with version 4... _everything seems to be broken_. The ambient light driver doesn't seem to be loaded at all -- "sudo kextstat | grep -i Big" returns nothing, and there is no dmesg logs for BigSurface itself, see attached file -- plus the voodooinput kext is not loaded and therefore the trackpad is not working either. I understand that the voodooinput and voodooI2CController kexts are identical in version 4 and version 3, however VoodooGPIO has been upgraded to version 1.20 from 1.1, plus VoodooSerial has been added. 

In version 4, VoodooI2CController, VoodooSerial and VoodooGPIO are confirmed to be loaded, but VoodooInput does not load or attach, neither does BigSurface itself. 

In version 3, BigSurface and its three plugins are loaded and working. On my machine, the ambient light sensor driver works and puts out values as shown by attached dmesg file, however macOS doesn't respond. As this is less important than battery readouts, we can ignore it for now if you'd like.

TL;DR the following issues are present in Version 4 of BigSurface, on my machine:
- BigSurface itself is not loading - therefore ALS driver is not working;
- VoodooInput is not loading, therefore trackpad is no longer working.

I have attached IORegs from my system for both versions, as well as the output of "sudo dmesg | grep Light" for both versions, named accordingly. Additionally I have included my config with irrelevant parts stripped in case I am missing a kext or something like that - I **do** have SSDT-SURFACE.aml enabled in my config and it is present in my ACPI folder. 

Any help would be greatly appreciated - more likely than not I am forgetting something, in which case I apologize. I have a SP7, i5-1035G4, and a model 1725 type cover.
Please let me know if you need anything else.

[version3grepsurfaceambientlight.txt](https://github.com/Xiashangning/BigSurface/files/7833863/version3grepsurfaceambientlight.txt)
^^this has most of the output stripped away as it is unnecessary. this is just to show that BigSurface version 3 is working, as the ALS driver puts out values (in dmesg).
[Version 3 working trackpad.ioreg.zip](https://github.com/Xiashangning/BigSurface/files/7833862/Version.3.working.trackpad.ioreg.zip)
[version 4 broken trackpad.ioreg.zip](https://github.com/Xiashangning/BigSurface/files/7833868/version.4.broken.trackpad.ioreg.zip)

[version4grepLight.txt](https://github.com/Xiashangning/BigSurface/files/7833866/version4grepLight.txt)
[config.plist.zip](https://github.com/Xiashangning/BigSurface/files/7833882/config.plist.zip)

Darkdestroyer636, 2022-01-10:
noticed this too, ioprobe seems to be set to wrong value, but everything else seems to be working

Darkdestroyer636, 2022-01-10:
update: after upgrading from macOS 12.0.1 to 12.1 it broke

Darkdestroyer636, 2022-01-11:
ok i attempted a big sur downgrade and it did not help

sauregurke, 2022-01-14:
Ah, didn’t think about setting probe value. I thought @Xiashangning had set it to 9999 as this worked with both Big Sur and Monterey. I will confirm.

Darkdestroyer636, 2022-01-20:
Anyone managed to get this working?

Xiashangning, 2022-01-23:
Sorry for the late reply, I was occupied by my university stuffs before (end of the semester) and take several days off to rest, I will investigate this issue now.

sauregurke, 2022-01-24:
No problem! I just finished up my first semester of universiry too so I’m right there with you.

misaka-20002, 2022-04-09:
macOS 12.1 probably modified something(just like Catalina to the BigSur broke the touchpad, VoodooInput not injected that time)

Xiashangning, 2022-05-15:
@sauregurke @KardCreator have you guys solved this problem?
I believe this is a config related issue
I will upload my Opencore config tomorrow so if you still cannot load `BigSurface` you can try mine

sauregurke, 2022-05-18:
Unfortunately not, but I’d be glad to look at your config and see if there’s an issue with mine!

Xiashangning, 2022-05-18:
@sauregurke See my repo https://github.com/Xiashangning/SurfacePro7Opencore

closed 2022-06-25 by Xiashangning