Xiashangning/BigSurface #28: Workout on Surface Touch Screen (not an issue)

plsh2me, 2022-05-25:
Hi dear, First I am thankful to you for your dedication and giving time out of your academic busyness. 
Anyway I am curious to know - when you are planning to work on touchscreen driver?

Thank you

Xiashangning, 2022-06-21:
Touch Screen done :)

plsh2me, 2022-06-21:
I will test this on my devices and report back to you.
Thank you

plsh2me, 2022-06-21:
Hi, it is my mistake I tried to install the kext with the ssdt. The system not even boots (restarts after selecting os). 
Now I found the kext has other requirements. 
To be sure - I have to install first IPTSDaemon and the ssdt, then the kext to get the touch screen working. Am I right?

Xiashangning, 2022-06-21:
No, only the ssd is needed when installing, if you run into anything wrong, that might suggest an incompatibility.
Please attach your screenshot of `Hackintool`'s PCI tab, your touchscreen device location might be different

Xiashangning, 2022-06-21:
Nope, I checked your ioreg, it should work, have you successfully enabled this SSDT? Otherwise there is no way it still panics

plsh2me, 2022-06-21:
Can you please upload your working efi folder here. As my system is not even booting with this kext. 
I also have the same problem in my sp7. May be I have the problem with my config.plist.

Xiashangning, 2022-06-22:
I already have a repo called [SurfacePro7Opencore](https://github.com/Xiashangning/SurfacePro7Opencore)
You can check that

Xiashangning, 2022-06-22:
you need to go into this folder to run the script

plsh2me, 2022-06-22:
I downloaded the folder. Then opened the the folder IPTSDaemon and drag the file install_daemon to the terminal and entered password.
And the output i got in terminal sent to you.
If I am doing wrong, please tell me in details.

Xiashangning, 2022-06-22:
you have to ENTER this folder in terminal
cd path_to_folder
./install_daemon.sh

plsh2me, 2022-06-22:
What I am doing wrong now - 

Last login: Wed Jun 22 11:01:33 on ttys000
mdpalashmiah@MacBook-Air ~ % cd ~/Desktop/IPTSDaemon-main/IPTSDaemon
mdpalashmiah@MacBook-Air IPTSDaemon % ./install_daemon.sh
Password:
cp: IPTSDaemon: No such file or directory
chmod: /usr/local/bin/IPTSDaemon: No such file or directory
chown: /usr/local/bin/IPTSDaemon: No such file or directory
cp: ipts_config: No such file or directory
mdpalashmiah@MacBook-Air IPTSDaemon %

plsh2me, 2022-06-22:
Sorry, You are right I was trying to install it from project folder not from prerelease folder.

Now downloaded the prerelease zip file, run it. Hope now it installed correctly.
Last login: Wed Jun 22 12:27:11 on ttys000
mdpalashmiah@MacBook-Air ~ % cd ~/Desktop/IPTSDaemon
mdpalashmiah@MacBook-Air IPTSDaemon % ./install_daemon.sh
Password:
mdpalashmiah@MacBook-Air IPTSDaemon % 

Am I correct now?

plsh2me, 2022-06-22:
Here is the ioreg of Surface Laptop 3 - 
[ioreg-SL3.zip](https://github.com/Xiashangning/BigSurface/files/8960612/ioreg-SL3.zip)

Xiashangning, 2022-06-23:
Good, closed as solved

closed 2022-06-23 by Xiashangning

Xiashangning, 2022-06-23:
PS: can you test this build for me on laptop3? If everything works fine, then you can switch back to released one
I have changed something so I want to ensure the compatibility for laptop3
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/8964401/BigSurface.kext.zip)

plsh2me, 2022-06-23:
Yes, It works.Last login: Thu Jun 23 09:27:28 on console
mdpalashmiah@Sahils-MBP ~ % sudo dmesg | grep -E "SurfaceSerial|SurfaceLaptop|SurfaceManagement|IntelPrecise"
Password:
[    0.428856]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    0.428894]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.468981]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.469462]: SurfaceManagementEngineDriver::Exit d0i3 mode
[    0.469743]: SurfaceManagementEngineDriver::We need to start the device
[    0.469770]: SurfaceManagementEngineDriver::Hardware is ready
[    0.469785]: SurfaceManagementEngineDriver::Link is established
[    0.470000]: SurfaceManagementEngineDriver::Start response message received
[    0.470124]: SurfaceManagementEngineDriver::Link layer initialization succeeded
[    0.470289]: SurfaceManagementEngineDriver::Capabilities response message received
[    0.470654]: SurfaceManagementEngineDriver::Enumeration response message received
[    0.471101]: SurfaceManagementEngineDriver::Property response message received
[    0.471214]: SurfaceManagementEngineDriver::Starting client...
[    0.479172]: IntelPreciseTouchStylusDriver::IPTS Device is ready
[    0.570059]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.572637]: SurfaceSerialHubDriver::SAM version 14.502.139
[    0.577426]: SurfaceSerialHubDriver::Surface Battery nub published!
[    0.582519]: SurfaceSerialHubDriver::Surface HID nub published!
[    0.687543]: SurfaceSerialHubDriver::Warning, given buffer_len(60) and received data_len(2) mismatched!
[   57.779841]: SurfaceSerialHubDriver::data length error!
[   57.779852]: SurfaceSerialHubDriver::msg dump: length: 46
[   57.779856]: SurfaceSerialHubDriver::aa 55 00 26 00 6a 4a 7d 80 15 00 02 03 15 00 00 
[   57.779863]: SurfaceSerialHubDriver::04 03 9d 06 72 01 00 00 00 00 00 00 00 00 00 00 
[   57.779869]: SurfaceSerialHubDriver::00 00 00 00 00 00 00 00 00 00 f0 1e 01 00 00 00 
[   57.787066]: SurfaceSerialHubDriver::Message received incomplete! Protential data loss!
[   57.787079]: SurfaceSerialHubDriver::msg dump: length: 2
[   57.787083]: SurfaceSerialHubDriver::aa 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
mdpalashmiah@Sahils-MBP ~ %

plsh2me, 2022-06-25:
I am facing random reboot after installing the new v6 kext on my SL3. After reboot I found every time the same the same report.
[panic report.zip](https://github.com/Xiashangning/BigSurface/files/8985598/panic.report.zip)

Xiashangning, 2022-06-26:
I can see nothing from your log, you need to set `keepsyms=1` in bootargs to reveal the exact panic location for me to debug.