Xiashangning/BigSurface #30: Wifi + bluetooth?

SmartBoy84, 2022-06-07:
Is Bluetooth + wifi support planned in this proposition for a fully working surface hackintosh?

Xiashangning, 2022-06-07:
I do not understand your question, intel wifi has already been supported by `AirportItwlm.kext`

closed 2022-06-07 by Xiashangning

Xiashangning, 2022-06-08:
But you should have told me all the information you have at the first place, right? Intel WIFI finally got supported only recently even though this has been an issue years ago. Supporting properly a WIFI device is not simply a dozen of codes. If you spend some time inspecting the linux's wifi driver, you would find that is a huge project and porting it to macOS requires tons of efforts. Given the fact that I do not process a such wifi card, and you wouldn't even want to install a Xcode, I don't see any possibility that it will be supported. :)

MeisterLone, 2023-04-02:
Hi, sorry to open this again. I am trying this on my Surface Laptop 1 with the same result. Everything works except bluetooth and wifi. 

In windows, the wifi shows as a "Marvell" bluetooth and wifi device. Do you know if this is supported by macos or any known kext?

Thank you for this project!