Xiashangning/BigSurface #31: Surface pro 6

cs59, 2022-06-09:
I tried to use this on sp6 and trackpad works fine but in macos settings it says no trackpad found, any support for this surface will be helpful, am also available if you want to test something, you can dm in telegram make support for sp6 through telegram @iOShAT

Xiashangning, 2022-06-21:
Hi, I am back, you really should take look at the solved issues because sp6 is supported in another issue. It is just not release yet.
Try the new release please, sp6 should work without any issue for now.

PS: if you want to make your touchscreen work, please attach a screenshot from hackintool PCI section. I need your touchscreen PCI device id and location information in DSDT.

fenglinnet, 2022-06-22:
I have an SB2, it should be the same device ID (9d3e) as SP6 , and I'm trying to use it to replace 34e4 in kext. The touchscreen is working properly.
 I want to say NB! 👍

Xiashangning, 2022-06-22:
@fenglinnet Okay, if your touchscreen is working, do you mind attach `sudo dmesg | grep -E "SurfaceSerial|SurfaceManagement|IntelPrecise"` here for me?

fenglinnet, 2022-06-23:
@Xiashangning here
[bigsurface.log](https://github.com/Xiashangning/BigSurface/files/8963318/bigsurface.log)

closed 2022-06-25 by Xiashangning