Xiashangning/BigSurface #33: Surface Book 3 - Touchscreen

MetalChris87, 2022-06-21:
Hi, Love your work.

Previous kext works perfectly on my Surface Book 3 bringing keyboard and touchpad plus everything else to my laptop, however when I tried to use the new kext it breaks the keyboard and touchpad, and Touchscreen still does not work.

Thank you

Xiashangning, 2022-06-21:
Wait a minute, you mean last release for surface laptop 3 enables your keyboard & touchpad on surface book 3 ?

Xiashangning, 2022-06-21:
As for the latest build, that is too sad.
Can you attach your ioreg and output of `sudo dmesg | grep -E "SurfaceSerial|SurfaceLaptop|SurfaceManagement|IntelPrecise"` ?

MetalChris87, 2022-06-21:
Yes the last release, version 5 brought keyboard and touchpad to my surface book 3 making it useable as a daily driver. 
I will when i get home to attach the ioreg and output log.

Thanks

MetalChris87, 2022-06-21:
On the Surface Book 3, I have pretty much everything except camera and touchscreen working. Oh and battery drains fast whilst in sleep mode.

SilverTsubasa, 2022-06-21:
> As for the latest build, that is too sad. Can you attach your ioreg and output of `sudo dmesg | grep -E "SurfaceSerial|SurfaceLaptop|SurfaceManagement|IntelPrecise"` ?

On the  Surface pro 7, touch screen not working 
```
voltageshift sudo dmesg | grep -E "SurfaceSerial|SurfaceLaptop|SurfaceManagement|IntelPrecise"
[    0.440315]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    0.441542]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.487758]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.488894]: SurfaceManagementEngineDriver::Exit d0i3 mode
[    0.489075]: SurfaceManagementEngineDriver::We need to start the device
[    0.489091]: SurfaceManagementEngineDriver::Hardware is ready
[    0.489101]: SurfaceManagementEngineDriver::Link is established
[    0.489308]: SurfaceManagementEngineDriver::Start response message received
[    0.489415]: SurfaceManagementEngineDriver::Link layer initialization succeeded
[    0.489532]: SurfaceManagementEngineDriver::Capabilities response message received
[    0.489664]: SurfaceManagementEngineDriver::Enumeration response message received
[    0.489758]: SurfaceManagementEngineDriver::Property response message received
[    0.489773]: SurfaceManagementEngineDriver::Starting client...
[    0.493676]: IntelPreciseTouchStylusDriver::IPTS Device is ready
[    0.589166]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.591736]: SurfaceSerialHubDriver::SAM version 14.502.139
[    0.596292]: SurfaceSerialHubDriver::Surface Battery nub published!
[    0.747338]: SurfaceSerialHubDriver::Timeout waiting for response
[    0.747347]: SurfaceLaptop3Nub::Failed to get data from SSH!
[    0.747350]: SurfaceSerialHubDriver::Failed to attach Surface Laptop3 nub! Ignore this if you are not a Laptop3 device
[    6.534503]: SurfaceManagementEngineDriver::Power gating isolation entry response received
[    6.534626]: SurfaceManagementEngineDriver::Enter d0i3 mode
[   23.714008]: SurfaceManagementEngineDriver::d0i3 set host ready
[   23.714042]: SurfaceManagementEngineDriver::Exit d0i3 mode
[   29.642872]: SurfaceManagementEngineDriver::Power gating isolation entry response received
[   29.642996]: SurfaceManagementEngineDriver::Enter d0i3 mode
[   42.746284]: SurfaceSerialHubDriver::Warning, received data with unknown tc 2, cid 3
[  155.537929]: SurfaceManagementEngineDriver::d0i3 set host ready
[  155.538018]: SurfaceManagementEngineDriver::Exit d0i3 mode
[  161.775347]: SurfaceManagementEngineDriver::Power gating isolation entry response received
[  161.775673]: SurfaceManagementEngineDriver::Enter d0i3 mode
```

Xiashangning, 2022-06-21:
no, it is working
you will need a user space daemon. See my another repo IPTSDaemon
you need to install that and together touchscreen will work

SilverTsubasa, 2022-06-21:
> no, it is working you will need a user space daemon. See my another repo IPTSDaemon you need to install that and together touchscreen will work

I know, i try it but touchscreen not working.
I think it my issues because it also doesn't work when i using voodoo for test touchscreen

Xiashangning, 2022-06-21:
> I think it my issues because it also doesn't work when i using voodoo for test touchscreen

No, voodoo cannot be used here. The only kext you need is BigSurface.kext and IPTSDaemon

SilverTsubasa, 2022-06-22:
> > I think it my issues because it also doesn't work when i using voodoo for test touchscreen
> 
> No, voodoo cannot be used here. The only kext you need is BigSurface.kext and IPTSDaemon
I working know, i forgot `brew install fmt inih`, thanks a lot

Xiashangning, 2022-06-25:
Closed as not responding, assuming solved

closed 2022-06-25 by Xiashangning

MetalChris87, 2022-10-11:
Yes it does. Worked perfectly.