Xiashangning/BigSurface #35: Support 10.14 Mojave on Surface Pro 6

stedio, 2022-06-25:
Hi, I've tried to install the kexts in OC, however somehow the kexts seems not properly loaded. May I ask if these kexts only works on Big Sur? Really would like to have the battery indicator works. Many thanks.

Xiashangning, 2022-06-25:
the lowest supported os version is big sur.
surface don’t have a nvidia card so i see no reason why not upgrade
all functions described in readme work for sp6

Xiashangning, 2022-06-25:
if you insist on mojave, then you can compile this project yourself as well

closed 2022-06-25 by Xiashangning

stedio, 2022-06-25:
Start from 10.15, the system space increased too much and will kill my 128 GB SSD. 

I tried to compile and target for 10.14, however the kext still doesn't load 😂

Xiashangning, 2022-06-26:
It might be that your other kext, especially Lilu and VirtualSMC, you need to get them upgraded to latest

stedio, 2022-06-26:
Now both Lilu and VirtualSMC updated to latest. 

The error msg from OC log: 

`00:000 00:000 OC: Prelinked injection BigSurface.kext\Contents\PlugIns\VoodooInput.kext () - Invalid Parameter
00:392 00:392 OC: Prelinked injection BigSurface.kext\Contents\PlugIns\VoodooI2CServices.kext () - Invalid Parameter
00:769 00:376 OC: Prelinked injection BigSurface.kext\Contents\PlugIns\VoodooGPIO.kext () - Invalid Parameter
01:188 00:419 OC: Prelinked injection BigSurface.kext\Contents\PlugIns\VoodooSerial.kext () - Invalid Parameter
01:689 00:500 OC: Prelinked injection BigSurface.kext () - Invalid Parameter`

Xiashangning, 2022-06-26:
that is because you have to change all kext including these plug-in kexts as well

stedio, 2022-06-26:
Aha, I forgot that. Now the kexts loaded successfully. Thank you very much. 

To enable the battery indicator, is it needed to load SMCBatteryManager, SMCProcessor, SMCLightSensor and SMCSuperIO? 

Currently, I only loaded VirtualSMC, and couldn't see the battery settings in system settings. 

Thank you

Xiashangning, 2022-06-26:
I need logs
`sudo dmesg | grep -E "VoodooUART|SurfaceSerial"`

stedio, 2022-06-26:
Surface-Pro-de-Bin:~ stedio$ sudo dmesg | grep -E "VoodooUART|SurfaceSerial"
Surface-Pro-de-Bin:~ stedio$ 

it seems return nothing?

Xiashangning, 2022-06-26:
you'll need `Debugenhancer.kext`

stedio, 2022-06-26:
thanks, here comes the log

SurfaceSerialHubDriver::Could not parse ACPI resources!
VoodooUARTController::UA00 device id: 9d27
VoodooUARTController::UA00 Designware UART version 3.14
VoodooUARTController::UA00 Too bad that we can only use timer

Xiashangning, 2022-06-26:
Strange, can you attach your dsdt as well?

stedio, 2022-06-26:
Sure, thank you. Please find attached zip file of all the dsdt. 

I used the DSDT from https://github.com/davidkathy/Surface-Pro6-OpenCore, and seems the battery indicator works fine for this author with Big Surface kext. 

[ACPI.zip](https://github.com/Xiashangning/BigSurface/files/8986698/ACPI.zip)

Xiashangning, 2022-06-26:
I need DSDT, not ssdt
Your problem is related to messing up with SSDT & DSDT. For this I cannot help you, as far as I can say, it is a miracle that his config works. You should find another working EFI instead or learn it yourslef.
We have already a lot of working sp6 so the driver `BigSurface` itself is fine.

stedio, 2022-06-26:
Thanks for your reply.
I just tried another SSDT (attached in the zip below). With this one, dmesg retunes nothing now. Is this correct? 

Surface-Pro-de-Bin:~ stedio$ sudo dmesg | grep -E "VoodooUART|SurfaceSerial"
Surface-Pro-de-Bin:~ stedio$ 

[ACPI_v2.zip](https://github.com/Xiashangning/BigSurface/files/8986973/ACPI_v2.zip)

stedio, 2022-06-27:
After double check the compiling, there is one dependency not properly compiled for 10.14. After fixing this, the kext works perfect now in 10.14. Thank you for your help!

Xiashangning, 2022-06-28:
Good :)

Xiashangning, 2022-07-06:
@stedio Hi, sorry to bother you but you've said that your device is Surface Pro 6, and after recompiling the kext yourself, `BigSurface` now works perfectly in 10.14, with all components (battery, type cover, buttons, touch screen) working. Is that right ?

stedio, 2022-07-06:
Hi, I haven't tested the touchscreen, the rest seems working fine. Except the touchpad lost connection after sleeping.

Xiashangning, 2022-07-06:
I have tried to put my SP7 sleeping for a whole night, but my touchpad still works and it can wake up the system. The related code doesn't have anything to do with power management, so I can only say that the problem is not on my driver's side. As for why, well, I have no clue. That is indeed weird.

stedio, 2022-07-06:
Indeed wired. 

But for sleeping, have you monitored the battery drain during sleeping? For me, it seems it is not really 'sleeping', as the battery drain is quite high, if compared with sleeping in windows.