Xiashangning/BigSurface #37: Trackpad stops working after closing lid

stedio, 2022-06-29:
Just noticed the trackpad stops working if the lid is closed for a longer time (10-20 mins), for shorter time, seems no such issue. 

Running 10.14 with recompiled Big Surface 

I guess it related to the power management? Any suggestions? Thank you

Xiashangning, 2022-06-30:
I haven't experienced your issue on my SP7 in big sur or monterey. The Type Cover is connected via USB, thus power managed by USB driver. If there is something wrong, that is your config's or USB driver's issue, not `BigSurface`, as it doesn't manage the power for Type Cover

closed 2022-06-30 by Xiashangning

stedio, 2022-07-01:
But it's weird that the keyboard works totally fine, I guess the keyboard and the touchpad are using the same USB port, as I can only see only 'Surface TypeCover' in the USB tree?