Xiashangning/BigSurface #38: Doesn't work on Surface Laptop Go, randomly causes kernel panics even though SSDT is installed.

unitictv, 2022-06-30:
Hey, I am 99% sure I installed this correctly, I added the SSDT, added all the "sub-kexts" and ordered them correctly. The trackpad still works, gestures don't. Or any of the other added features. It basically just acts like VoodoI2C. I was following [the guide made by olm3ca](https://github.com/olm3ca/Surface-Laptop-Go), and it uses this kext. Yet it doesn't add any of the features. Is there something I am missing in the configuration? Thanks in advance.

Xiashangning, 2022-07-03:
First of all, what is not working when you don't use `BigSurface`? (What do you expect this driver to drive)
I am not clear about the hardware on Surface Laptop Go.

unitictv, 2022-07-03:
Battery percentage, trackpad gestures and sleep.

Xiashangning, 2022-07-03:
I need the kernel panic logs (need boot-args `keepsyms=1`) and your ioreg file.

Xiashangning, 2022-07-03:
BigSurface has nothing to do with sleep issues, that is related to ACPI or Opencore's config
As for trackpad and battery, it should work since you are on intel 10th platform
Send me your ioreg file

unitictv, 2022-07-03:
Here's my IOReg file and the .panic file.
[Logs.zip](https://github.com/Xiashangning/BigSurface/files/9034329/Logs.zip)

Xiashangning, 2022-07-03:
Panic(CPU 0, time 22055348729): NMIPI for spinlock acquisition timeout, spinlock: 0xffffff8016a810e8, spinlock owner: 0xffffff8690a54b20, current_thread: 0xffffff8690a54b20, spinlock_owner_cpu: 0x0
RAX: 0x00000000e7893876, RBX: 0x0000000000000002, RCX: 0xffffff935a0f5b80, RDX: 0x0000000000000009
RSP: 0xffffffa03c542f88, RBP: 0xffffffa03c542fd0, RSI: 0x0000000000000002, RDI: 0xffffff8016840e08
R8:  0x0000000000000003, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0x0000000000000008, R13: 0xffffff8016840e40, R14: 0x0000000000000070, R15: 0xffffffa07ff93320
RFL: 0x0000000000000002, RIP: 0xffffff8015a2f0c0, CS:  0x0000000000000008, SS:  0x0000000000000000
Backtrace (CPU 0), Frame : Return Address
0xffffffa03c542e30 : 0xffffff8015bc272b mach_kernel : _interrupt + 0x12b
0xffffffa03c542e80 : 0xffffff8015a2fd39 mach_kernel : _return_to_iret + 0x159
0xffffffa03c542fd0 : 0xffffff8015a2fbdd mach_kernel : _hndl_allintrs + 0x11d
0xffffffa07ff93400 : 0xffffff8015aea8f8 mach_kernel : _work_interval_port_type_render_server + 0x4fd8
0xffffffa07ff93470 : 0xffffff8015b5c6d3 mach_kernel : _upl_abort_range + 0xac3
0xffffffa07ff93560 : 0xffffff8015d0971e mach_kernel : _cluster_read_ext + 0x14ee
0xffffffa07ff93730 : 0xffffff8015d083ba mach_kernel : _cluster_read_ext + 0x18a
0xffffffa07ff93790 : 0xffffff8018bf0f27 com.apple.filesystems.apfs : _apfs_nstream_read + 0x1be
0xffffffa07ff93840 : 0xffffff8018bf1ec0 com.apple.filesystems.apfs : _apfs_inode_getxattr + 0x1c9
0xffffffa07ff93920 : 0xffffff8018bf1cec com.apple.filesystems.apfs : _apfs_vnop_getxattr + 0xac
0xffffffa07ff93970 : 0xffffff8015d5652a mach_kernel : _VNOP_GETXATTR + 0x5a
0xffffffa07ff939e0 : 0xffffff8016f320e9 com.apple.AppleFSCompression.AppleFSCompressionTypeZlib : _compression_decode_buffer + 0xd02
0xffffffa07ff93a40 : 0xffffff8016f32447 com.apple.AppleFSCompression.AppleFSCompressionTypeZlib : _compression_decode_buffer + 0x1060
0xffffffa07ff93b00 : 0xffffff8016f31d81 com.apple.AppleFSCompression.AppleFSCompressionTypeZlib : _compression_decode_buffer + 0x99a
0xffffffa07ff93b90 : 0xffffff8015d6aa87 mach_kernel : _decmpfs_read_compressed + 0x677
0xffffffa07ff93d00 : 0xffffff8018bfaf8b com.apple.filesystems.apfs : _apfs_vnop_read + 0x2d9
0xffffffa07ff93d90 : 0xffffff8015d43dcc mach_kernel : _utf8_normalizeOptCaseFoldAndMatchSubstring + 0x72c
0xffffffa07ff93e30 : 0xffffff801604c038 mach_kernel : _read_nocancel + 0x328
0xffffffa07ff93ee0 : 0xffffff801604c19d mach_kernel : _pread_nocancel + 0xbd
0xffffffa07ff93f40 : 0xffffff801613e9de mach_kernel : _unix_syscall64 + 0x2ce
0xffffffa07ff93fa0 : 0xffffff8015a301f6 mach_kernel : _hndl_unix_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.filesystems.apfs(1677.141.3)[45812160-E4A7-307A-B604-91D6A782126B]@0xffffff8018baa000->0xffffff8018d19fff
            dependency: com.apple.driver.AppleEFINVRAM(2.1)[A506788E-BC7C-3C87-BD70-376984C225D3]@0xffffff8016efe000->0xffffff8016f07fff
            dependency: com.apple.driver.AppleEffaceableStorage(1.0)[F1C24692-B3A0-3F31-A777-CA17069CD3DD]@0xffffff8016f11000->0xffffff8016f16fff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[E6C5D15B-3934-3236-932A-1EEF294F5E2B]@0xffffff801736d000->0xffffff8017373fff
            dependency: com.apple.iokit.IOStorageFamily(2.1)[FFF27081-0E9D-3E7F-8807-C018C8E46A09]@0xffffff801866e000->0xffffff801867ffff
            dependency: com.apple.kec.corecrypto(11.1)[17F7A252-0406-3D10-90BD-A51F4394481D]@0xffffff8018d47000->0xffffff8018dd8fff
            dependency: com.apple.security.AppleImage4(3.0.0)[BE766F1D-D061-3436-98BA-AF65AD47F2EF]@0xffffff8016f7d000->0xffffff8016f8dfff
         com.apple.AppleFSCompression.AppleFSCompressionTypeZlib(1.0)[B0DACBCF-5CFF-3C57-9743-5A8A7EE73627]@0xffffff8016f26000->0xffffff8016f36fff

Process name corresponding to current thread: MRT
Boot args: -igfxdvmt -igfxcdc keepsyms=1 debug=0x100 alcid=1 swd_panic=1 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20G527

Kernel version:
Darwin Kernel Version 20.6.0: Tue Feb 22 21:10:41 PST 2022; root:xnu-7195.141.26~1\/RELEASE_X86_64
Kernel UUID: 40A47F9A-078F-3F9B-B353-53A194CC6F1D
KernelCache slide: 0x0000000015800000
KernelCache base:  0xffffff8015a00000
Kernel slide:      0x0000000015810000
Kernel text base:  0xffffff8015a10000
__HIB  text base: 0xffffff8015900000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 22057794838
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000000522bf4ddf
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000003caa28f70 0x0000000000000000
panic(cpu 4 caller 0x<ptr>): \"Attempt to test no-preempt spinlock 0xffffff8016a810e8 in preemptible context\"@\/System\/Volumes\/Data\/SWE\/macOS\/BuildRoots\/a0c6c82cc8\/Library\/Caches\/com.apple.xbs\/Sources\/xnu\/xnu-7195.141.26\/osfmk\/kern\/locks.c:732
Backtrace (CPU 4), Frame : Return Address
0xffffffa07eb13690 : 0xffffff8015a8b26d mach_kernel : _handle_debugger_trap + 0x3fd
0xffffffa07eb136e0 : 0xffffff8015bd2993 mach_kernel : _kdp_i386_trap + 0x143
0xffffffa07eb13720 : 0xffffff8015bc2f8a mach_kernel : _kernel_trap + 0x55a
0xffffffa07eb13770 : 0xffffff8015a2fa2f mach_kernel : _return_from_trap + 0xff
0xffffffa07eb13790 : 0xffffff8015a8aa8d mach_kernel : _DebuggerTrapWithState + 0xad
[... cut ...]
            dependency: com.apple.iokit.IOStorageFamily(2.1)[FFF27081-0E9D-3E7F-8807-C018C8E46A09]@0xffffff801866e000->0xffffff801867ffff
            dependency: com.apple.kec.corecrypto(11.1)[17F7A252-0406-3D10-90BD-A51F4394481D]@0xffffff8018d47000->0xffffff8018dd8fff
            dependency: com.apple.security.AppleImage4(3.0.0)[BE766F1D-D061-3436-98BA-AF65AD47F2EF]@0xffffff8016f7d000->0xffffff8016f8dfff

Process name corresponding to current thread: MRT

This is your extracted kp log, as you can see, it has nothing to do with `BigSurface`, this is your own installation's problem

unitictv, 2022-07-03:
OK, I can resolve that, how about the other issue?

Xiashangning, 2022-07-03:
Add `DebugEnhancer.kext` and send me output of `sudo dmesg | grep -E "SurfaceSerial|VoodooUART"`

Xiashangning, 2022-07-03:
And of course use the latest version of BigSurface

unitictv, 2022-07-03:
Here's the output:
```
[    0.646259]: VoodooUARTController::UA00 device id: 34a8
```

Xiashangning, 2022-07-03:
Impossible, do you have BigSurface.kext enabled in config.plist?

unitictv, 2022-07-03:
[image](https://user-images.githubusercontent.com/56695928/177033648-9be77270-bf92-4142-9ace-d024e08e1eaf.png)
All of them.

unitictv, 2022-07-03:
The SSDT too.
[image](https://user-images.githubusercontent.com/56695928/177033723-c39e5b03-dce1-4b5b-9954-6aa8832ae737.png)

Xiashangning, 2022-07-03:
Does your BigSurface version up to date?

unitictv, 2022-07-03:
Latest one, 6.0.

Xiashangning, 2022-07-03:
Try again with this `sudo dmesg | grep -E "SurfaceSerial|VoodooUART"`

unitictv, 2022-07-03:
Still `[    0.646259]: VoodooUARTController::UA00 device id: 34a8`

Xiashangning, 2022-07-03:
Send me your `BigSurface.kext`

unitictv, 2022-07-03:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034448/BigSurface.kext.zip)

Xiashangning, 2022-07-03:
What about the new ioreg?

unitictv, 2022-07-03:
[CMacBook Air.zip](https://github.com/Xiashangning/BigSurface/files/9034460/CMacBook.Air.zip)

Xiashangning, 2022-07-03:
There is definitely something wrong on your config.
There must have a log like this `VoodooUARTController::UA00 Designware UART version...`. This line is written in the start function, so if VoodooUART shows up in ioreg that means start function returns no error. Thus this line will be executed.

unitictv, 2022-07-03:
I am so sorry, I though the UART id was the only relevant piece of information, here's the rest.
```
[    0.646491]: VoodooUARTController::UA00 Designware UART version 3.14
[    0.646497]: VoodooUARTController::UA00 Too bad that we can only use timer
```

Xiashangning, 2022-07-03:
...

unitictv, 2022-07-03:
I am so sorry for the inconvenience, I don't know why I thought that.

Xiashangning, 2022-07-04:
Use this one and send me all output of `sudo dmesg | grep SurfaceSerial`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9037791/BigSurface.kext.zip)

unitictv, 2022-07-04:
`[    0.611936]: Couldn't alloc class "SurfaceSerialHubDriver"`

Xiashangning, 2022-07-04:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9039333/BigSurface.kext.zip)

unitictv, 2022-07-04:
Unfortunately still `[    1.272416]: Couldn't alloc class "SurfaceSerialHubDriver"

Xiashangning, 2022-07-06:
I need full log of `sudo dmesg` then
But be aware that the possibility of resolving this by me is nearly 0. I am afraid that it falls into your "1% case of wrong installment". 
This is more likely a OC config problem to me, since this type of error only happens when macOS cannot find the class --"SurfaceSerialHubDriver" described in Info.plist of a kext -- `BigSurface.kext` in this case.
However, this class definitely exists, because the same kext works on all other devices. at least not with this problem.

unitictv, 2022-07-07:
Sorry for the late reply, [here](https://github.com/Xiashangning/BigSurface/files/9064052/sudo.dmesg.txt) is the full `sudo dmesg` output log.

Xiashangning, 2022-07-07:
There contains no `SurfaceSerialHubDriver` in your log...

Xiashangning, 2022-07-08:
Well, now 100% sure it is your own problem, see issue #40 for a "normal" output of Surface Laptop Go. Battery is proved to be working, I don't know what happened to your device.

closed 2022-07-08 by Xiashangning

IsaahE, 2022-11-13:
It's not working for me either in this exact way. Installed exactly how it says and quadruple checked everything. I have the SSDT and Kext installed and the kexts are configured in the right order.