Xiashangning/BigSurface #39: Surface Pro 6 latest kext don't works

cs59, 2022-07-03:
I have surface pro 6 with macos monterey latest installed on it. before i was using your old kext which u made for someone issue before if i remembered and tried the kext and it working fine but afcourse no trackpad found in settings but trackpad and all gestures work and also hardware power volume keys, other fearures like battery afcourse not work, and I tried your ssdt and that latest bigsurface kext and when i booted with your latest kext then it not booting, If you have time to make this kext fully or mostly support sp6 then am ready for all information u want also I can install linux and send what u want for this kext to make work for sp6, hope u reply I will wait
[image]
g

cs59, 2022-07-03:
Current bigsurface kext am using is this -
[BigSurfaceSP6.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034163/BigSurfaceSP6.kext.zip)

Xiashangning, 2022-07-03:
I need ioreg, disable BIgSurface and send me the ioreg

cs59, 2022-07-03:
> I need ioreg, disable BIgSurface and send me the ioreg

okay, should i remove the bigsurface fully then send ioreg?

Xiashangning, 2022-07-03:
just not enable it in config.plist

cs59, 2022-07-03:
> just not enable it in config.plist

okay

cs59, 2022-07-03:
> just not enable it in config.plist
Opened ioreg which information should i send u?
[image]

Xiashangning, 2022-07-03:
save this

cs59, 2022-07-03:
> save this

Ok sending

Xiashangning, 2022-07-03:
???
You HAVE TO SAVE THIS TO A FILE AND SEND THAT FILE, in `File`->`Save`
you attached NOTHING

cs59, 2022-07-03:
> ???
> You HAVE TO SAVE THIS TO A FILE AND SEND THAT FILE, in `File`->`Save`
> you attached NOTHING
`

> ???
> You HAVE TO SAVE THIS TO A FILE AND SEND THAT FILE, in `File`->`Save`
> you attached NOTHING

Here’s link: can’t attack files from device i think so here link https://drive.google.com/file/d/1AUv9wy4scFcC90Uf4tsdzh6txAWTqOmQ/view?usp=drivesdk

Xiashangning, 2022-07-03:
Why? compress it to a zip file and drag it here
I cannot open this link

cs59, 2022-07-03:
> Why? compress it to a zip file and drag it here
> I cannot open this link

Ok wait

cs59, 2022-07-03:
> Why? compress it to a zip file and drag it here
> I cannot open this link
here file Korey
[LJSS’s MacBook Pro.zip](https://github.com/Xiashangning/BigSurface/files/9034228/LJSS.s.MacBook.Pro.zip)

Xiashangning, 2022-07-03:
Try this one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034240/BigSurface.kext.zip)

cs59, 2022-07-03:
> Try this one, remember to reenable BigSurface
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034240/BigSurface.kext.zip)

Ok, just replace the kext with this one in kext folder and enable it, ryt?

cs59, 2022-07-03:
> Try this one, remember to reenable BigSurface
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034240/BigSurface.kext.zip)

Stuck here
[image]

cs59, 2022-07-03:
> Try this one, remember to reenable BigSurface [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034240/BigSurface.kext.zip)

Stuck her
[image]
e

Xiashangning, 2022-07-03:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034256/BigSurface.kext.zip)

cs59, 2022-07-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034256/BigSurface.kext.zip)

Different error last output every boo
[image]
[image]
t

cs59, 2022-07-03:
> > [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034256/BigSurface.kext.zip)
> 
> Different error last output every boot [image] [image] t

btw it gone past before 1 error 'previous shutdown error'

Xiashangning, 2022-07-03:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034271/BigSurface.kext.zip)

cs59, 2022-07-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034271/BigSurface.kext.zip)

this one working bro, trackpad and gestures work but as usual setting don't find trackpad, power volume button working. can we make more compatibility for sp6
[image]
?

Xiashangning, 2022-07-03:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034289/BigSurface.kext.zip)
If this one works, then go download `IPTSDaemon` from my other repo, install it as instructed and test if touch screen works

cs59, 2022-07-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034289/BigSurface.kext.zip)
> If this one works, then go download `IPTSDaemon` from my other repo, install it as instructed and test if touch screen works

Okay

cs59, 2022-07-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034289/BigSurface.kext.zip)
> If this one works, then go download `IPTSDaemon` from my other repo, install it as instructed and test if touch screen works

Can’t install bro getting errors
[image]

cs59, 2022-07-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034289/BigSurface.kext.zip)
> If this one works, then go download `IPTSDaemon` from my other repo, install it as instructed and test if touch screen works

Seems like it’s not copying the file dunno why, let me try disabling the sip
[image]

cs59, 2022-07-03:
> > [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9034289/BigSurface.kext.zip)
> > If this one works, then go download `IPTSDaemon` from my other repo, install it as instructed and test if touch screen works
> 
> Seems like it’s not copying the file dunno why, let me try disabling the sip, else will copy the files manually and see [image]

Okay solved the error, just have to -R
[image]

Xiashangning, 2022-07-03:
I don't understand what you are doing.
Really, a kind reminder: do you install `BigSurface.kext` like this?

cs59, 2022-07-03:
> I don't understand what you are doing.
> Really, a kind reminder: do you install `BigSurface.kext` like this?

No bro, bigsurface i replaced in windows from diskgenius and working perfectly fine, and for IPTSDaemon, when i run install sh file then it shows many errors, so what i done is that i opened sh file and manually copy all commands one by one and accoring to their location i cd in terminal and then pasted commands one by one and installed like that

cs59, 2022-07-03:
> I don't understand what you are doing.
> Really, a kind reminder: do you install `BigSurface.kext` like this?

Also see this, i install the daemon successfully, also it doesn’t execute the last command if not run with sudo, if i run with sudo then it works see
[image]

Xiashangning, 2022-07-03:
> Okay solved the error, just have to -R

Also, that is not SOLVED, that is WORSE
Why would you install anything from the project folder ???????
Here is what you will do:
You go to the `release section` of IPTSDaemon repo main page
You download the pre-release zip file
You unzip the file you downloaded
You open Terminal
You cd to that folder
You run the install script

cs59, 2022-07-03:
> Also, that is not SOLVED, that is WORSE Why would you install anything from the project folder ??????? Here is what you will do: You go to the `release section` of IPTSDaemon repo main page You [...]

Extremely sorry bro, i tried to see release but I thought we have to so it manually, release name really too small btw sorry 
[image]

cs59, 2022-07-03:
> Extremely sorry bro, i tried to see release but I thought we have to so it manually, release name really too small btw sorry [image]

Okay all good now, I will follow next 
[image]

cs59, 2022-07-03:
> Also, that is not SOLVED, that is WORSE Why would you install anything from the project folder ??????? Here is what you will do: You go to the `release section` of IPTSDaemon repo main page You [...]

All good now, will follow next things
[image]

Xiashangning, 2022-07-03:
No, not good> You do not need `sudo`, rerun the script

Xiashangning, 2022-07-03:
If everything works as expected, your touch screen should work

cs59, 2022-07-03:
> If everything works as expected, your touch screen should work

Okay, rerun the script and no errors at all, but touchscreen not working as expected lol 
[image]

cs59, 2022-07-03:
> > If everything works as expected, your touch screen should work
> 
> Okay, rerun the script and no errors at all, but touchscreen not working as expected lol [image]

Also bro, I tried to check if the daemon starts after restarting my laptop automatically but the daemon not started as i checked it by running last command of sh script after restrating, it only enabled daemon when we run that command

cs59, 2022-07-03:
> Also bro, I tried to check if the daemon starts after restarting my laptop automatically but the daemon not started as i checked it by running last command of sh script after restrating, it only [...]

[image]

Xiashangning, 2022-07-03:
run `IPTSDaemon` manually and see what is the error

cs59, 2022-07-03:
> run `IPTSDaemon` manually and see what is the error

Here,
[image]

Xiashangning, 2022-07-03:
`sudo dmesg | grep -E "SurfaceMana|IntelPrecise"`

cs59, 2022-07-03:
> sudo dmesg | grep -E "SurfaceMana|IntelPrecise"

Runnes no error comes, now?

cs59, 2022-07-03:
> sudo dmesg | grep -E "SurfaceMana|IntelPrecise"

Same, Could not find IntelPreciseTouchusDriver when trying to run IPTSDaemon

Xiashangning, 2022-07-03:
Then you should add `DebugEnhancer.kext` and reboot.
I need output of this

cs59, 2022-07-03:
> Then you should add `DebugEnhancer.kext` and reboot.
> I need output of this

Okay, will get this kext and will reboot, and will message you

cs59, 2022-07-03:
> Then you should add `DebugEnhancer.kext` and reboot.
> I need output of this

Ok, added debugenchancer kext and now booted, now what to do next to get output?

Xiashangning, 2022-07-03:
Go to Terminal and send me output of `sudo dmesg | grep -E "SurfaceMana|IntelPrecise"`

cs59, 2022-07-03:
> Open Terminal and send me output of `sudo dmesg | grep -E "SurfaceMana|IntelPrecise"`

Sadly it’s not showing any output, you see can here, debug kext also enabled
[image]
[image]

cs59, 2022-07-03:
> sudo dmesg | grep -E "SurfaceMana|IntelPrecise"

I seen on DebugEnchancer website that workaround is going on macos 12 and above, that why it maybe not showing any output

cs59, 2022-07-04:
> Try this one and send me `sudo dmesg | grep -E "SurfaceMana|IntelPrecise"`
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9037894/BigSurface.kex
[image]
t.zip)

okay finally got output

cs59, 2022-07-04:
> > Try this one and send me `sudo dmesg | grep -E "SurfaceMana|IntelPrecise"`
> > [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9037894/BigSurface.kex
> 
> [image]
> 
> okay finally got output

what the heeeeeckkkkkk, touchscreen perfectly man lol am crying bro thnx so much lol never expected specially touchscreen seriously u r very genius bro u makes impossible to possible, I hope we can now fix more things and increase compatibility for SP6 :)

Xiashangning, 2022-07-04:
Try this one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9038339/BigSurface.kext.zip)

cs59, 2022-07-04:
> Try this one
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9038339/BigSurface.kext.zip)

Stuck here
[image]
[image]

Xiashangning, 2022-07-04:
I need your dsdt

cs59, 2022-07-04:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9039388/BigSurface.kext.zip)

worked but any changes?
Also am getting random clicks after my touchscreen was started working

https://user-images.githubusercontent.com/68501788/177156672-c8a7701d-fed0-452b-9899-bf94471fbf5d.MOV

Xiashangning, 2022-07-04:
i need logs
sudo dmesg | grep SurfaceSerial

cs59, 2022-07-04:
> i need logs
> sudo dmesg | grep SurfaceSerial

[image]

cs59, 2022-07-04:
> i need logs
> sudo dmesg | grep SurfaceSerial

Update: the random clicking problem was seems solved now :),
Noe possible to get this last thing ‘battery’?

cs59, 2022-07-05:
> > i need logs
> > sudo dmesg | grep SurfaceSerial
> 
> Update: the random clicking problem was seems solved now :), Now possible to get this last thing ‘battery’?

Okay it’s not fixed, even it’s getting worst now, any command to disable it and if fix also available?

[12 more replies]

closed 2022-07-06 by Xiashangning