Xiashangning/BigSurface #41: Nothing works on surface pro 5

kevin-Curry, 2022-07-14:
I’ve tested everything. But it seems nothing works on my laptop. Including battery and trackpad. Stared and waiting for touchscreen for surface pro 5!

Xiashangning, 2022-07-16:
Trackpad should be working, as for battery and touchscreen, I need your device id for UART and IntelManagementEngine. You can either send me your screenshot of `hackintool`'s PCIe tab or find them in device manager in Windows.

kevin-Curry, 2022-07-17:
OK I'll try

kevin-Curry, 2022-07-19:
[image]
there's the screenshot. look for the touchscreen and battery

Xiashangning, 2022-07-19:
I need ioreg, try the pre-release kext first, battery and touchscreen should work

kevin-Curry, 2022-07-19:
only v5 works

kevin-Curry, 2022-07-19:
v6 and pre-release doesn't work

Xiashangning, 2022-07-19:
What is not working? Attach output of `sudo dmesg | grep -E "SurfaceSerial|VoodooUART|SurfaceBattery|SurfaceMana"`

kevin-Curry, 2022-07-19:
And ioreg/Users/kevinzhou/Library/Messages/Attachments/ea/10/B1A5E6BE-8B79-4E3F-99B8-3C6A6213B8FF/图像2022-7-19 上午10.59.heic

kevin-Curry, 2022-07-19:
Nothing is working and no output after I type that.

Xiashangning, 2022-07-19:
you need debugenhancer.kext

kevin-Curry, 2022-07-19:
where can I download

kevin-Curry, 2022-07-19:
hmmm.
None of your release can drive the trackpad.
I downloaded a version that edited by other one.
I don't know which version actually did he use.
My OpenCore version is 0.8.1 and macOS Monterey.
I don't know if it made something wrong.

kevin-Curry, 2022-07-19:
Sorry.
Version 3.0 does work.
But 1/2/4/5/6/6.1 doesn't work.

Xiashangning, 2022-07-19:
I need you to install DebugEnhancer.kext (just search it on github) and attach the output and the ioreg saved from ioregistryexplorer.app
It is meaningless to try different version. That is a waste of time

kevin-Curry, 2022-07-19:
[image]
[image]
Is these what you want?

Xiashangning, 2022-07-19:
Have you updated your Lilu and VirtualSMC to the latest version?

Xiashangning, 2022-07-19:
What about the output? `sudo dmesg | grep -E "SurfaceSerial|VoodooUART|SurfaceBattery|SurfaceMana" -A 5 -B 5`

Xiashangning, 2022-07-19:
Try this one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9141941/BigSurface.kext.zip)

kevin-Curry, 2022-07-20:
Still not working.

Xiashangning, 2022-07-20:
I NEED THE OUTPUT, PLZ
How would you expect me to help you debug without telling me what is wrong?

kevin-Curry, 2022-07-20:
[image]

Xiashangning, 2022-07-20:
sudo dmesg | grep -E "SurfaceSSurfaceB|SurfaceM" -A 10

kevin-Curry, 2022-07-20:
[image]

Xiashangning, 2022-07-20:
sudo dmesg | grep -E "SurfaceS|SurfaceB|SurfaceM" -A 10
sorry

kevin-Curry, 2022-07-20:
[image]

Xiashangning, 2022-07-20:
you should remove smcbatterymanager

kevin-Curry, 2022-07-20:
Yeah, GREAT.
Trackpad and Battery works.
But touchscreen still not working.

kevin-Curry, 2022-07-20:
What command shall I type this time?

Xiashangning, 2022-07-20:
you should install IPTSDaemon, see my other repo

closed 2022-07-20 by Xiashangning

kevin-Curry, 2022-07-20:
OK.
 IT WORKS!!!
 AWESOME!