Xiashangning/BigSurface #44: Surface Book 3 over heating

MetalChris87, 2022-07-20:
Hi,

I had everything working on Monterey except for bluetooth as the BigSurface kext won't work when bluetooth kext is also installed, no idea, but I am fine not having Bluetooth.

My issue is the screen part of my Surface Book 3 where the CPU etc all sits is over heating. It doesn't happen every time, I can be using for hours with no issues everything great including touch screen, and then it will over heat for no reason. Or I can boot up after not being turned on for hours and it will be over heat immediately. And sometimes when it's asleep I will touch the lid and I can feel it extremely hot get worried and turn it off for a few hours.  I use my laptop for work and so I had to revert back to windows for now. 
Any ideas? I was thinking maybe performance mode is set too high? not sure.

BTW I also get this same over heating issue with BigSur.

Thanks

kevin-Curry, 2022-07-21:
Wait.
You meant book 3 can touch right

MetalChris87, 2022-07-21:
Surface Book 3 has working touch screen. Yes.

Xiashangning, 2022-07-21:
BigSurface has nothing to do with buletooth at all
As for your overheating problem, plz first make sure that it is CAUSED by BigSurface. For example, if you disable BigSurface, will the overheating be gone?

MetalChris87, 2022-07-21:
I know, I was just trying to convey that everything is working great except for bluetooth as for some reason it clashes with the BigSurface Kext. I can only have one active, but that doesn't bother me. 
I will boot up MacOs Monterey tonight and test without BigSurface to see how it reacts. 

Thanks

MetalChris87, 2022-07-21:
I disabled BigSurface and the overheating issue remained. I edited the Plist for BigSurface to change the performance mode from 1 to 0 and immediately after reboot the fans turned on rapidly. I will try different values to see which works best for me as I feel it has slowed down my laptop a little bit.

Xiashangning, 2022-07-22:
Interesting, so actually the `best performance` mode works best

closed 2022-07-22 by Xiashangning

kevin-Curry, 2022-07-24:
> Surface Book 3 has working touch screen. Yes.

Would you please send me a link of you EFI folder plz.

MetalChris87, 2022-07-24:
My Efi for Monterey is here 
https://drive.google.com/file/d/1pmqYzsL6TwmGODvsHqhrRWfjYjxlRSEy/view?usp=sharing

But I think the Efi in here is version 5. Download version 6 and follow the steps on this GitHub for touchscreen functionality.

kevin-Curry, 2022-07-25:
> My Efi for Monterey is here https://drive.google.com/file/d/1pmqYzsL6TwmGODvsHqhrRWfjYjxlRSEy/view?usp=sharing
> 
> But I think the Efi in here is version 5. Download version 6 and follow the steps on this GitHub for touchscreen functionality.

wait, 15''or13.5''

MetalChris87, 2022-07-25:
My laptop is 15inch.