Xiashangning/BigSurface #47: SP6 Monterey - ALS Auto Brightness

EvoSems, 2022-07-28:
Hello Friend,

Firstly, Thank you for all your work! Simply Amazing!

My Only trouble so Far is getting ALS to Auto Dim etc

What other Kext/SSDT are Required or not Required?
ALS0, PNLF, SMCLightSensor?

I can manually change Brightness with Slider as well as Keyboard controls.
Automatically Adjust Brightness is Checked.

I feel I am missing something very simple...

I am a little green on the scene, but am woking on a Ventura Voyage

Many Thanks

E

Xiashangning, 2022-07-28:
I need your dsdt and ioreg

Xiashangning, 2022-07-28:
> ALS0, PNLF, SMCLightSensor?
None of these are needed as they are not written for surface devices

EvoSems, 2022-07-29:
So to be clear. I only need the following?
[image]

Xiashangning, 2022-07-29:
Sorry, PNLF is still needed.
VirtualSMC is missing, when I say `SMCLightSensor` I mean only this plugin, the main kext `VirtualSMC`  is indispensable.
PS: I doubt that you lack serval kexts as well, like wifi & buletooth etc

Xiashangning, 2022-07-29:
Go to Windows' device manager to find out where your Ambient Light Sensor is attached to. (I cannot find it in your DSDT)

EvoSems, 2022-07-30:
Hi Again. thanks for the info. I have now set up correctly!

Brightness Control works manually now. (auto is checked) but doesn't function.

Windows im not sure how to find in device manager? I had a good look. but I do know it 100% change automatically in windows so there is 1 sensor present 100%.

Good news is I have Ventura installed and working. the only change needed is OC 0.8.3 and Lilu.kext 1.6.2

Xiashangning, 2022-07-30:
I tried to search for SP6's light sensor hardware info, it seems to be different from SP7 (apds9960) so that would require a complete new driver for your ambient light sensor to work. I am afraid that I cannot do this for you. You could only depend on yourself.

closed 2022-07-30 by Xiashangning

EvoSems, 2022-07-30:
Thankyou. how did you find apds9960 in windows? I will try to find and fix it.

many thanks for all your hard work

cs59, 2022-07-31:
> Thankyou. how did you find apds9960 in windows? I will try to find and fix it.
> 
> many thanks for all your hard work

hey, I also own a SP6 device and have bigsur installed, and I did like to test SP6 EFI if u can give it to me