Xiashangning/BigSurface #48: MacOs Ventura und BigSurface shutdown

Peterro61, 2022-07-28:
Hello.
I installed Ventura and it works for now. What doesn't work is turning it off normally. After switching off, the computer restarts with the message "Rebooted because an error was detected" After that, the entire top bar is missing. Turning it off by holding down the power button and then turning it back on works normally. If I load the Bigsurface.kext before voodoinput.kext everything works except the battery indicator is gone

Peterro61, 2022-07-28:
Sorry, it is a SP7

Xiashangning, 2022-07-28:
The support for Ventura is not planned as the released version is not coming out yet. Sorry for that.
I will leave this open until I upgrade to Ventura and take a look at this bug.

SheltangW, 2022-08-06:
Hey dude, thanks for your art, and does Bigsurface also work on Monterey? Or just on Big sur? Thans!

Xiashangning, 2022-08-06:
@SheltangW Why not? Officially supported on 10.15-12

SheltangW, 2022-10-19:
> @SheltangW Why not? Officially supported on 10.15-12

Thanks man, it works

backdash756, 2022-10-28:
Maybe I can help a bit here, there is a kernel panic when shutting down Ventura:

`panic(cpu 2 caller 0xffffff801c34aee3): Kernel trap at 0xffffff801c24ab71, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000042, CR3: 0x0000000023006000, CR4: 0x00000000003626e0
RAX: 0xffffff8b7687b238, RBX: 0xffffff86ab262c00, RCX: 0x0000000000000001, RDX: 0x0000000000000001
RSP: 0xffffffe281bc7dd0, RBP: 0xffffffe281bc7e00, RSI: 0xffffff95102b3100, RDI: 0x0000000000000002
R8:  0x000000000000ffff, R9:  0x00000000ffffffff, R10: 0x0000000000000001, R11: 0xffffff8b785f8550
R12: 0x0000000000000000, R13: 0xffffff951067dc70, R14: 0x0000000000000001, R15: 0x0000000000000002
RFL: 0x0000000000010202, RIP: 0xffffff801c24ab71, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000042, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Panicked task 0xffffff86aaf28218: 163 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff9043cd2598, Frame : Return Address
0xffffffe281bc77b0 : 0xffffff801c1edf9d 
0xffffffe281bc7800 : 0xffffff801c35b786 
0xffffffe281bc7840 : 0xffffff801c34aa10 
0xffffffe281bc7890 : 0xffffff801c188951 
0xffffffe281bc78b0 : 0xffffff801c1ee27d 
0xffffffe281bc79a0 : 0xffffff801c1ed929 
0xffffffe281bc7a00 : 0xffffff801c9e0ecb 
0xffffffe281bc7af0 : 0xffffff801c34aee3 
0xffffffe281bc7c70 : 0xffffff801c34abd2 
0xffffffe281bc7cc0 : 0xffffff801c188951 
0xffffffe281bc7ce0 : 0xffffff801c24ab71 
0xffffffe281bc7e00 : 0xffffff801e8d9d22 
0xffffffe281bc7e30 : 0xffffff801c8e4683 
0xffffffe281bc7ea0 : 0xffffff801c9134ae 
0xffffffe281bc7ee0 : 0xffffff801c8e38df 
0xffffffe281bc7f60 : 0xffffff801c8f0337 
0xffffffe281bc7fa0 : 0xffffff801c18819e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[96D09D0F-DCBC-317F-A955-2BAAEB46FCFF]@0xffffff801e8cb000->0xffffff801e92bfff
            dependency: com.apple.iokit.IOReportFamily(47)[3115840D-90F4-383F-8991-D3CF8891C24C]@0xffffff801ec40000->0xffffff801ec42fff

Process name corresponding to current thread (0xffffff9043cd2598): kernel_task

Mac OS version:
22A380

Kernel version:
Darwin Kernel Version 22.1.0: Sun Oct  9 20:14:54 PDT 2022; root:xnu-8792.41.9~2/RELEASE_X86_64
Kernel UUID: BF7C9676-EF23-3E8D-A2E2-25DAC63091B6
roots installed: 0
KernelCache slide: 0x000000001be00000
KernelCache base:  0xffffff801c000000
Kernel slide:      0x000000001bedc000
Kernel text base:  0xffffff801c0dc000
__HIB  text base: 0xffffff801bf00000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: YES
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 2063154673247
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000001e05d99f9be
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000004bfcc8ad0 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8042dc7000 - 0xffffffa042dc7000
  . PGZ   : 0xffffff8042dc7000 - 0xffffff8044dc8000
  . VM    : 0xffffff8044dc8000 - 0xffffff85115c7000
  . RO    : 0xffffff85115c7000 - 0xffffff86aadc7000
  . GEN0  : 0xffffff86aadc7000 - 0xffffff8b775c7000
  . GEN1  : 0xffffff8b775c7000 - 0xffffff9043dc7000
  . GEN2  : 0xffffff9043dc7000 - 0xffffff95105c7000
  . GEN3  : 0xffffff95105c7000 - 0xffffff99dcdc7000
  . DATA  : 0xffffff99dcdc7000 - 0xffffffa042dc7000
  Metadata: 0xffffffe28ddf5000 - 0xffffffe2addf5000
  Bitmaps : 0xffffffe2addf5000 - 0xffffffe2b3df5000

`

plsh2me, 2022-10-30:
@Xiashangning Hi, I have tested the kext for macOS Ventura on SL3 and SP7.

On SL3: Kext works without any issue (except the sleep problem-still there).

SP7: Kext works, but only when I shut down the device it restarts. Rather than this the kext functions normally it should be. I added the output of - sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB" and the panic report. 

So when you have time you please have a look on these.
Thank you very much.

[kext log & panicreport.zip](https://github.com/Xiashangning/BigSurface/files/9895594/kext.log.panicreport.zip)

Xiashangning, 2022-10-30:
Please add keepsyms=1 to bootargs to get the debug symbols

backdash756, 2022-10-30:
This is the same behavior as mine and seems to be the same panic. Here also my panic with keepsyms=1:
[panic.zip](https://github.com/Xiashangning/BigSurface/files/9895949/panic.zip)

plsh2me, 2022-10-30:
@Xiashangning  It is related with the type cover. Without type cover it shutdown normally and also turn on without panic report.

Rainringgoaway, 2022-11-01:
problem came from Voodool.

anschein, 2022-11-02:
> Hello. I installed Ventura and it works for now. What doesn't work is turning it off normally. After switching off, the computer restarts with the message "Rebooted because an error was detected" [...]

Same problem here,using SP7,and I'm following load the Bigsurface.kext before others,shutdown and reboot works fine,just lost touchpad and battert indicator

bitoy81, 2022-11-03:
> @Xiashangning Hi, I have tested the kext for macOS Ventura 13.0 on SL3 and SP7.
> 
> On SL3: Kext works without any issue (except the sleep problem-still there).
> 
> SP7: Kext works, but only when I shut down the device restarts. Rather than this the kext functions normally it should be. I added the output of - sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB" and the panic report.
> 
> So when you have time, please have a look on these. Thank you very much.
> 
> [kext log & panicreport.zip](https://github.com/Xiashangning/BigSurface/files/9895594/kext.log.panicreport.zip)

@plsh2me what is that sleep issue? i got SL3 too. is it delay to waking up and going to sleep that is my issue since monterey.

balopez83, 2022-11-11:
@Xiashangning First off thank you for all your hard work. Have you had any progress on this KP issue? I am happy to assist with any coding or testing as needed. I have a GitHub supporting the SP4/SB 1st gen which also uses your kext and the same problem exists for both devices. I also have added your Surface Keyboard code into a vanilla Voodooi2c per your direction from a previous ticket I submitted to get the SP3 running with touch and it also has the same issue of KP on shutdown/reboot.

billabongbruno, 2022-11-29:
@Xiashangning , hello there, once again.
Just chiming in to wonder if there any news regarding this issue and if you indeed plan to upgrade to Ventura or not.

billabongbruno, 2022-12-01:
> No, this seems to be a bug on VoodooI2C side. Right now i am afraid nothing could be done.

I see.
Well, hopefully VoodooInput gets edited to support Ventura, eventually.

Thank you anyway.

piticrobert, 2022-12-01:
Same problem here, on Surface Laptop Go.

dclive, 2022-12-03:
> Same problem here, on Surface Laptop Go.

Can you share your EFI for SLG?  I'm having a lot of trouble even getting ...anywhere... on the SLG.  I'm happy to stick with 12.x. 

Adding the BigSurface causes a kp (with stack displayed referencing AppleSMC and IONVMeController) at bootup in my 12.5.1.  But not using it causes walls of text and a NOT symbol, so I do not have a working EFI yet, with or without this...

dclive, 2022-12-04:
(prior question now resolved) 

I've posted a good EFI for a SLG on my repository if anyone is interested in a great starting point for the Surface Laptop Go.  Everything works, including sleep/wake and touchpad/battery/touchscreen, using 12.6.1.  

Thanks for posting that Ventura has a problem with this with the SLG!

misaka-20002, 2022-12-07:
Well it's confirmed a VoodooI2CHID bug
located VoodooI2CTouchscreenHIDEventDriver.cpp

dclive, 2022-12-07:
Sorry, I'm not clear on the details. 

Is the SHUTDOWN CRASH a confirmed bug [this, I am aware of] or is the INABILITY TO SLEEP a confirmed bug [this, I wasn't aware of].  Can you confirm?

misaka-20002, 2022-12-08:
ofc KP

kprinssu, 2022-12-09:
@Xiashangning I am one of the devs on VoodooI2C, I would like to track down what is the causing the panic. As I don't exactly have hardware that causes this panic, I am unable to track down where exactly the panic occurs.

Would you be able to ping me on Gitter so we can collaborate on fixing this together?

Xiashangning, 2022-12-10:
@kprinssu Sure ! I have upgraded my device to Ventura and will look into this issue on my side.

kprinssu, 2022-12-30:
@Xiashangning This is not an issue with VoodooI2C. This is an issue with VoodooInput.

LuxferreX, 2023-01-08:
Same issues on Ventura，look forward to solving that.

Rainringgoaway, 2023-01-11:
> @Xiashangning This is not an issue with VoodooI2C. This is an issue with VoodooInput. Please see my [quick explanation](https://github.com/VoodooI2C/VoodooI2C/issues/507#issuecomment-1367947816).
> 
> Edit: This seemed to be an issue with VoodooInput and have proposed a [PR](https://github.com/acidanthera/VoodooInput/pull/23) to fix this panic.

waiting for big surface update. thx @kprinssu efforts

balopez83, 2023-02-05:
@XiashangningI attempted to recompile your kext with the updated VoodooInput and latest VoodooI2C code. I was successful however it looks like this issue still persists. Do you have any thoughts on where the issue might be? I also recompiled a clean VoodooI2C and VoodooI2CHID kext and also noticed this issue still persists on Surface devices so it seems that this still may be more than the proposed VoodooInput?

billabongbruno, 2023-02-07:
@balopez83 @Xiashangning 
In regards to this, I have done some experiments.

By using any give clean version of VoodooI2C, the trackpad is not recognized as one (therefore we lose gestures and acceleration is off), but it does not produce any panics.

By using a mix-match of @Xiashangning 's repo and VoodooI2C, I have come to the conclusion that by using the latest kexts from @acidanthera and @Xiashangning 's VoodooGPIO.kext from the latest BigSurface release, we get the trackpad to be recognized again, but the panic persists.

Hence, the logical conclusion that I draw from this is that VoodooGPIO.kext needs to be edited somehow in order to reflect the new changes made to the new VoodooInput.kext.

So, in a way, for now, VoodooGPIO.kext is the offending kext which is not allowing this problem to be fixed.

balopez83, 2023-02-07:
@Xiashangning & @billabongbruno I need to correct my earlier statement. VoodooI2C with the latest VoodooInput and other code changes does work but of course the keyboard is not working though the trackpad does and with gestures. I have attempted to create my own VoodooI2CHID.kext using only the two SurfaceTypeCoverDriver files extracted from BigSurface kext files and am able to drive the keyboard with full support for the trackpad and gestures. It is recognized as expected in IORegistryExplorer. In this configuration it still crashes on reboot. 

I need to dig further into the code but suspect this may be an issue requiring code edits that mirror the "WillTerminate" that was required in VoodooInput and I will investigate this further. I also suspect, though is less likely, that the assignment of external USB2 in usb mapping might also be contributing. Its possible and I will test, that assigning it to "internal" in usb mapping might make a difference, though again, I suspect this is unlikely. 

Once I dig further I will post my changes and request assistance to test. I am sure @Xiashangning is likely really busy as many of us are and so understand why this is likely not something he is able to fix right away. Thank you though for all your work and hopefully we can help contribute to this project to keep it going.

balopez83, 2023-02-08:
@billabongbruno can you tell me what happens when you boot with this VoodooI2CHID.kext attached? You should use a vanilla VoodooI2C.kext v2.7.1 with it. 
[VoodooI2CHID.kext.zip](https://github.com/Xiashangning/BigSurface/files/10681817/VoodooI2CHID.kext.zip)

I expect no touch screen and keyboard and touch pad should work fine. I do not know if it will cause a KP on reboot/shutdown

balopez83, 2023-02-08:
@Xiashangning & @billabongbruno I think I figured out the problem. It is the "willTerminate" code however the changes in the VoodooInput.kext and in two of drivers within the VoodooI2C code were not enough to resolve the issues on Surface devices. I have been able to get stable boot, restart, and shutdown with the attached modified VoodooI2C.kext and VoodooI2CHID.kext which fully implement the "willTerminate" code throughout the entire driver stacks (literally in every driver within the two kexts). I will do my best to get the same changes implemented within the BigSurface kext and post here pending an official update

Please test the following two kexts to see if there you all experience any KP issues. Keep in mind the touch screen will not be driven with these kexts as I simply was attempting to resolve the KP problems and will see about getting the touch screen driven in the official BigSurface kext after I make the same updates there. 

[VoodooI2C.kext.zip](https://github.com/Xiashangning/BigSurface/files/10682484/VoodooI2C.kext.zip)
[VoodooI2CHID.kext.zip](https://github.com/Xiashangning/BigSurface/files/10682485/VoodooI2CHID.kext.zip)

billabongbruno, 2023-02-08:
> VoodooI2CHID

If I used VoodooI2CHID, I couldn't even use the Keyboard or Trackpad.
I only used the VoodooI2C.

billabongbruno, 2023-02-08:
> 

With a vanilla VoodooI2CHID I couldn't get the keyboard or trackpad to work at all.

billabongbruno, 2023-02-08:
> 

Will try these now and report ASAP.

billabongbruno, 2023-02-08:
@balopez83 , after having tried the kexts you attached, I can't get the trackpad to work at all (not even working erratically - it simply does nothing).
This happened with VoodooSerial loaded and not loaded.
Whenever VoodooI2CHID is loaded, it prevents the trackpad from functioning.
Although, I must note that the keyboard now works (when loading VoodooI2CHID), which didn't happen with the vanilla version (nor the latest ones I get from Dortania build Repo - which can be found here, FYI: https://dortania.github.io/builds/).

So, for now, I guess I have to live with the KP.

seven-of-eleven, 2023-02-08:
@balopez83 @billabongbruno , I tried the two kexts above and had no issues with them.
1. The keyboard and trackpad both worked without issue.
2. The touchscreen did not work (which was expected). 
3. Shutdown worked without issue (no kernel panic). 

For clarity I installed the two kexts (and updated the config.plist), and also removed BigSurface.kext and disabled the SSDT-SURFACE.aml.

For reference I'm on Ventura 13.1. Let me know if you need any further details. Thanks.

balopez83, 2023-02-08:
@seven-of-eleven this is great news and I am glad to hear it is working as expected. @billabongbruno I forgot to mention that all BigSurface kexts and ssdts should be disabled during this test. Also make sure that any other VoodooInput (serial, GPIO, etc.) are also disabled. The only items that should be enabled are the ones included within those two kexts. Make sure the order is correct as well in the config.plist with VoodooGPIO first, VoodooI2CServices second, VoodooInput third, then VoodooI2C and then VoodooI2CHID. 

Let me know if everything works as expected and I will continue working on the BigSurface kext now to get my changes integrated. I am having some issues with the build process that I am working through. Once I can get it to build I will upload my changes.

balopez83, 2023-02-08:
Hi @billabongbruno & @seven-of-eleven , please try the following custom BigSurface kext and see if it drives everything. I had some issues when compiling that might have broken some stuff. Make sure to disable the previous kexts and re-enable the BigSurface SSDT.
[BigSurface-custom.kext.zip](https://github.com/Xiashangning/BigSurface/files/10689326/BigSurface-custom.kext.zip)

balopez83, 2023-02-09:
Alright!!!! @billabongbruno , @seven-of-eleven , @Xiashangning , @kprinssu I think we are good to go. I have tested on a Surface Book and it looks like everything is now working. I had to add the "willTerminate" code to all "drivers" and a few other code files to get it to stop having a KP on reboot/shutdown and there is a lot lol.

Here is the BigSurface using current release code modified with changes to stop the KP. Keep in mind @Xiashangning just changed the code and I was not able to get that code running with my changes but since its not incorporated into a release yet I am not going to worry about it. 

[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/10691251/BigSurface.kext.zip)

@Xiashangning I am attaching a zip of my build folder so you can see where I made changes to your code to get things to work. I would submit a change however your code has now changed with the addition of the new satellite kext and my changes won't work just yet with those changes. 

[BigSurface-Build-Folder.zip](https://github.com/Xiashangning/BigSurface/files/10691321/BigSurface-Build-Folder.zip)

seven-of-eleven, 2023-02-09:
@balopez83 I tested your modified BigSurface.kext and everything is working:
1. Keyboard, touchpad, touch screen, all working as expected.
2. Shutdown works without kernel panic and no reboot.

### Thank you!

Let me know if you need any further detail.
Specs -  SP7 i5 256GB SSD | 8GB ram

balopez83, 2023-02-09:
@seven-of-eleven glad I could help

billabongbruno, 2023-02-09:
> Alright!!!! @billabongbruno , @seven-of-eleven , @Xiashangning , @kprinssu I think we are good to go. I have tested on a Surface Book and it looks like everything is now working. I had to add the [...]

Sorry, I completely missed this, as our timezones are probably very apart. lol

I will have a go at it and report back.

I will test everything out, including a fresh install, which I previously had issues with too, having to force shutdown in order to reach the next stage of the installation process.

I will try to debug as much as I can and post it here, seeing as you already closed the other issue in your repo (as well you should - lol).

I am currently using my SP4 at work, so I won't have it on hand for a few more hours, but as soon as I do, I will give it a go.

Regardless if it works 100% or not (it most likely does), I would like to take the time to thank you, @balopez83 , for all your hard work and persistence in the pursuit of finding the solution to this issue and many others before it.

Your work has made a lot of people happy, for sure.
I am one of them, I know that much.

And even though he is probably busy, thank you as well, @Xiashangning, too.

And thank you, @kprinssu for having the first go at this and dishing out the PR which solved the issue for other users.

Best regards,
Bruno.

Xiashangning, 2023-02-09:
Hi guys! :)
Indeed I was quite busy these 2 months. But that was not all academics‘ fault.
After the exams, first it was the Chinese new year and then I went on vacations. XD
But now I am back again!
Thanks you all. You all have put a lot of effort into this annoying issue. @balopez83  @billabongbruno @seven-of-eleven .

> @Xiashangning I am attaching a zip of my build folder so you can see where I made changes to your code to get things to work. I would submit a change however your code has now changed with the addition of the new satellite kext and my changes won't work just yet with those changes.

@balopez83 Well I have made some research on my side too, the repo's code now is WIP as I intend to incorporate the latest update of the touch screen code from `linux-surface`. I would like to see your PR request so that your contribution can be correctly marked. What is the problem with the latest code of the repo exactly? I only moved all things related to HID to a separate kext as I can see. (`BigSurfaceHIDDriver`) You should use `git clone -recursive` to clone the entire project. If you only changed those HIDEventDriver's code, you can just fork `BigSurfaceHIDDriver` and submit a PR request there.

billabongbruno, 2023-02-09:
> Hi guys! :) Indeed I was quite busy these 2 months. But that was not all academics‘ fault. After the exams, first it was the Chinese new year and then I went on vacations. XD But now I am back [...]

Hey, @Xiashangning , welcome back again.

Thank you, but unfortunately all I was able to do was report, test and share my findings.
The heavy lifting was basically @balopez83 .

Once again, welcome back.

Hopefully this project continues to get more and more followers.

Best regards,
Bruno.

Xiashangning, 2023-02-09:
> Thank you, but unfortunately all I was able to do was report, test and share my findings.

No, for me the contribution is not depend on anyone's 'capabilities', but on the time and effort he/she spent.

Xiashangning, 2023-02-09:
New release done :)

Giant-Li, 2024-07-29:
Hello, I'm also having this problem on my surface pro 4-i5/8/256 running Ventura.
In detail, I update OC to 1.0 and all the kexts to the latest under Montery, then update Ventura through system updates.

Here's some of my plist:
1. Machine type: MacBook pro 14.1，2017
2. Value of PciRoot(0x0)/Pci(0x2,0x0) was set to 00001B59 to fake as Kaby Lake GPU
3. Add lilucpu=9 to 7C436110-AB2A-4BBB-A880-FE41995C9F82

Under Montery the shutdown/restart function worked properly. But after updating Ventura
1. Selection of shutdown from apple menu results in restart every time. And will get a notification screen after restarting.
2. The type cover lost connection sometimes, nothing can be done but restart. And the type cover seems not support hot plugin.

I read through this thread and didn't find a direct answer. And I noticed that the thread was marked as closed, so I hope someone will read my issue and give some help.

Thanks in advance.

balopez83, 2024-07-29:
@Giant-Li please see my repo for the surface pro 4. This issue has been resolved and does not impact my repo. The patching for lilucpu is not needed for one plus other better patches are implemented. My repo is not using the latest OC version but always using the latest version of kexts or OC does not always add features or stability. I will update my repo soon but encourage you to use my repo as is as it is working perfect with everything at this time. Currently my repo will support up to Sonoma officially.