Xiashangning/BigSurface #50: Can't boot with SSDT-SURFACE.aml

kevin-Curry, 2022-08-01:
Surface Laptop 3.
if I add SSDT-SURFACE.aml, it cannot boot.

kevin-Curry, 2022-08-01:
[image]
And this is my ACPI folder.
I don't know if something make it wrong.

kevin-Curry, 2022-08-01:
Sorry,actually nothing works. Including keyboard and touchscreen. Using version 6.1

kevin-Curry, 2022-08-01:
If I add ssdt surface
It cannot even boot

Xiashangning, 2022-08-01:
I need `sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceM|IntelP"`

Xiashangning, 2022-08-01:
Try this one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9235657/BigSurface.kext.zip)

PS: AWAC, PNLF-CFL, XOSI are not needed.

kevin-Curry, 2022-08-02:
> Try this one [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9235657/BigSurface.kext.zip)
> 
> PS: AWAC, PNLF-CFL, XOSI are not needed.

Not working.
And no output I added debugenhancer.kext

Xiashangning, 2022-08-02:
what about ioreg?

bitoy81, 2022-08-02:
use 6.0. it works fine in monterey including touch. 6.1 dont work on my surface laptop 3.

kevin-Curry, 2022-08-03:
> use 6.0. it works fine in monterey including touch. 6.1 dont work on my surface laptop 3.

Can you share your efi folder plz

Xiashangning, 2022-08-03:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9247532/BigSurface.kext.zip)

bitoy81, 2022-08-03:
@kevin-Curry i got an i5 10th gen 16gig surface laptop 3 13" i believe you got i7. but here is my config.plist. @Xiashangning im using some of the surface pro 7 from your project can you please check if i need to make changes in the plist especially the device properties. thank you in advance
[config surface laptop3.plist.zip](https://github.com/Xiashangning/BigSurface/files/9247778/config.surface.laptop3.plist.zip)

Xiashangning, 2022-08-03:
@bitoy81 I’ve sent a new build, plz test it

Xiashangning, 2022-08-03:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9247532/BigSurface.kext.zip)

@kevin-Curry @bitoy81 Please test this build, with SSDT-SURFACE

ja350exl, 2022-08-03:
Does not work on my Surface Book 2 either, on Monterey 12.1 (No effect at all). Tried the version above as well, together with 6.1 and 6.0.

Xiashangning, 2022-08-03:
this is not helping, I need to understand what is going wrong. not booting? keyboard/touchpad failed to work? battery?
All of these are based on the test build above.
i need sudo dmesg | grep -E 'SurfaceS|SurfaceB|SurfaceM|SurfaceH|IntelP' and the file saved from ioregistryexplorer.app

kevin-Curry, 2022-08-03:
I can't even boot with it. How can I give you the output?

Xiashangning, 2022-08-03:
I was speaking to ja350exl for that.
but even with the latest test build above you cannot boot into system?

kevin-Curry, 2022-08-03:
Yes.

ja350exl, 2022-08-03:
sudo dmesg | grep -E 'SurfaceS|SurfaceB|SurfaceM|SurfaceH|IntelP' returns an empty result, none of the features work (no battery icon showing up, volume button not working, etc) - it is as if I did not have the kext installed.
I am unable to upload the IOregistry exported file as the extension is not supported, hence I changed the extension (from .IOREG)
Thanks for the help
[IORegistryExplorer'sOutput.txt](https://github.com/Xiashangning/BigSurface/files/9252691/IORegistryExplorer.sOutput.txt)
.

Xiashangning, 2022-08-03:
> sudo dmesg | grep -E 'SurfaceS|SurfaceB|SurfaceM|SurfaceH|IntelP' returns an empty result,

@ja350exl That means you have to install DebugEnhancer.kext

ja350exl, 2022-08-03:
Ok, I will thanks

bitoy81, 2022-08-03:
@Xiashangning the new test build kext is not working for my surface laptop 3. i even tried both SSDT-SURFACE from 6.0 and 6.1 build. hope new build will still work with surface laptop 3. 6.0 works just fine including touch.

ja350exl, 2022-08-03:
I cannot boot after adding the kext. Im stuck here [image]

Xiashangning, 2022-08-04:
> @Xiashangning the new test build kext is not working for my surface laptop 3. i even tried both SSDT-SURFACE from 6.0 and 6.1 build. hope new build will still work with surface laptop 3. 6.0 works just fine including touch.

Yes, I know. Sorry for the bug. From 6.0 to 6.1 I just tried to integrate support for Surface Laptop 1&2 as well. Something must go wrong during my rewriting.
@kevin-Curry @bitoy81 @ja350exl Try this new build. This one should allow you to boot into the system and give some output. I need output of `sudo dmesg | grep -E "SurfaceS|SurfaceH"`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9256391/BigSurface.kext.zip)

PS: for the SSDT, always use the newest one.

Xiashangning, 2022-08-04:
It is reported to be solved. Plz try this build above. I will leave this here for your response.

bitoy81, 2022-08-04:
> Update: Try using this build. It is reported to work on another SL3 https://github.com/Xiashangning/IPTSDaemon/files/9256974/BigSurface.kext.zip

@Xiashangning 6.1  build works now. thank you so much. i know there are still work to be done specially on the sleep function. is it normal if i  put my SL3 to sleep to wait around 40 sec for it to sleep and to wake up around 10 sec to fully open? just want to know if its same timeframe for other surface devices.

closed 2022-08-04 by Xiashangning