Xiashangning/BigSurface #53: Battery indicator only working for tablet battery in Surface Book 2

ja350exl, 2022-08-08:
Thanks for the great work, @Xiashangning ! Everything as mentioned is working on the Surface Book 2. However, due to its dual battery configuration, the indicator does not reflect the percentage correctly. Would you mind looking into that? Thanks :)

MetalChris87, 2022-08-18:
I have the same issue with Surface Book 3. https://dortania.github.io/OpenCore-Post-Install/laptop-specific/battery.html
This link has a section all about dual battery laptops and how to patch VirtualSMC to pick this up. 
I will be giving this a try later if you also wanna look at it.

ja350exl, 2022-08-18:
Ok thanks. Will try

Xiashangning, 2022-08-18:
I have noted this issue, but i won’t be available these days. when i have extra time to solve this, i will let you know and send you test build
patching virtualsmc is useless, don’t waste your time trying

ja350exl, 2022-08-18:
Ok noted with thanks. :)

jlempen, 2022-10-08:
Hi all!
I tried setting the `BatteryCount` key to `2` in BigSurface's Info.plist file, hoping that my 15-Inch Surface Book 3 would pick up the second (and much larger) battery in the keyboard base, but it doesn't seem to have any kind of effect. The SB3 still only picks up the smaller battery in the clipboard (the tablet part).
@MetalChris87, @ja350exl, have you tried the "SSDT-BATC" method described in [acidanthera's VirtualSMC dual battery support readme](https://github.com/acidanthera/VirtualSMC/blob/master/Docs/Dual%20Battery%20Support.md)?
I have the impression that the SSDT-BATC method would break BigSurface's battery code.

balopez83, 2023-02-09:
Not sure if it will help as I know the hardware is a little different but you can try to use my Batt C code that I made for the Surface Book 1. You will need to match it to the DSDT of the SB3 but I don't think it will hurt the code in the BigSurface kext. You can always try it and remove it if it doesnt work. 

I use the BigSurface kext in my repo as well and the batt c method was the only way to get both batteries to properly show in macOS. Granted the hardware is different so your mileage may vary.

https://github.com/balopez83/Surface_Pro_4_and_Surface_Book_Hackintosh