Xiashangning/BigSurface #55: Request to add Surface Go 3

exp13579, 2022-08-11:
I have a surface go 3 i3 version with 128gb SSD and 8gb RAM, the UART device id is 9D66. I wonder whether  it will work for my device. Thank you.

Xiashangning, 2022-08-11:
interesting, i thought go 3 won’t need this. plz send me ioreg saved from ioregistryexplorer.app

exp13579, 2022-08-11:
Let me find a time to install macOS and I will get you back with the file. Thanks in advance.

exp13579, 2022-08-15:
[sg3.zip](https://github.com/Xiashangning/BigSurface/files/9334344/sg3.zip)
OK, here you go. Many thanks.

Xiashangning, 2022-08-19:
Nope, just like other go series, your device is completely a normal laptop. After I examined your ioreg and DSDT, the only thing you need is some DSDT/SSDT modifications.
For example, you already have EC and battery present in your DSDT, just the name won't be recognised by macOS
[image]
Thus you should consult to the official guide on SSDT and then your battery will be fine.

closed 2022-08-19 by Xiashangning