Xiashangning/BigSurface #56: Mouse clicks randomly whilst I am typing

MetalChris87, 2022-08-19:
On my Surface Book 3 when I am typing anything my touchpad/mouse seems to click itself and this moves to the cursor so it looks something like this "I went for a walk to forest and saw a squirrel the local" so obviously it should have been I went for a walk to the local forest and saw a squirrel but because the mouse cursor clicks itself it messes up my typing. I use the laptop for work and so I constantly have to retype emails. 

Any help would be appreciated. 

Thanks

Xiashangning, 2022-08-25:
Got it, I will try to add a short silence time for touchpad when typing

Xiashangning, 2022-09-08:
Hi, I am back from school stuff. Sorry for the delay. I've checked the code, well, in fact there already exists a config in BigSurface's Info.plist -> IOKitPersonalities -> Surface Touchpad HID Event Driver -> QuietTimeAfterTyping. The unit is ms so by default it will silence the touchpad for 200ms = 0.2 second after typing event. You can change this to something bigger, like 500 and try to see if it is improved or not.

closed 2022-09-08 by Xiashangning