Xiashangning/BigSurface #58: Do Ambient Light Sensor Work？

wxxsfxyzm, 2022-08-20:
Newly installed Monterey on my SP7, but find no Auto Adjust Brightness in Display Settings, I don't know whether it is a problem from EFI or from BigSurface.kext
[image]
.

Xiashangning, 2022-08-25:
Well, have you installed SSDT-SURFACE ? Also you shouldn't have SMCLightSensor.kext in your config.
If everything is as required, then send me `sudo dmesg | grep "SurfaceA"` with DebugEnhancer.kext enabled

wxxsfxyzm, 2022-09-03:
efi maintainer BadStorm replied me that he didn't include your newest release. Updating manually solves the problem. Sorry for that.

closed 2022-09-08 by Xiashangning