Xiashangning/BigSurface #59: Surface Laptop Go brightness only works at 5 bars

frykher, 2022-08-20:
When using SSDT-PNLF-CML or SSDT-PNLF, I am able to control the brightness but the backlight only works when the slider is at 5 bars. Without these patches the display is bright but I am not able to control it.

Relevant parts of my EFI:
SSDT-PNLF-CFL.aml
SSDT-SURFACE.aml
SSDT-XOSI.aml
BigSurface.kext
BrightnessKeys.kext (brightness keys were fully functional even without this but removing it still gave this issue)

Additionally this is my igpu setup:
[image]

Let me know if anything else is needed from me. I really appreciate the work you've done to help me and others get their laptops running well at all
[ACPI.zip](https://github.com/Xiashangning/BigSurface/files/9388079/ACPI.zip)
[Drivers.zip](https://github.com/Xiashangning/BigSurface/files/9388080/Drivers.zip)
[image]
[config.plist](https://gist.github.com/frykher/510ba7210087b037bb19b25572c08f62)

frykher, 2022-08-20:
If it's relevant, waking after sleep does not turn on the display either

Xiashangning, 2022-08-25:
It is a SSDT related error I believe. You don't need BrightnessKeys.kext.
My suggestion is that you should go check the latest version of SSDT-PNLF and choose it according to your platform.

> I am able to control the brightness but the backlight only works when the slider is at 5 bars

I don't understand this, with slider more or less than 5 bars, what happens?
Also, `Whatevergreen` has some backlight patches for 10 gen graphics, you should enable that.
For reference, this is what I have
[image]

frykher, 2022-08-25:
Sorry for not being clear. When the slider is more or less than 5 bars the backlight does not work at all. I will try the fixes you proposed here.

frykher, 2022-08-26:
Thank you so much for all your help!! Not only do the brightness keys function properly, but waking from sleep works right as well (with like a 2 sec delay).

closed 2022-08-26 by frykher