Xiashangning/BigSurface #61: Surface firmware update aug 26 broke big surface kext

bitoy81, 2022-08-30:
Anyone have issues with their keyboard and touchpad? Looks like the aug firmware update broke the big surface kext. Lucky for people that did not update. Hope @Xiashangning will save the day. Thanks in advance.

plsh2me, 2022-08-30:
Yes. Same problem.
I found on windows - the keyboard became removable device after that update. If I press to remove then keyboard and touchpad stops working. This is very strange!!! Microsoft might solve this issue.
So it's sure that related to the bios update. 
Hope @Xiashangning will find a solution for that.

JChamberlain28, 2022-09-04:
I have SL 2 and have a non-working keyboard as well. Can provide any info that is requested for fixing.

Xiashangning, 2022-09-08:
@plsh2me @bitoy81 I need output of `sudo dmesg | grep SurfaceH`. Remember to enable DebugEnhancer.kext

plsh2me, 2022-09-08:
@Xiashangning 
Output -Last login: Thu Sep  8 22:13:19 on console
mdpalashmiah@Shoaibs-Air ~ % sudo dmesg | grep SurfaceH
Password:
[    0.745472]: SurfaceHIDNub::Woke up
[    0.756711]: SurfaceHIDDevice:0x10000038a start
[    0.777954]: SurfaceHIDDevice:0x10000038b start

bitoy81, 2022-09-08:
[    0.765168]: SurfaceHIDNub::Woke up
[    0.776789]: SurfaceHIDDevice:0x10000037b start
[    0.800615]: SurfaceHIDDevice:0x10000037c start

plsh2me, 2022-09-10:
I can confirm that for Surface Laptop 3 - the issue is related with bios. I have found two SL3, of them one has old bios which still works perfectly, and another has updated bios which not working.
[image]
[image]

Infligo, 2022-09-18:
Thanks for reporting this issue. I wasted so much time troubleshooting my new install until I found this thread. BIOS updates through Windows are the absolute worst. I didn't know it was turned on by default for these devices. I would have disabled it if I knew.

@Xiashangning Is this fixable, and do you need anything else from us besides the ioreg? Thank you!

Xiashangning, 2022-09-25:
@iloveps4 @plsh2me @Infligo @bitoy81 please test this test build.
it probably still doesn't work (build for SL1,2, but it will help me understand which part it fails to work. I have read linux's code but failed to find the difference, they seem to have no issue with this at all), so attach the output of `sudo dmesg | grep -E "SurfaceS|SurfaceH"`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9639132/BigSurface.kext.zip)

plsh2me, 2022-09-25:
@Xiashangning 
Output -Last login: Sun Sep 25 14:54:38 on console
mdpalashmiah@Shoaibs-Air ~ % sudo dmesg | grep -E "SurfaceS|SurfaceH"    
Password:
[    0.443634]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    0.443639]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.487703]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.588767]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.591266]: SurfaceSerialHubDriver::SAM version 14.602.139
[    0.596053]: SurfaceSerialHubDriver::Surface Battery nub published!
[    0.746121]: SurfaceSerialHubDriver::Timeout waiting for response
[    0.750791]: SurfaceSerialHubDriver::Surface HID nub published!
[    0.750824]: SurfaceHIDNub::Woke up
[    0.759874]: SurfaceHIDDevice:0x10000038b start
[    0.781919]: SurfaceHIDDevice:0x10000038c start
[    0.844949]: SurfaceSerialHubDriver::Warning, given buffer_len(60) and received data_len(2) mismatched!

Xiashangning, 2022-09-25:
Interesting, so the driver is loaded without any problem, but SAM doesn't send events anymore. Your keyboard still not responsive right? I am wondering if linux has this problem. Can you install a linux and verify whether that works?

plsh2me, 2022-09-25:
@Xiashangning To install linux- is there any specific procedure to be follow or enough to install normally?

Xiashangning, 2022-09-25:
Go check the `linux-surface` and install it according to their instructions. Also, you can try downgrade your firmware using this script [surface-uefi-firmware](https://github.com/linux-surface/surface-uefi-firmware)

plsh2me, 2022-09-25:
@Xiashangning Ubuntu 22.04.1 LTS running without any issue. Not even need to install something extra.

Infligo, 2022-09-30:
@Xiashangning `sudo dmesg | grep -E "SurfaceS|SurfaceH"` does not yield any results for me. Do you want me to send the output for `sudo dmesg` instead?

jlempen, 2022-10-02:
Hi all!
I can confirm that this issue affects the **Surface Book 3** series as well.
I have two Surface Book 3 devices, one of them got a firmware update a few days ago and the keyboard and trackpad won't work anymore with the BigSurface kext on macOS Monterey. 
The "HID Keyboard Device" got removable in Windows 11. 
The keyboard as well as the trackpad work just fine in (Manjaro) Linux, with or without the [linux-surface](https://github.com/linux-surface/linux-surface) kernel.
The device with the UEFI 15.11.140 stopped working, while the device with the older UEFI 13.101.140 still works perfectly.
It seems that the keyboard/trackpad of any Surface device stops working in macOS with the BigSurface kext once it gets updated to UEFI v15.xx.xx.
I'll try to downgrade to UEFI v13.xx.xx in order to fix the issue on the affected system.
@Xiashangning thank you so much for your brilliant work. Is there something we can do to help you resolve this issue?
[image]
[image]

Xiashangning, 2022-10-02:
Sadly nothing yet, this is indeed a weird and not so easy problem. I am investigating it right now.

agngwrnt, 2022-10-07:
@Xiashangning @jlempen 
can you explain how to downgrade this UEFI System? Thank You!

jlempen, 2022-10-07:
@agngwrnt 
This script [surface-uefi-firmware](https://github.com/linux-surface/surface-uefi-firmware) found in the [linux-surface](https://github.com/linux-surface/linux-surface) repo is designed for flashing Surface UEFI firmware updates and has pretty clear instructions for downgrading the UEFI on a Surface device.
For the Surface Book 3 series, the driver update pack from Microsoft which contains the last UEFI firmware version which works with the BigSurface kext is `SurfaceBook3_Win11_22000_22.072.22541.0.msi`
The above script works great for unpacking the driver update pack and processing the UEFI firmware .cab files.
Unfortunately, I encounter [this unresolved issue](https://github.com/linux-surface/surface-uefi-firmware/issues/23) when I try to downgrade the UEFI using the `fwupdmgr` command. The "firmware signature missing or not trusted" error could be bypassed by setting `OnlyTrusted=false` in `/etc/fwupd/daemon.conf`, but I don't want to take the risk of bricking my Surface Book 3 by doing so...
You may do so, but at your own risk :-)
I think it is much safer to wait until Xiashangning finds a way to solve our problem.

Xiashangning, 2022-10-11:
@Infligo @bitoy81 @plsh2me @jlempen @agngwrnt Guys, please test this build :)
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9760205/BigSurface.kext.zip)

jlempen, 2022-10-12:
@Xiashangning, thanks a lot for your test build. I won't have access to the SB3 with the problematic firmware until Saturday, but I'll test your new build right away when I get back to my place :-)

agngwrnt, 2022-10-12:
Hi @Xiashangning thanks for update, unfortunately I was downgrade my System UEFI, Intel ME, and SAM, can I confirm: 
1. uninstall the three drivers in device manager (windows).
2. downgrade those three drivers in linux.

agngwrnt, 2022-10-12:
Hi @Xiashangning thanks for update, unfortunately I was downgrade my System UEFI, Intel ME, and SAM. can I confirm everything went well after:

1. Uninstall the three drivers in device manager (UEFI, ME, SAM in windows).
2. Downgrade those three drivers in linux (see reference in [surface-uefi-firmware](https://github.com/linux-surface/surface-uefi-firmware)) (in my case UEFI 13.101.140, ME 13.0.2101.0, SAM 14.502.139) (Note: if you only downgrade UEFI its seem not work for me)
3. After that bigsurface.kext (old) running normally in (SL3 in my case).

Everything works normally, only the touchscreen doesn't work.

agngwrnt, 2022-10-12:
maybe, any suggest for fixing my touchscreen?

Xiashangning, 2022-10-12:
try this build, if it works without any problems for keyboard and touchpad. Then you can directly upgrade your firmware again. You have my words :) It will work

plsh2me, 2022-10-12:
> @Infligo @bitoy81 @plsh2me @jlempen @agngwrnt Guys, please test this build :) [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9760205/BigSurface.kext.zip)

@Xiashangning Hi, I have tried this kext. Device restarts after login to os.

bitoy81, 2022-10-12:
> > @Infligo @bitoy81 @plsh2me @jlempen @agngwrnt Guys, please test this build :) [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9760205/BigSurface.kext.zip)
> 
> @Xiashangning Hi, I have tried this kext. Device restarts after login to os.

same here my device restart boot loop

bitoy81, 2022-10-12:
my file vault is enabled so keyboard works before os completely loads. since i got file vault enabled it ask for the password before apple logo loads. after loading it shows the screen desktop and it reboots.

Xiashangning, 2022-10-12:
So with this build your keyboard works again (but has some bug causing kp)? Right?

plsh2me, 2022-10-12:
> what do you mean after login, do you use your external keyboard or the built in keyboard works when login?

After I gave login password, during loading OS, It's restarts. 

I didn't connect any keyboard. Cause already virtual keyboard was enabled and I was using only that.

Xiashangning, 2022-10-12:
Please test your keyboard and touchpad when possible. What about this one?
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768076/BigSurface.kext.zip)

bitoy81, 2022-10-12:
> So with this build your keyboard works again (but has some bug causing kp)? Right?

keyboard is working even with the old kext pre os loading. not sure if keyboard or touch pad works since it reboots right away.

bitoy81, 2022-10-12:
> Please test your keyboard and touchpad when possible. What about this one? [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768076/BigSurface.kext.zip)

will try it now

bitoy81, 2022-10-12:
> > Please test your keyboard and touchpad when possible. What about this one? [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768076/BigSurface.kext.zip)
> 
> will try it now

still restarts. tried moving touchpad before it restart but no movement

plsh2me, 2022-10-12:
> Please test your keyboard and touchpad when possible. What about this one?
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768076/BigSurface.kext.zip)

@Xiashangning My system boots in. 
No output by -Last login: Wed Oct 12 21:06:22 on console
mdpalashmiah@Shoaibs-Air ~ % sudo dmesg | grep -E "SurfaceS|SurfaceH"
Password:
mdpalashmiah@Shoaibs-Air ~ % 

Got last panic details - 

>panic(cpu 0 caller 0xffffff80095d0833): Kernel trap at 0xffffff800ee138e2, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x00000000000000a0, CR3: 0x000000000f555000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff86c636c5c0, RCX: 0xffffff8b8e3b2d40, RDX: 0x0000000000000004
RSP: 0xffffffe283e9b9a0, RBP: 0xffffffe283e9b9c0, RSI: 0x0000000000000003, RDI: 0x0000000000000000
R8:  0x000000000000003c, R9:  0x0000000000000000, R10: 0xffffff85282c2698, R11: 0x00000001a610a000
R12: 0xffffff8b8e4647a0, R13: 0x0000000000000001, R14: 0x0000000000000000, R15: 0x0000000000000004
RFL: 0x0000000000010286, RIP: 0xffffff800ee138e2, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00000000000000a0, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Panicked task 0xffffff8b8e5fd670: 175 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff8b8de6faa0, Frame : Return Address
0xffffffe283e9b350 : 0xffffff800947fdad mach_kernel : _handle_debugger_trap + 0x41d
0xffffffe283e9b3a0 : 0xffffff80095e10d6 mach_kernel : _kdp_i386_trap + 0x116
0xffffffe283e9b3e0 : 0xffffff80095d0443 mach_kernel : _kernel_trap + 0x4d3
0xffffffe283e9b430 : 0xffffff800941fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffe283e9b450 : 0xffffff800948017d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffe283e9b570 : 0xffffff800947f936 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffe283e9b5d0 : 0xffffff8009d15093 mach_kernel : _panic + 0x84
0xffffffe283e9b6c0 : 0xffffff80095d0833 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffe283e9b840 : 0xffffff80095d0516 mach_kernel : _kernel_trap + 0x5a6
0xffffffe283e9b890 : 0xffffff800941fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffe283e9b8b0 : 0xffffff800ee138e2 com.xavier.BigSurface : __ZN16SurfaceHIDDriver12getRawReportE20SurfaceHIDDeviceTypehPht + 0x22
0xffffffe283e9b9c0 : 0xffffff800ee25f2d com.xavier.BigSurface : __ZN16SurfaceHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x8d
0xffffffe283e9ba20 : 0xffffff800bcd8730 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice19updateElementValuesEPjj + 0x1f4
0xffffffe283e9bb90 : 0xffffff800edf6d02 com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver15getElementValueEP12IOHIDElement + 0x62
0xffffffe283e9bbd0 : 0xffffff800edf93ba com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver13parseElementsEv + 0x27a
0xffffffe283e9bc90 : 0xffffff800edf7ff4 com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver11handleStartEP9IOService + 0x2b4
0xffffffe283e9bd00 : 0xffffff800ee1a36c com.xavier.BigSurface : __ZN40VoodooI2CPrecisionTouchpadHIDEventDriver11handleStartEP9IOService + 0x2c
0xffffffe283e9bd30 : 0xffffff800bce6e69 com.apple.iokit.IOHIDFamily : __ZN17IOHIDEventService5startEP9IOService + 0xd5
0xffffffe283e9bd80 : 0xffffff800edf987c com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver5startEP9IOService + 0x2c
0xffffffe283e9bdc0 : 0xffffff8009c19cae mach_kernel : __ZN9IOService14startCandidateEPS_ + 0x10e
0xffffffe283e9be20 : 0xffffff8009c1976e mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0xdce
0xffffffe283e9bef0 : 0xffffff8009c18810 mach_kernel : __ZN9IOService14doServiceMatchEj + 0x3e0
0xffffffe283e9bf50 : 0xffffff8009c1b837 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x157
0xffffffe283e9bfa0 : 0xffffff800941f19e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[F847D524-DF1D-3389-AEDB-9F059F8AB7E9]@0xffffff800bcc9000->0xffffff800bd25fff
            dependency: com.apple.iokit.IOReportFamily(47)[13547D50-3F54-31F3-ADC4-69ABFE0D5ABA]@0xffffff800c073000->0xffffff800c075fff
         com.xavier.BigSurface(1.1)[49204420-EA64-3340-835E-07183F4004EC]@0xffffff800edf4000->0xffffff800ee94fff
            dependency: as.vit9696.Lilu(1.6.2)[264B15BE-8923-3A33-A9F5-8F0FFBB80595]@0xffffff800d464000->0xffffff800d48ffff
            dependency: as.vit9696.VirtualSMC(1.3.0)[40CF5E80-248F-35B7-833D-2DABF724B391]@0xffffff800d493000->0xffffff800d4a9fff
            dependency: com.alexandred.VoodooI2CServices(1.0)[8AA1B471-9422-37CE-B4E3-24E62A432CA5]@0xffffff800ed8f000->0xffffff800ed99fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CD3677CB-1370-3425-AB61-064B4F0508E6]@0xffffff800bb8c000->0xffffff800bb8dfff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[F847D524-DF1D-3389-AEDB-9F059F8AB7E9]@0xffffff800bcc9000->0xffffff800bd25fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[31170FF9-33B7-341F-865B-4E0D7872EF82]@0xffffff800c035000->0xffffff800c061fff
            dependency: com.apple.iokit.IOSMBusFamily(1.1)[2D0607F4-96F1-3D3F-BF7B-E00984DFBA9E]@0xffffff800c100000->0xffffff800c100fff
            dependency: com.xavier.VoodooSerial(1.2)[8AD9505D-8E0B-3760-8440-BEA6BBBB68B3]@0xffffff800edc9000->0xffffff800edeafff
            dependency: org.coolstar.VoodooGPIO(1.2.0)[5E20327A-12E0-3F09-87FB-99E397D4BB31]@0xffffff800eda3000->0xffffff800edbefff

Process name corresponding to current thread (0xffffff8b8de6faa0): kernel_task
[... cut ...]
|IOKitRegistryCompatibility	1
|EndpointSecurity	1
@Dont_Steal_Mac_OS_X	7.0.0
@kec.!AEncryptedArchive	1
>driverkit.serial	6.0.0
|IOSerial!F	11
@kext.triggers	1.0
>DspFuncLib	340.2
@kext.OSvKernDSPLib	529
>!UAudio	416.2
>!AAudioClockLibs	140.1
>usb.IOUSBHostHIDDevice	1.2
>!AGraphicsControl	6.5.7
>!AHDA!C	340.2
|IOHDA!F	340.2
|IOAudio!F	340.2
@vecLib.kext	1.2.0
>!ABacklightExpert	1.1.0
|IONDRVSupport	597
>!ASMBus!C	1.0.18d1
|IO!BSerialManager	9.0.

Touch Screen not working anymore. 
Trackpad not loads on system preferences.

Xiashangning, 2022-10-12:
Try this, I need logs `sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768609/BigSurface.kext.zip)

plsh2me, 2022-10-12:
> Try this, I need logs `sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"`
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768609/BigSurface.kext.zip)

Nothing works.

Output - 
mdpalashmiah@Shoaibs-Air ~ % sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"
Password:
[    0.455015]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    0.455021]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.455257]: SurfaceButtonDriver::Found valid GPIO interrupts
[    0.455264]: SurfaceButtonDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.486913]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.555445]: SurfaceButtonDriver::Surface ACPI button device found!
[    0.555552]: SurfaceButtonDevice:0x100000357 start
[    0.555962]: SurfaceButtonDevice:0x100000357 creating interfaces
[    0.587950]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.590484]: SurfaceSerialHubDriver::SAM version 14.602.139
[    0.656373]: SurfaceButtonDriver::Woke up
[    0.656382]: SurfaceButtonDevice:0x100000357 message: 0xe0000230 from: 0x100000294 0
[    5.781091]: SurfaceButtonDevice:0x100000357 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[    5.893654]: SurfaceButtonDevice:0x100000357 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[   30.588795]: SurfaceSerialHubDriver::Surface Battery nub published!
[   30.588966]: SurfaceBatteryNub::Woke up
[   30.592987]: SurfaceBatteryDriver::Plugin submitted
[   30.609500]: SurfaceBatteryDriver::Woke up
[   30.739349]: SurfaceSerialHubDriver::Timeout waiting for response
[   30.743079]: SurfaceSerialHubDriver::Surface HID nub published!
[   30.743131]: SurfaceHIDNub::Woke up
[   30.746500]: SurfaceSerialHubDriver::Unexpected response from event-enable request
[   30.746503]: SurfaceHIDDriver::HID event registration failed!
[   30.955114]: SurfaceBatteryDriver::Got BST/PSR update event
[   52.842688]: 1 duplicate report for Sandbox: AssetCacheLocato(508) deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: triald(467) deny(1) iokit-open-user-client AppleKeyStoreUserClientSandbox apply: AssetCache[509] <bytes>Sandbox apply: distnoted[510] <bytes>Sandbox apply: UsageTrackingAge[491] <bytes>Sandbox apply: ContainerMetadat[500] <bytes>Sandbox apply: mediasharingd[433] <bytes>1 duplicate report for Sandbox: triald(467) deny(1) iokit-open-user-client AppleKeyStoreUserClientSandbox: ContainerMetadat(500) deny(1) user-preference-read kCFPreferencesAnyApplicationSurfaceBatteryDriver::Got BST/PSR update event
[   54.763718]: Sandbox apply: AssetCacheTether[511] <bytes>Sandbox apply: parsec-fbf[468] <bytes>Sandbox apply: cloudpaird[441] <bytes>Sandbox: CommCenter(464) deny(1) mach-lookup com.apple.lsd.mapdbSystem Policy: CommCenter(464) deny(1) file-read-data /private/var/folders/m5/mb_nrdr53b1fck8v243hdskr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSystem Policy: UserEventAgent(76) deny(1) file-read-data /private/var/db/installcoordinationd/Library/InstallCoordinationSystem Policy: UserEventAgent(76) deny(1) file-read-data /private/var/db/installcoordinationd/LibrarySystem Policy: UserEventAgent(76) deny(1) file-read-data /private/var/db/installcoordinationdSystem Policy: CommCenter(464) deny(1) file-read-metadata /private/var/folders/m5/mb_nrdr53b1fck8v243hdskr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstore2 duplicate reports for System Policy: CommCenter(464) deny(1) file-read-metadata /private/var/folders/m5/mb_nrdr53b1fck8v243hdskr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSandbox: CommCenter(464) deny(1) mach-lookup com.apple.lsd.modifydb22 duplicate reports for Sandbox: CommCenter(464) deny(1) mach-lookup com.apple.lsd.modifydbSandbox: parsec-fbf(468) deny(1) file-read-data /Users/mdpalashmiah/Library/Caches/GeoServicesSandbox: CommCenter(464) deny(1) mach-lookup com.apple.lsd.modifydbSurfaceBatteryDriver::Got BST/PSR update event
[   60.096638]: SurfaceBatteryDriver::Got BST/PSR update event
[   62.215576]: SurfaceBatteryDriver::Got BST/PSR update event
[   79.258058]: SurfaceBatteryDriver::Got BST/PSR update event
[   80.011430]: Sandbox: com.apple.Safari(540) deny(1) user-preference-write com.apple.Safari.HistorySurfaceBatteryDriver::Got BST/PSR update event
[   88.339334]: SurfaceBatteryDriver::Got BST/PSR update event
[   88.849362]: successfully parsed 1/4 entries for file observations.dbSurfaceBatteryDriver::Got BST/PSR update event

Infligo, 2022-10-12:
The stars aligned, and I ended up giving my SL3 to my dad since he needed a new laptop and got myself an upgraded model. I found an SL3 with an i7 that was still on the older UEFI version, so I can't help test anymore since the keyboard and trackpad work for me in the current BigSurface release. Thanks for all of your hard work on this kext!

Xiashangning, 2022-10-12:
@plsh2me @bitoy81 
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768777/BigSurface.kext.zip)

plsh2me, 2022-10-12:
> @plsh2me @bitoy81
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768777/BigSurface.kext.zip)

Like previous one.

Output:
mdpalashmiah@Shoaibs-Air ~ % sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"
Password:
[    0.450530]: SurfaceSerialHubDriver::Found valid UART bus and GPIO interrupt!
[    0.451750]: SurfaceSerialHubDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.452463]: SurfaceButtonDriver::Found valid GPIO interrupts
[    0.452472]: SurfaceButtonDriver::Got GPIO Controller! VoodooGPIOIceLakeLP
[    0.499157]: SurfaceSerialHubDriver::Got UART Controller! VoodooUARTController
[    0.552592]: SurfaceButtonDriver::Surface ACPI button device found!
[    0.552708]: SurfaceButtonDevice:0x100000357 start
[    0.553075]: SurfaceButtonDevice:0x100000357 creating interfaces
[    0.599347]: SurfaceSerialHubDriver::Surface Serial Hub found!
[    0.601919]: SurfaceSerialHubDriver::SAM version 14.602.139
[    0.654158]: SurfaceButtonDriver::Woke up
[    0.654211]: SurfaceButtonDevice:0x100000357 message: 0xe0000230 from: 0x100000293 0
[    5.863135]: SurfaceButtonDevice:0x100000357 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[    6.040319]: SurfaceButtonDevice:0x100000357 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[   30.600611]: SurfaceSerialHubDriver::Surface Battery nub published!
[   30.600756]: SurfaceBatteryNub::Woke up
[   30.604540]: SurfaceBatteryDriver::Plugin submitted
[   30.619203]: SurfaceBatteryDriver::Woke up
[   30.750751]: SurfaceSerialHubDriver::Timeout waiting for response
[   30.754378]: SurfaceSerialHubDriver::Surface HID nub published!
[   30.754422]: SurfaceHIDNub::Woke up
[   39.216142]: SurfaceHIDDriver::here
[   39.291947]: SurfaceHIDDriver::here
[   39.304226]: SurfaceHIDDriver::here
[   39.373741]: SurfaceHIDDriver::here
[   39.465967]: SurfaceHIDDriver::here
[   39.538835]: SurfaceHIDDriver::here
[   39.616403]: SurfaceHIDDriver::here
[   39.699968]: SurfaceHIDDriver::here
[   39.791289]: SurfaceHIDDriver::here
[   39.905243]: SurfaceHIDDriver::here
[   39.999230]: SurfaceHIDDriver::here
[   40.110632]: SurfaceHIDDriver::here
[   40.205415]: SurfaceHIDDriver::here
[   40.316533]: SurfaceHIDDriver::here
[   40.410978]: SurfaceHIDDriver::here
[   40.521521]: SurfaceHIDDriver::here
[   40.617765]: SurfaceHIDDriver::here
[   40.729046]: SurfaceHIDDriver::here
[   47.124044]: 1 duplicate report for Sandbox: TrustedPeersHelp(395) deny(1) user-preference-write com.apple.TrustedPeersHelperSandbox: itunescloudd(390) deny(1) mach-lookup com.apple.private.corewifi-xpcSandbox: itunescloudd(390) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox apply: pkd[401] <bytes>Sandbox apply: secinitd[404] <bytes>Sandbox apply: containermanager[407] <bytes>1 duplicate report for Sandbox: itunescloudd(390) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: itunescloudd(390) deny(1) user-preference-write com.apple.mobileipodSandbox apply: identityservices[403] <bytes>Sandbox apply: usernoted[408] <bytes>Sandbox apply: QuickLookUIServi[402] <bytes>Sandbox apply: contactsd[411] <bytes>Sandbox apply: rapportd[410] <bytes>Sandbox apply: siriactionsd[406] <bytes>Sandbox apply: donotdisturbd[412] <bytes>Sandbox: donotdisturbd(412) deny(1) mach-lookup com.apple.tccd.system1 duplicate report for Sandbox: donotdisturbd(412) deny(1) mach-lookup com.apple.tccd.systemSandbox: siriactionsd(406) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox apply: CalendarAgent[409] <bytes>Sandbox apply: routined[413] <bytes>Sandbox apply: NotificationCent[414] <bytes>SurfaceBatteryDriver::Got BST/PSR update event
[   48.268120]: uipc_accept: peer disconnected unp_gencnt 164uipc_accept: peer disconnected unp_gencnt 172uipc_accept: peer disconnected unp_gencnt 180schedule_before_userret(2873): 1/4 tasks scheduledhook_thread_userret calleduserret(2873): invoking callback; 0/4 remainingSandbox apply: com.apple.quickl[436] <bytes>Sandbox apply: AirPlayUIAgent[451] <bytes>Sandbox apply: Siri[447] <bytes>SurfaceBatteryDriver::Got BST/PSR update event
[   53.119162]: Sandbox apply: diagnosticextens[508] <bytes>SurfaceBatteryDriver::Got BST/PSR update event
[   54.268433]: Sandbox: triald(478) deny(1) iokit-open-user-client AppleKeyStoreUserClient1 duplicate report for Sandbox: triald(478) deny(1) iokit-open-user-client AppleKeyStoreUserClientSandbox: assistantd(499) deny(1) iokit-open-user-client AppleKeyStoreUserClient1 duplicate report for Sandbox: assistantd(499) deny(1) iokit-open-user-client AppleKeyStoreUserClientSandbox: commerce(512) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plist1 duplicate report for Sandbox: commerce(512) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: commerce(512) deny(1) file-write-create /Users/mdpalashmiah/Library/HTTPStorages/commerceSandbox: commerce(512) deny(1) file-read-data /private/var/db/nsurlstoraged/dafsaData.binuipc_accept: peer disconnected unp_gencnt 297SurfaceBatteryDriver::Got BST/PSR update event
[   62.569563]: SurfaceBatteryDriver::Got BST/PSR update event
[   62.768995]: 1 duplicate report for Sandbox: cloudpaird(453) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: cloudpaird(453) deny(1) system-fsctl _IO('h', 47)Sandbox: cloudpaird(453) deny(1) mach-lookup com.apple.metadata.mdssuccessfully parsed 1/4 entries for file Assetssuccessfully parsed 1/4 entries for file FrameworkCachessuccessfully parsed 1/4 entries for file 29f01b5c701f9fb880c7098678ed8f898726e05aSurfaceBatteryDriver::Got BST/PSR update event

Xiashangning, 2022-10-12:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768950/BigSurface.kext.zip)

plsh2me, 2022-10-12:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768950/BigSurface.kext.zip)

System reboots (boot loop) after login screen.

bitoy81, 2022-10-13:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9768950/BigSurface.kext.zip)

same result boot loop

Xiashangning, 2022-10-13:
Can you try disable BigSurface
 to get into the system and get me the kernel panic log ? I am really confused why it will kp

plsh2me, 2022-10-13:
@Xiashangning Please check the panic log - 
[last panic log.zip](https://github.com/Xiashangning/BigSurface/files/9779100/last.panic.log.zip)

bitoy81, 2022-10-13:
> Can you try disable BigSurface to get into the system and get me the kernel panic log ? I am really confused why it will kp

Panic(CPU 0, time 11723903370): NMIPI for spinlock acquisition timeout, spinlock: 0xffffff80050f2190, spinlock owner: 0xffffff9507f16540, current_thread: 0xffffff9507f16540, spinlock_owner_cpu: 0x0
RAX: 0x0000000000000000, RBX: 0xffffff99d373fa08, RCX: 0x00000000f903a344, RDX: 0xffffff903a344000
RSP: 0xffffffb0dd48a5d8, RBP: 0xffffffb0dd48a640, RSI: 0xffffff8508271a90, RDI: 0x00000000000049f1
R8:  0xffffffa08dae8740, R9:  0x00000000000000b0, R10: 0xffffff8508271a90, R11: 0xffffff9507f16540
R12: 0xffffff80050f2148, R13: 0xffffffa0c3f6e620, R14: 0x00000000000000b0, R15: 0x0000000000004004
RFL: 0x0000000000000087, RIP: 0xfffff6a340001c30, CS:  0x0000000000000008, SS:  0x0000000000000010
Panicked task 0xffffff9507318d10: 1 threads: pid 328: mount_apfs
Backtrace (CPU 0), panicked thread: 0xffffff9507f16540, Frame : Return Address
0xffffffb0dcec3f70 : 0xffffff80041cfc27 mach_kernel : _interrupt + 0xd7
0xffffffb0dcec3fd0 : 0xffffff800401fc1d mach_kernel : _hndl_allintrs + 0x11d
0xffffffb0dd48a640 : 0xffffff800485bfe7 mach_kernel : __ZNK25IOGeneralMemoryDescriptor9MetaClass5allocEv + 0x37
0xffffffb0dd48a670 : 0xffffff800485d55f mach_kernel : __ZN18IOMemoryDescriptor16withAddressRangeEyyjP4task + 0x2f
0xffffffb0dd48a6b0 : 0xffffff8006d875e2 com.apple.iokit.IOStorageFamily : __ZL11dkreadwritePv9dkrtype_t + 0x3be
0xffffffb0dd48a770 : 0xffffff8004369dd8 mach_kernel : _spec_strategy + 0x448
0xffffffb0dd48a7d0 : 0xffffff8004360502 mach_kernel : _VNOP_STRATEGY + 0x42
0xffffffb0dd48a810 : 0xffffff800730ce07 com.apple.filesystems.apfs : _nx_bio_doread + 0x1b5
0xffffffb0dd48a860 : 0xffffff800730c871 com.apple.filesystems.apfs : __vnode_dev_read + 0x2a5
0xffffffb0dd48a8f0 : 0xffffff800730b562 com.apple.filesystems.apfs : _vnode_dev_read + 0x15
0xffffffb0dd48a910 : 0xffffff80074269cc com.apple.filesystems.apfs : _obj_read + 0x16a
0xffffffb0dd48a960 : 0xffffff8007425c2d com.apple.filesystems.apfs : _obj_get + 0x76d
0xffffffb0dd48aa80 : 0xffffff80073fd095 com.apple.filesystems.apfs : _btree_node_get_internal + 0x1a3
0xffffffb0dd48ab70 : 0xffffff80073fcee3 com.apple.filesystems.apfs : _btree_node_get + 0x76
0xffffffb0dd48abf0 : 0xffffff800740319c com.apple.filesystems.apfs : _btree_node_copy + 0x1e6
0xffffffb0dd48ad40 : 0xffffff80074031d7 com.apple.filesystems.apfs : _btree_node_copy + 0x221
0xffffffb0dd48ae90 : 0xffffff8007409108 com.apple.filesystems.apfs : _btree_copy + 0x7b
0xffffffb0dd48aee0 : 0xffffff80073426e9 com.apple.filesystems.apfs : _supplemental_tree_copy + 0x50
0xffffffb0dd48af20 : 0xffffff8007342aa6 com.apple.filesystems.apfs : _supplemental_tree_revert + 0x2f1
0xffffffb0dd48afd0 : 0xffffff80073185a3 com.apple.filesystems.apfs : _revert_to_snapshot + 0x6c3
0xffffffb0dd48b080 : 0xffffff800738bb49 com.apple.filesystems.apfs : _apfs_vfsop_mount + 0x207f
0xffffffb0dd48b8a0 : 0xffffff800432e390 mach_kernel : _vfs_iskernelmount + 0x1550
0xffffffb0dd48bb00 : 0xffffff800432fc22 mach_kernel : ___mac_mount + 0x422
0xffffffb0dd48bf00 : 0xffffff800432f7ee mach_kernel : _mount + 0x3e
0xffffffb0dd48bf40 : 0xffffff800478918b mach_kernel : _unix_syscall64 + 0x1fb
0xffffffb0dd48bfa0 : 0xffffff8004020236 mach_kernel : _hndl_unix_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.iokit.IOStorageFamily(2.1)[24743B95-072D-31C4-A59D-7FF38502C0C4]@0xffffff8006d78000->0xffffff8006d8efff
         com.apple.filesystems.apfs(1934.141.2)[A4D6592B-135F-3622-BBFA-2FB698814463]@0xffffff8007307000->0xffffff8007485fff
            dependency: com.apple.driver.AppleEFINVRAM(2.1)[7201FB4A-CE2A-3EDA-92DE-053285346974]@0xffffff80055b2000->0xffffff80055bbfff
            dependency: com.apple.driver.AppleEffaceableStorage(1.0)[53D12D3C-8326-3B2E-8FD6-204D97AEEF08]@0xffffff80055c8000->0xffffff80055cdfff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[752EFEA5-C07D-3246-AC42-A2C447402B68]@0xffffff8005ad3000->0xffffff8005adafff
            dependency: com.apple.iokit.IOStorageFamily(2.1)[24743B95-072D-31C4-A59D-7FF38502C0C4]@0xffffff8006d78000->0xffffff8006d8efff
            dependency: com.apple.kec.corecrypto(12.0)[864DE2A3-7343-32A3-AC76-68DF89128C87]@0xffffff80074bf000->0xffffff800753ffff
            dependency: com.apple.security.AppleImage4(4.2.0)[C55B9734-D46F-3F23-9B96-13FD4402B872]@0xffffff800564f000->0xffffff8005667fff

Process name corresponding to current thread (0xffffff9507f16540): mount_apfs
Boot args: keepsyms=1 swd_panic=1 igfxonln=1 -igfxblr -igfxcdc -igfxdvmt alcid=35 -noDC9 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21G83

Kernel version:
Darwin Kernel Version 21.6.0: Wed Aug 10 14:25:27 PDT 2022; root:xnu-8020.141.5~2/RELEASE_X86_64
Kernel UUID: D0DC2052-71F4-3DCB-B189-BEE5A59B3B42
KernelCache slide: 0x0000000003e00000
KernelCache base:  0xffffff8004000000
Kernel slide:      0x0000000003e10000
Kernel text base:  0xffffff8004010000
__HIB  text base: 0xffffff8003f00000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 11725869635
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000002baeaa1ce
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000eb202639a 0x0000000000000000
[... cut ...]
0xffffffb0dd0e31c0 : 0xffffff8004311444 mach_kernel : _cluster_read_ext + 0x174
0xffffffb0dd0e3230 : 0xffffff80073529a4 com.apple.filesystems.apfs : _apfs_nstream_read + 0x295
0xffffffb0dd0e3300 : 0xffffff800735391c com.apple.filesystems.apfs : _apfs_inode_getxattr + 0x1a7
0xffffffb0dd0e33e0 : 0xffffff800735376a com.apple.filesystems.apfs : _apfs_vnop_getxattr + 0xa3
0xffffffb0dd0e3410 : 0xffffff800435ff3a mach_kernel : _VNOP_GETXATTR + 0x5a
0xffffffb0dd0e3480 : 0xffffff80055f1e5b com.apple.AppleFSCompression.AppleFSCompressionTy

Xiashangning, 2022-10-13:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9780869/BigSurface.kext.zip)

plsh2me, 2022-10-13:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9780869/BigSurface.kext.zip)

Same thing. System rebooting after login screen.

bitoy81, 2022-10-14:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9780869/BigSurface.kext.zip)

same result here is my new panic log

panic(cpu 2 caller 0xffffff8012fd07e3): Kernel trap at 0xffffff80187f28b2, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000088, CR3: 0x0000000018f63000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffffaa02d4c3a0, RCX: 0xffffffa06a6b9a40, RDX: 0x0000000000000004
RSP: 0xffffffffd98039a0, RBP: 0xffffffffd98039c0, RSI: 0x0000000000000003, RDI: 0x0000000000000000
R8:  0x000000000000003c, R9:  0x0000000000000040, R10: 0xffffff906b121698, R11: 0x00000003887d0000
R12: 0xffffff96d1322a40, R13: 0x0000000000000001, R14: 0x0000000000000000, R15: 0x0000000000000004
RFL: 0x0000000000010286, RIP: 0xffffff80187f28b2, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000088, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Panicked task 0xffffffa53687e670: 171 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff9b9c7baaa0, Frame : Return Address
0xffffffffd9803350 : 0xffffff8012e7fd6d mach_kernel : _handle_debugger_trap + 0x41d
0xffffffffd98033a0 : 0xffffff8012fe1086 mach_kernel : _kdp_i386_trap + 0x116
0xffffffffd98033e0 : 0xffffff8012fd03f3 mach_kernel : _kernel_trap + 0x4d3
0xffffffffd9803430 : 0xffffff8012e1fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffffd9803450 : 0xffffff8012e8013d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffffd9803570 : 0xffffff8012e7f8f6 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffffd98035d0 : 0xffffff8013714e03 mach_kernel : _panic + 0x84
0xffffffffd98036c0 : 0xffffff8012fd07e3 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffffd9803840 : 0xffffff8012fd04c6 mach_kernel : _kernel_trap + 0x5a6
0xffffffffd9803890 : 0xffffff8012e1fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffffd98038b0 : 0xffffff80187f28b2 com.xavier.BigSurface : __ZN16SurfaceHIDDriver12getRawReportE20SurfaceHIDDeviceTypehPht + 0x22
0xffffffffd98039c0 : 0xffffff8018804efd com.xavier.BigSurface : __ZN16SurfaceHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x8d
0xffffffffd9803a20 : 0xffffff80156d8730 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice19updateElementValuesEPjj + 0x1f4
0xffffffffd9803b90 : 0xffffff80187d5c82 com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver15getElementValueEP12IOHIDElement + 0x62
0xffffffffd9803bd0 : 0xffffff80187d833a com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver13parseElementsEv + 0x27a
0xffffffffd9803c90 : 0xffffff80187d6f74 com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver11handleStartEP9IOService + 0x2b4
0xffffffffd9803d00 : 0xffffff80187f933c com.xavier.BigSurface : __ZN40VoodooI2CPrecisionTouchpadHIDEventDriver11handleStartEP9IOService + 0x2c
0xffffffffd9803d30 : 0xffffff80156e6e69 com.apple.iokit.IOHIDFamily : __ZN17IOHIDEventService5startEP9IOService + 0xd5
0xffffffffd9803d80 : 0xffffff80187d87fc com.xavier.BigSurface : __ZN33VoodooI2CMultitouchHIDEventDriver5startEP9IOService + 0x2c
0xffffffffd9803dc0 : 0xffffff8013619a1e mach_kernel : __ZN9IOService14startCandidateEPS_ + 0x10e
0xffffffffd9803e20 : 0xffffff80136194de mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0xdce
0xffffffffd9803ef0 : 0xffffff8013618580 mach_kernel : __ZN9IOService14doServiceMatchEj + 0x3e0
0xffffffffd9803f50 : 0xffffff801361b5a7 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x157
0xffffffffd9803fa0 : 0xffffff8012e1f19e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[54BC444D-71AC-3031-B356-D00F6D96B80B]@0xffffff80156c9000->0xffffff8015725fff
            dependency: com.apple.iokit.IOReportFamily(47)[66824C9A-7BC1-3F5D-A056-D789C8A86B8C]@0xffffff8015a73000->0xffffff8015a75fff
         com.xavier.BigSurface(1.1)[FFB75347-6C77-3DF7-AAC9-7AF36DD2BB1F]@0xffffff80187d3000->0xffffff8018873fff
            dependency: as.vit9696.Lilu(1.6.1)[7FE96C72-5555-3D86-BA9C-36D9177168FB]@0xffffff80184dd000->0xffffff8018508fff
            dependency: as.vit9696.VirtualSMC(1.3.0)[40CF5E80-248F-35B7-833D-2DABF724B391]@0xffffff8018519000->0xffffff801852ffff
            dependency: com.alexandred.VoodooI2CServices(1.0)[8AA1B471-9422-37CE-B4E3-24E62A432CA5]@0xffffff80185da000->0xffffff80185e4fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5481E015-DEF4-31B0-8649-67C6445D5CBA]@0xffffff801558c000->0xffffff801558dfff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[54BC444D-71AC-3031-B356-D00F6D96B80B]@0xffffff80156c9000->0xffffff8015725fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[22B97C0A-53BF-3843-B4B5-4DE866C375B0]@0xffffff8015a35000->0xffffff8015a61fff
            dependency: com.apple.iokit.IOSMBusFamily(1.1)[2D0607F4-96F1-3D3F-BF7B-E00984DFBA9E]@0xffffff8015b00000->0xffffff8015b00fff
            dependency: com.xavier.VoodooSerial(1.2)[8AD9505D-8E0B-3760-8440-BEA6BBBB68B3]@0xffffff801860a000->0xffffff801862bfff
            dependency: org.coolstar.VoodooGPIO(1.2.0)[5E20327A-12E0-3F09-87FB-99E397D4BB31]@0xffffff80185b4000->0xffffff80185cffff

Process name corresponding to current thread (0xffffff9b9c7baaa0): kernel_task
Boot args: keepsyms=1 swd_panic=1 igfxonln=1 -igfxblr -igfxcdc -igfxdvmt alcid=35 -noDC9 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21G83

Kernel version:
[... cut ...]
  . PGZ   : 0xffffff90690db000 - 0xffffff906b0dc000
  . VM    : 0xffffff906b0dc000 - 0xffffff95378db000
  . RO    : 0xffffff95378db000 - 0xffffff96d10db000
  . GEN0  : 0xffffff96d10db000 - 0xffffff9b9d8db000
  . GEN1  : 0xffffff9b9d8db000 - 0xffffffa06a0db000
  . GEN2  : 0xffffffa06a0db000 - 0xffffffa5368db000
  . GEN3  : 0xffffffa5368db000 - 0xffffffaa030db000
  . DATA  : 0xffffffaa030db000 - 0xffffffb0690db000
  Metadata: 0xffffffffd9fee000 - 0xfffffffff9fee000
  Bitmaps : 0xfffffffff9fee000 - 0xfffffffffffee000

Xiashangning, 2022-10-14:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9789565/BigSurface.kext.zip)

Xiashangning, 2022-10-14:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9789759/BigSurface.kext.zip)

plsh2me, 2022-10-14:
> BigSurface.kext.zip

Nothing works. 
Even no output -
sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"

Xiashangning, 2022-10-14:
impossible, please try download it again, delete the original one and copy the new build into the folder

plsh2me, 2022-10-14:
Ok, you are right. 
Now deleted old one and copy/paste new one. System rebooted and cleaned the nvram.
Screen touch working.
Trackpad on system > preferences loaded.
Keyboard & Trackpad not working.

Xiashangning, 2022-10-15:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9792833/BigSurface.kext.zip)

jlempen, 2022-10-15:
Hi @Xiashangning,
Thank you so much for your hard work on this issue!
With your latest BigSurface.kext from tonight, **the keyboard is working again** on my 13-Inch SB3!
The trackpad is still dead, though.

Xiashangning, 2022-10-15:
Wow, good news :)

Xiashangning, 2022-10-15:
Try this then @jlempen 
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9792903/BigSurface.kext.zip)

jlempen, 2022-10-15:
> Try this then @jlempen
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9792903/BigSurface.kext.zip)

This one kernel panics when reaching the login screen.

plsh2me, 2022-10-15:
> BigSurface.kext.zip

With this kext also keyboard working on SL3. 

OutPut- 

> Try this then @jlempen
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9792903/BigSurface.kext.zip)

And with this - rebooting after login screen.

bitoy81, 2022-10-15:
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9792833/BigSurface.kext.zip)

thank you @Xiashangning  one more issue to go. SL3. the newest link also reboots on me.

[32 more replies]

closed 2022-10-16 by bitoy81