Xiashangning/BigSurface #63: Surface Laptop 2 keyboard/trackpad not working on Mar 10 firmware

JChamberlain28, 2022-09-06:
I have a surface laptop 2 (8250u).
I have MacOS Monterey installed and BigSurface kext.

I am in Mar 10 firmware update so before the keyboard changes from Aug 26.

My trackpad works but no gestures. My keyboard does not work. Other features such as battery status seem to work.

I have noticed on boot these messages and wondered if this indicates anything.

If I know what files or logs to provide I can provide. Also happy to troubleshoot if anyone has had this problem and resolved it before.

Xiashangning, 2022-09-08:
Does your keyboard work BEFORE the firmware update?

JChamberlain28, 2022-09-10:
I have not updated my firmware to the one that has been reported in other issues (Aug 26). Support for SL keyboard was released after the firmware I am on, so I think your right it's not firmware that is the problem. Any further steps for me to troubleshoot would be extremely helpful ☺️

Xiashangning, 2022-09-24:
Sorry for the late reply, too occupied these days. Anyway, please test this test build.
In case of not working, attach `sudo dmesg | grep -E "SurfaceS|SurfaceB|SurfaceH"`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9639132/BigSurface.kext.zip)

Xiashangning, 2022-09-25:
trackpad shouldn't be an issue as long as BigSurface is properly loaded, battery also.

Xiashangning, 2022-09-29:
Try this one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9674555/BigSurface.kext.zip)

JChamberlain28, 2022-10-02:
> @JChamberlain28 @RockJesus @SmartBoy84 Sorry for the late reply, too occupied these days. Anyway, please test this test build. In case of not working, attach `sudo dmesg | grep -E "SurfaceS|SurfaceB|SurfaceH"` [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9639132/BigSurface.kext.zip)

All good. I appreciate you finding time to reply since you don't have to. I will test this in next few days. I'm also quite occupied currently.

Xiashangning, 2022-10-02:
@SmartBoy84 Try this one. Also does battery and trackpad works?
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9692487/BigSurface.kext.zip)

Xiashangning, 2022-10-04:
> What the hell? It just started working after a random number of reboots

Yes, if you compare your last two output above you will find some clue.  The reason why it fails is because that the communication from the hardware is really lossy, and it needs to retrive its report descriptor which is quite long (>100 bytes) .  So maybe sometimes it will work. The only thing I have changed in the last build is to force it to try multiple times for report descriptor.

Xiashangning, 2022-10-04:
> trackpad gestures (working with a stock VoodooI2C and whatever)

Are you using the latest VoodooI2C that is working? I need logs. `sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB|VoodooI2C"`

Xiashangning, 2022-10-04:
I need logs, and ioreg as well.

Xiashangning, 2022-10-04:
> Keyboard wasn't working this time round btw

Reboot and resend me logs and ioreg(please use IORegistryExplorer.app and save the file) please. I need them when your keyboard is working.

Xiashangning, 2022-10-04:
> Nope, I have no knowledge in CPU area. From what you described, I think this is a OC related issue (maybe something is misconfigured)

Xiashangning, 2022-10-04:
format please

Xiashangning, 2022-10-04:
@SmartBoy84 Try this one.
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9708693/BigSurface.kext.zip)

Xiashangning, 2022-10-04:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9710669/BigSurface.kext.zip)

Xiashangning, 2022-10-05:
No idea, sorry

RockJesus, 2022-10-06:
> sudo dmesg | grep -E "SurfaceS|SurfaceB|SurfaceH"
[out.txt](https://github.com/Xiashangning/BigSurface/files/9727017/out.txt)
still not working on sl

RockJesus, 2022-10-06:
> @JChamberlain28 @RockJesus @SmartBoy84 Sorry for the late reply, too occupied these days. Anyway, please test this test build. In case of not working, attach `sudo dmesg | grep -E [...]

[out.docx](https://github.com/Xiashangning/BigSurface/files/9727042/out.docx)
still not workin on sl

Xiashangning, 2022-10-06:
@RockJesus You should use the last build
This one

> @JChamberlain28 @SmartBoy84 Try this build, if no error, this will be the release version.
> [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9710669/BigSurface.kext.zip)

And attach the output immediately after you login, (with DebugEnhancer installed)

RockJesus, 2022-10-08:
> 

finally work on sl with touchpad &keyboard .  thx
How do i get touchscreen work?

Xiashangning, 2022-10-11:
Because I am still working on some other fixes.
BTW, can you turn on your caps lock on the keyboard?

Xiashangning, 2022-10-14:
@RockJesus
> How do i get touchscreen work?

First you need to find out where your touchscreen is attached to, I2C or MEI, send me your ioreg

Xiashangning, 2022-10-14:
@SmartBoy84 
> No, my caps lock doesn't turn on - it does toggle caps however
Try this build
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9780869/BigSurface.kext.zip)

RockJesus, 2022-10-15:
> 

touchscreen has worked by IPTSDaemon.
next how to get ALS work?thank u

Xiashangning, 2022-10-15:
> is your setup working fine on MacOS

Yes, cpu can boost to 3.5GHz.
However, your SL and my SP7 are VERY DIFFERENT devices and share no common config at all. Basically everything in your laptop is different.

Xiashangning, 2022-10-15:
@RockJesus Unfortunately MS uses different hardware so ALS totally relies on you to find the hardware and write the code.

Xiashangning, 2022-10-17:
No

Xiashangning, 2022-10-17:
@SmartBoy84  And plz try this test build to see if caps lock led can be turned on
if it is in an opposite way, please attach the output `sudo dmesg | grep "SurfaceS|SurfaceH"`
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9798780/BigSurface.kext.zip)

Xiashangning, 2022-10-18:
disable BigSurface and send me the panic log plz, you need to set keepsyms=1 in boot-args
If you don’t set, then please set it and enable the bigsurface to cause the kernel panic, then disable it so you can get detailed panic logs

pigworlds, 2022-10-19:
I hit panic when using this kext [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9798780/BigSurface.kext.zip):

The panic logs below come from Ventura, and the same panic happens to Monterey and lower too.

<details>
<summary>Panic logs</summary>

```
panic(cpu 2 caller 0xffffff800f34aff3): Kernel trap at 0xffffff8014300b53, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000249a1000, CR4: 0x00000000003626e0
RAX: 0xffffff952358000c, RBX: 0xffffff86be833a00, RCX: 0x0000000000000000, RDX: 0xffffff9523580b00
RSP: 0xffffffa07bf17c50, RBP: 0xffffffa07bf17c80, RSI: 0xffffff8b8a7969a4, RDI: 0x0000000000000000
R8:  0x000000000000000c, R9:  0xffffff8b8a796998, R10: 0x000000000000000c, R11: 0xffffffbdc7165000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0xffffff80142eace0, R15: 0x0000000000000001
RFL: 0x0000000000010286, RIP: 0xffffff8014300b53, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Panicked task 0xffffff9057378218: 189 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff8b8ab0ab30, Frame : Return Address
0xffffffa07bf17630 : 0xffffff800f1ee0ad mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffa07bf17680 : 0xffffff800f35b896 mach_kernel : _kdp_i386_trap + 0x116
0xffffffa07bf176c0 : 0xffffff800f34ab20 mach_kernel : _kernel_trap + 0x3e0
0xffffffa07bf17710 : 0xffffff800f188951 mach_kernel : _return_from_trap + 0xc1
0xffffffa07bf17730 : 0xffffff800f1ee38d mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffa07bf17820 : 0xffffff800f1eda39 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffffa07bf17880 : 0xffffff800f9e100b mach_kernel : _panic + 0x84
0xffffffa07bf17970 : 0xffffff800f34aff3 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffa07bf17af0 : 0xffffff800f34ace2 mach_kernel : _kernel_trap + 0x5a2
0xffffffa07bf17b40 : 0xffffff800f188951 mach_kernel : _return_from_trap + 0xc1
0xffffffa07bf17b60 : 0xffffff8014300b53 com.xavier.BigSurface : __ZN16SurfaceHIDDriver13eventReceivedEP13SurfaceHIDNub20SurfaceHIDDeviceTypePht + 0x83
0xffffffa07bf17c80 : 0xffffff80142e374f com.xavier.BigSurface : __ZN13SurfaceHIDNub13eventReceivedEhhhhPht + 0x11f
0xffffffa07bf17cc0 : 0xffffff80142ebb28 com.xavier.BigSurface : __ZN22SurfaceSerialHubDriver14processMessageEv + 0xa88
0xffffffa07bf17e40 : 0xffffff80142eb061 com.xavier.BigSurface : __ZN22SurfaceSerialHubDriver8_processEPht + 0x2b1
0xffffffa07bf17e70 : 0xffffff80142eaf3a com.xavier.BigSurface : __ZN22SurfaceSerialHubDriver8_processEPht + 0x18a
0xffffffa07bf17ea0 : 0xffffff80142ead5f com.xavier.BigSurface : __ZN22SurfaceSerialHubDriver21processReceivedBufferEP22IOInterruptEventSourcei + 0x7f
0xffffffa07bf17ed0 : 0xffffff800f914e44 mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x114
0xffffffa07bf17f20 : 0xffffff800f91373e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x12e
0xffffffa07bf17f60 : 0xffffff800f912d77 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x37
0xffffffa07bf17fa0 : 0xffffff800f18819e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.xavier.BigSurface(1.1)[FDEA9EFD-F5FA-326C-BD8F-FBD3B78DCD4F]@0xffffff80142e2000->0xffffff8014383fff
            dependency: as.vit9696.Lilu(1.6.2)[264B15BE-8923-3A33-A9F5-8F0FFBB80595]@0xffffff801300f000->0xffffff801303afff
            dependency: as.vit9696.VirtualSMC(1.3.0)[40CF5E80-248F-35B7-833D-2DABF724B391]@0xffffff801303e000->0xffffff8013054fff
            dependency: com.alexandred.VoodooI2CServices(1.0)[8AA1B471-9422-37CE-B4E3-24E62A432CA5]@0xffffff801427d000->0xffffff8014287fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[62C52EB0-4D0B-3A00-B4B0-A1841EA15AB4]@0xffffff801178f000->0xffffff8011790fff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[85585968-5B32-38C5-A3C5-1A93D002C67F]@0xffffff80118cb000->0xffffff801192bfff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[8CD5D885-96C0-37C2-9526-BB7B0C243555]@0xffffff8011bff000->0xffffff8011c2efff
            dependency: com.apple.iokit.IOSMBusFamily(1.1)[94016052-A586-3635-8D02-3B13C4E7E569]@0xffffff8011cd3000->0xffffff8011cd3fff
            dependency: com.xavier.VoodooSerial(1.2)[8AD9505D-8E0B-3760-8440-BEA6BBBB68B3]@0xffffff80142b7000->0xffffff80142d8fff
            dependency: org.coolstar.VoodooGPIO(1.2.0)[5E20327A-12E0-3F09-87FB-99E397D4BB31]@0xffffff8014291000->0xffffff80142acfff

Process name corresponding to current thread (0xffffff8b8ab0ab30): kernel_task
Boot args: keepsyms=1

Mac OS version:
22A5373b

```

</details>

Xiashangning, 2022-10-19:
@pigworlds what is your device? SL1 or SL2?
Please try this.
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9803784/BigSurface.kext.zip)

pigworlds, 2022-10-19:
> @pigworlds what is your device? SL1 or SL2? Please try this. [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9803784/BigSurface.kext.zip)

It was on a SL1 device. Your latest kext works great.

Xiashangning, 2022-10-20:
@pigworlds Good, happy to know updates for SL3 doesn't introduce bugs for SL1&2.
@SmartBoy84 since another SL1 user here reported it working without bugs, please recheck your oc config.

closed 2022-10-20 by Xiashangning