Xiashangning/BigSurface #71: Surface Pro Keyboard Question

balopez83, 2022-10-10:
Hi @Xiashangning , This isn't an issue but a question. I am working on a Surface Pro 3 and trying to get the keyboard to work with VoodooI2Chid and am unable to get both trackpad and keyboard working at the same time. 

I noticed that you resolved the issue and you made note of a custom VoodooI2Chid kext ver 2.4.2 in the original VoodooI2C GitHub that did fix this issue. It appears that that version does not work in Big Sur and newer and so need to update it. Since touch support requires the original voodooi2c version to work I need to simply integrate the keyboard fix and wanted to know if you could provide the fix you used to get the keyboard and touchpad working?

Thanks in advance

Xiashangning, 2022-10-10:
Just use BigSurface.kext

closed 2022-10-10 by Xiashangning

balopez83, 2022-10-10:
Hi @Xiashangning I am using a Surface Pro 3 which your BigSurface.kext doesn't support touch screen on. While the keyboard and trackpad work great using your kext I have to use the original voodooi2c to get touch support. 

Would you be able to provide what you changed to get the keyboard and trackpad to work separate of the other changes you made?

Xiashangning, 2022-10-11:
Well. in this case you will need to patch VoodooI2C on your own. Just grab my SurfaceTypeCoverDriver and add it to original VoodooI2C.kext