Xiashangning/BigSurface #73: Touchpad & Touchscreen on Microsoft Surface Laptop 2

lzhoang2601, 2022-10-17:
I used kext and SSDT-Surface with macOS Monterey

Touchscreen not working. Touchpad works but no gestures

Before that I also had problem getting only touchpad or keyboard at the same time, but changed the patch and they work at the same time

Please help. Thanks

Xiashangning, 2022-10-17:
Try this one.
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9799852/BigSurface.kext.zip)

PS: touchscreen needs IPTSDaemon to work with.

lzhoang2601, 2022-10-17:
I tryed but stuck 
[image]
[config.plist.txt](https://github.com/Xiashangning/BigSurface/files/9800367/config.plist.txt)

lzhoang2601, 2022-10-17:
Solved with disabled SSDT-GPI0. But some time keyboard and touchpad not work. Touchpad does work no gesture

lzhoang2601, 2022-10-17:
I had to reboot several times to get into macOS. How to solve?

lzhoang2601, 2022-10-17:
Working perfect with kext in https://github.com/Xiashangning/BigSurface/issues/63#issuecomment-1267831020

Xiashangning, 2022-10-17:
please try the latest build, either in #63 or above

Xiashangning, 2022-10-17:
This one
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9801375/BigSurface.kext.zip)

lzhoang2601, 2022-10-17:
Installed IPTSDaemon but touchscreen not working

Xiashangning, 2022-10-17:
do you follow the installation instructions?
And does this build above work for you

lzhoang2601, 2022-10-17:
> This one [BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9801375/BigSurface.kext.zip)

I can't start macOS with kext

P/s: Sorry, i forget install fmt and inih for touchscreen work

lzhoang2601, 2022-10-17:
> Working perfect with kext in [#63 (comment)](https://github.com/Xiashangning/BigSurface/issues/63#issuecomment-1267831020)

I always have to boot up twice to get the keyboard to work

Xiashangning, 2022-10-17:
> 

Upgrade Lilu & VirtualSMC to the latest version. Try again.

lzhoang2601, 2022-10-17:
I used last commit for all kext

Xiashangning, 2022-10-17:
What about this ?
You won't need BrightnessKeys.kext btw.
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/9803797/BigSurface.kext.zip)

Xiashangning, 2022-10-20:
Another SL1 user in #63 report it working without bugs, since SL1 and SL2 are very similar, and my kext treats SL1 & 2 identically, please recheck your OC config

closed 2022-10-20 by lzhoang2601