Xiashangning/BigSurface #74: Battery status on Microsoft Surface Book

lzhoang2601, 2022-10-20:
I used kext and SSDT-Surface with macOS Monterey. But battery status is not working, what do I need to do to make it work?

Thanks.

Xiashangning, 2022-10-20:
I need logs `sudo dmesg | grep -E "SurfaceS|SurfaceB"`

lzhoang2601, 2022-10-21:
[SSDT.zip](https://github.com/Xiashangning/BigSurface/files/9840588/SSDT.zip)
[config.plist.txt](https://github.com/Xiashangning/BigSurface/files/9840594/config.plist.txt)

I used two patches and kext ACPIBatteryManager. Battery status and sound button worked

lzhoang2601, 2022-10-21:
> I need logs `sudo dmesg | grep -E "SurfaceS|SurfaceB"`

Nothing show in terminal, DebugEnhancer loaded

Xiashangning, 2022-10-21:
I have checked Surface Book's DSDT, it does not use the surface aggregator module so there is no need to use BigSurface.kext at all.

closed 2022-10-21 by Xiashangning