Xiashangning/BigSurface #75: Version 6.2 breaks Trackpad / Battery / Volume Keys

backdash756, 2022-10-26:
When I update to version 6.2 or 6.3, my trackpad is no longer recognized as a Magic trackpad. It only works as a mouse.
Battery indicator and volume key's are no longer recognized.
If I then downgrade version 6.0 again it still doesn't work, I have to restore the whole EFI folder where version 6.0 was in.

Xiashangning, 2022-10-26:
What device

backdash756, 2022-10-26:
Sorry, I forgot: Surface Pro 7

Xiashangning, 2022-10-26:
That's impossible, it's your oc config's problem. I am using a SP7 so SP7 will be guaranteed to work

Xiashangning, 2022-10-26:
attach your config.plist here

closed 2022-10-26 by Xiashangning

backdash756, 2022-10-26:
What can be responsible for that? I really don't change ANYTHING except BigSurface.kext and SSDT-Surface.aml.
Strange, as i said, it remains broken even if i go back to 6.0.
If I restore the EFI folder from the backup, everything works again.
Here is my config.plist
[config.plist.zip](https://github.com/Xiashangning/BigSurface/files/9868886/config.plist.zip)

Xiashangning, 2022-10-26:
Have you upgraded Lilu and VirtualSMC to the latest version?

backdash756, 2022-10-26:
Yes, Lilu 1.6.3 and VirtualSMC 1.3.1

Xiashangning, 2022-10-26:
please read the README, the loading sequence is not right