Xiashangning/BigSurface #76: Support older version of macOS

pigworlds, 2022-10-27:
First big thank you for this great work! Without BigSurface.kext I won't able to use keyboard on SL1.

I tried BigSurface.kext on Mojave, it doesn't seems to work. Further look show that BigSurface.kext only run on macOS 10.15+.

Getting support on older version of OS isn't complicated. You just need to set MACOSX_DEPLOYMENT_TARGET to lower version, and fix another one line code to let the kext compatible with older OS.

I have created a diff https://github.com/Xiashangning/BigSurface/compare/dev...pigworlds:BigSurface:downlevel and verified working well from Sierra and above. Choosing Sierra as the baseline because it is the oldest OS that can run on SL1. In addition to this diff, the similar MACOSX_DEPLOYMENT_TARGET change need apply to all three submodules too.

Would you mind incorporate this to support older version of macOS?

Xiashangning, 2022-10-27:
To be honest, I cannot understand why someone would prefer an older version of OS. Unless you give me a good enough reason, I am not going to add support for 10.14 and older.

RockJesus, 2022-10-28:
> 

我的是sl1用了你最新的版本，大写键可以使用，但是led不会亮，这个功能倒是无所谓，需要的话我可以帮你测试，QQ 2994638

pigworlds, 2022-10-28:
Adding older OS support gives people the choices of the OS that works best. I’m always running the latest OS on the bleeding edge, but not all people would do. Just like you add Catalina support a while ago, I would appreciate lower level of OS to be supported.

pigworlds, 2022-10-28:
> PS: since you have a SL1, may I ask you to test if your caps lock led is turned on correctly?

The Caps lock functionality works fine. But the Caps lock led keeps off.

Xiashangning, 2022-10-28:
Fine, even though it is against my will, I will adjust the OS support down to Serria in next release. :)
For now, you can only try to build it yourself.

closed 2022-10-28 by Xiashangning