Xiashangning/BigSurface #79: [Surface Book 3] Trackpad lagging/skipping

educovas, 2022-11-21:
Surface Book 3 15", tried this amazing kext on my laptop but I'm having this weird skipping on the trackpad, couldn't find any way to fix this. Tested on macOS Monterey 12.6 and MacOS Ventura 13.0. Thank you for your time!

Xiashangning, 2022-11-23:
Make sure using the latest version of BigSurface, enable DebugEnhancer.kext and send me the output of `sudo dmesg | grep -E "SurfaceS|SurfaceH"`

educovas, 2022-11-24:
[log.txt](https://github.com/Xiashangning/BigSurface/files/10084459/log.txt)

educovas, 2022-11-24:
Yes, that's all I have, I did try to reproduce the problem before and after running the command. Also ran the command 3 times with the same result.
I have both Lilu debug and DebugEnhancer enabled.

jlempen, 2022-11-25:
Hi @Xiashangning, hi all,
I can confirm this issue with two different Surface Book 3 devices, a 13.5" i5 and a 15" i7.
The trackpad works great on both devices with the BigSurface 6.3 kext, but the pointer/cursor will skip/hang for about a tenth of a second every few seconds.
I paired a Magic Trackpad 2 with both devices and the pointer/cursor is buttery smooth when using the MT2.
Also, the Intel Power Gadget shows no CPU spike of any kind when the skipping/hanging occurs.
I'll try to send you a debug log when I have one of the devices on hand this week-end.
Thanks again for your amazing work!

yaseenshaik, 2023-01-13:
I can confirm the cursor skip issue exists on the surface laptop 3 as well on Monterey 12.6.1.

jlempen, 2023-01-13:
Thanks @yaseenshaik for your feedback. I was very busy and totally forgot about this issue and set aside the SB3. It's surprising that no one else is reporting this issue if both the SB3 and the SL3 are affected by it. IMHO, this issue makes these systems more or less unusable for everyday work.

While researching on hacking a mid-2019 Razer Blade 15 Advanced, I stumbled upon this interesting thread on the [VoodooI2C repo](https://github.com/VoodooI2C/VoodooI2C/issues/456#issuecomment-859701344). That looks very similar to our skipping issue. It was referenced at the end of [this thread](https://github.com/VoodooI2C/VoodooI2C/issues/319).

As @Xiashangning doesn't have an SB3 or an SL3, we'll have to look into it ourselves if we want to help him fixing it.

Xiashangning, 2023-01-13:
From logs I cannot see any errors or warnings produced, so indeed this is very weird for me. 
BTW, when the lagging issue happens, are u using the keyboard at the same time?

sxcsfan, 2023-01-13:
I also encountered this problem on SL3 with Ventura. @Xiashangning while the symptom occurs I was not using keyboard at the same time.

jlempen, 2023-01-17:
Hi @Xiashangning,
Here's my log from the output of `sudo dmesg | grep -E "SurfaceS|SurfaceH"`. Apart from the two `SurfaceHIDDevice` addresses, it's exactly the same as the log from @educovas. I'm afraid there's not much more useful information here.
[log_jlempen.txt](https://github.com/Xiashangning/BigSurface/files/10438998/log_jlempen.txt)

I put the DEBUG version of `DebugEnhancer.kext` in the kext folder and enabled it with the `-dbgenhdbg` boot argument. Apart from that, all other kexts, as well as OpenCore itself, are the RELEASE versions. In my config.plist, i have `Misc -> Debug -> Target = 3` and `AppleDebug = False`. There's no `-v` flag in my boot arguments neither. Is that correct or am I a total idiot? :-)

The load order of the BigSurface kexts is as specified in your README.md.

Just for the sake of completeness, here's what's in my ACPI, Drivers and Kext folders:

[image]
[image]
[image]

yaseenshaik, 2023-02-09:
Thank you, very much @Xiashangning! We will take whatever we can get!

Xiashangning, 2023-02-10:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/10706086/BigSurface.kext.zip)
Try this one

yaseenshaik, 2023-02-11:
Thanks for your hard work. We appreciate it.

Xiashangning, 2023-02-13:
@yaseenshaik so do you still have the trackpad lagging issue?

yaseenshaik, 2023-02-13:
Hi @Xiashangning. 

Unfortunately, I can still see the lag with the new kext you linked. But IPTSDaemon was causing high CPU loads and had to be unloaded.
[image]

Thanks for your work.

Anyin2020, 2023-03-05:
我的laptop3也出现了该问题，触控板间歇性跳跃和迟滞

yaseenshaik, 2023-03-13:
Hi @Xiashangning , Just giving some feedback on this release. Though the skipping is still present, I've gotten used to it so much that I don't even notice it happening. Without IPTSDaemon, the touch/click works and scrolling doesn't work. But, this isn't a issue as I don't use the touchscreen anyway. Thanks again.

Xiashangning, 2023-03-26:
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/11071424/BigSurface.kext.zip)
Use this build, boot into the system, continuously using your touchpad (without lifting your finger) until the skipping issue happens, DO NOT lift your finger, keep moving it around for a while until it is usable again (back to normal), then go to terminal and please attach the output `sudo dmesg | grep -E "VoodooUART|SurfaceS|SurfaceH|SurfaceB"`

ROCKERLIQUID, 2023-04-24:
Surface Laptop 3, Monterey 12.6.5, OC 0.8.1
[LogTouchpad.txt](https://github.com/Xiashangning/BigSurface/files/11309336/LogTouchpad.txt)

Xiashangning, 2023-05-23:
After some analysis on the output, sadly I have completely no clue why it will lag.
Will look into it when I have more time

MuyuenHoshino, 2023-09-22:
Hi @Xiashangning , I also encountered this problem on SL3 with both Ventura and Monterey. Did you test it under BigSur ,will it become normal if I go back to BigSur ?
Thanks

MuyuenHoshino, 2024-02-11:
> Same here. I am using SL3 with Ventura. Tried both 6.5 and 6.3 release but no luck. Does anybody who resolve the laggy trackpad issue can kindly share their firmware version? I guess this could be the issue.

Hello! Have you solved the problem? Could you please share the solution?

jc-bao, 2024-02-11:
I am afraid not. 😢 
The laggy trackpad really destroy the MacOS experience. 😞

MuyuenHoshino, 2024-02-17:
> I am afraid not. 😢 The laggy trackpad really destroy the MacOS experience. 😞

I am using Sonoma now, the trackpad have less lagging probility than Monterey. And it seems a 120Hz monitor also makes sense. If you need, I can share my efi.

jc-bao, 2024-02-17:
\

> > I am afraid not. 😢 The laggy trackpad really destroy the MacOS experience. 😞
> 
> I am using Sonoma now, the trackpad have less lagging probility than Monterey. And it seems a 120Hz monitor also makes sense. If you need, I can share my efi.

That would be very helpful! I am wondering what do you change or you merely upgrade the system? 
Besides, you mean when you use the 120Hz external display the problem is gone?

MuyuenHoshino, 2024-02-17:
> That would be very helpful! I am wondering what do you change or you merely upgrade the system? Besides, you mean when you use the 120Hz external display the problem is gone?

I don't know the your lagging degree. I have tried bigsur,monterey and sonoma, with almost the same efi (except for wifi and bluetooth)  ,sonoma is the most fluent. At the same time, cursor in bigsur and monterey is very very lagging. And when using 120Hz external display, it seems to skip a shorter distance than 60Hz.Or it could just be an illusion

jc-bao, 2024-02-18:
> > > > I am afraid not. 😢 The laggy trackpad really destroy the MacOS experience. 😞
> > > 
> > > 
> > > I am using Sonoma now, the trackpad have less lagging probility than Monterey. And it seems a 120Hz monitor also makes sense. If you need, I can share my efi.
> > 
> > 
> > That would be very helpful! I am wondering what do you change or you merely upgrade the system? Besides, you mean when you use the 120Hz external display the problem is gone?
> 
> I don't know the your lagging degree. I have tried bigsur,monterey and sonoma, with almost the same efi (except for wifi and bluetooth) ,sonoma is the most fluent. At the same time, cursor in bigsur and monterey is very very lagging. And when using 120Hz external display, it seems to skip a shorter distance than 60Hz, or it could just be an illusion. Please provide me your email so that I can send you my efi

I see. My email is jc-bao@outlook.com. Thank you!

tashda, 2024-03-15:
Just wanted to chime in that I'm experiencing the exact same thing on a SL 3 15" i7.
@MuyuenHoshino If you want to share your EFI with me, my email is kenneth@skabel.se and I'd be happy to look over differences/similarities to exclude what _could_ be the cause of this.

MuyuenHoshino, 2024-03-26:
> Just wanted to chime in that I'm experiencing the exact same thing on a SL 3 15" i7. @MuyuenHoshino If you want to share your EFI with me, my email is [kenneth@skabel.se](mailto:kenneth@skabel.se) and I'd be happy to look over differences/similarities to exclude what _could_ be the cause of this.

Sorry for replying you late, I have already sent efi.

jlempen, 2024-07-04:
Hey all,

**I found the solution for a buttery-smooth trackpad without lagging/skipping** on my Surface Laptop 3 15" i7. This should work for the Surface Book 3 as well.

You have to downgrade your UEFI firmware to the last known working version `13.101.140.0` for both the SL3 and SB3 and downgrade the `BigSurface.kext` to [version 6.2](https://github.com/Xiashangning/BigSurface/releases/download/v6.2/BigSurface.zip).

This UEFI firmware is included in the March 01 2022 release update package for the SL3 (SurfaceLaptop3_Win10_19041_22.011.9779.0.msi) and the May 24 2022 release update package for the SB3 (SurfaceBook3_Win10_19041_22.034.40281.0.msi). As of today, both update packages are not available on Micro$oft's support site anymore, but you'll find them on the [Softpedia website](https://drivers.softpedia.com).

Follow the very clear instructions on the [surface-uefi-firmware Github](https://github.com/linux-surface/surface-uefi-firmware) to downgrade the UEFI firmware. 

In a nutshell:
1. Boot with a Linux Live ISO, preferably a Debian, Arch or Fedora based distribution, as you'll have to install the three packages `msiextract`, `gcab` and `dos2unix` for the downgrade script to work.
2. Download the script from the [surface-uefi-firmware Github](https://github.com/linux-surface/surface-uefi-firmware) and add `OnlyTrusted=false` in `/etc/fwupd/daemon.conf`. 
3. Run the `repack.sh` script, which will unpack the MSI update package, extract all UEFI firmwares from it and generate fwupd metadata for it. Once the script finishes you'll find a list of cab files inside the `fwupdates` folder. 
4. Using `fwupd`, install all the firmware files at once.

This will reboot the laptop and downgrade all UEFI firmwares at once, which takes around 10 minutes. You'll see progress bars with different colours depending on which type of firmware is being flashed.
Once the process is done, your laptop will restart a few times and seem to hang on the Surface logo for 20 or 30 seconds each time, this is normal. Then it will restart for good to your OpenCore picker.
Now restart while holding the F4/Volume Up key to check the firmware version in the UEFI. In the Firmware section, `System UEFI` should now show `13.101.140`.

Reboot into macOS. The keyboard and trackpad should be working, but the trackpad will still lag/skip as before. To fix that, download [BigSurface version 6.2](https://github.com/Xiashangning/BigSurface/releases/download/v6.2/BigSurface.zip) and replace your current `BigSurface.kext`. Do do so, **delete** your current `BigSurface.kext` in the OC kext folder and in your `config.plist` and add v6.2 in the specified order:

[image]

Reboot and you're done. Enjoy a perfect trackpad on your Surface Hackintosh!

A couple of caveats:

- If you are using Windows on the laptop, you'll have to find a way to prevent Windows Update from updating the firmware to the latest version again! I don't know how to do that, but DuckDuckGo is your friend.
- If you are using Xiashangning's [IPTS daemon](https://github.com/Xiashangning/IPTSDaemon) to enable the touchscreen, you'll probably have to downgrade that as well to an [older version](https://github.com/Xiashangning/IPTSDaemon-old). This shouldn't be an issue for Surface Laptop 3 owners (who needs a touchscreen on a non-foldable, non 2-in-1 style laptop), but it could be problematic for Surface Book 3 users.

jlempen, 2024-07-04:
@Xiashangning 
For your information, after I downgraded my Surface Laptop 3's UEFI fimware to the last known good version 13.101.140.0, BigSurface.kext v6.5, v6.4 and v6.3 still exhibit the same lagging/skipping trackpad. Downgrading BigSurface.kext to v6.2 fixes the issue, which leads me to think that commit #538bf71 introduced the skipping/lagging trackpad for SL3 and SB3 users.

jc-bao, 2024-07-04:
@jlempen Thank you so much for the information! I have tried downgrading the firmware but did not change the BigSurface.text version. That seems to be the issue. Again, I really appreciate your detailed instructions! I will try it out today.

ROCKERLIQUID, 2024-07-04:
> buttery-smooth trackpad

@jlempen thanks a lot, works absolutely great on my Surface Laptop 3 i5-1035G7.

jc-bao, 2024-07-04:
@ROCKERLIQUID Which version of MacOS and OpenCore are you using for `6.2`? My original config cannot boot 😢

ROCKERLIQUID, 2024-07-04:
@jc-bao i'm using 0.9.8, Sonoma 14.4.1, make sure you delete BigSurface and OC Snapshot on propertree and then add the 6.2 BigSurface.
not doing this leads to crashing for some reason.

jc-bao, 2024-07-04:
@ROCKERLIQUID Thanks for the instruction! I have figured it out. It is just soooo smooth. Huge thanks to @jlempen!

tashda, 2024-07-21:
Big thanks from here as well, @jlempen  !
I've had to dual boot Linux on my SL3 when I needed to do something as this was bugging me out and I haven't had the time to get it working, but after applying the old UEFI firmware and downgrading to 6.2 it is buttery smooth!!

Finally!

jlempen, 2024-07-31:
@jc-bao @ROCKERLIQUID @tashda 
Please have a look at [this thread](https://github.com/Xiashangning/BigSurface/issues/127#issuecomment-2261648714). As you are `Surface Laptop 3` or `Surface Book 3` users, I could use your help for testing `hibernatemode 25` and my forked `BigSurface` which should fix the trackpad lagging/skipping with `BigSurface v6.5`.
Thanks!

bitoy81, 2024-08-28:
guys still have the older version on IPTS daemon that will work for big surface kext  6.3. accidentally deleted my old copy. new version doesn't work for my SL3.

jlempen, 2024-12-19:
**This issue is now easily fixed by following my simplified instructions on how to downgrade the UEFI firmware using a Linux live USB stick:**

[Surface Laptop 3](https://github.com/jlempen/Surface-Laptop-3-OpenCore#downgrading-the-uefi-firmware)

[Surface Book 3](https://github.com/jlempen/Surface-Book-3-OpenCore#downgrading-the-uefi-firmware)

Then downgrade the [BigSurface.kext to version 6.2](https://github.com/Xiashangning/BigSurface/releases/download/v6.2/BigSurface.zip) by removing the current `BigSurface.kext` and its dependencies and adding the `BigSurface.kext` v6.2 and its dependencies to your `EFI -> OC -> Kexts` folder and `config.plist` file in the following order:

[image]

Or even easier, use my comprehensive repos for the [Surface Laptop 3](https://github.com/jlempen/Surface-Laptop-3-OpenCore) and [Surface Book 3](https://github.com/jlempen/Surface-Book-3-OpenCore) to install and setup macOS on your laptop.

jlempen, 2024-12-19:
I'll reopen this so it gets more visibility for Surface Laptop 3 and Surface Book 3 users.