Xiashangning/BigSurface #80: 关于Surface Pro 6触屏的问题

123MEMZ, 2022-11-23:
两位作者大大，我想来咨询关于SP6触屏的问题。我已下载并安装上您那个"BigSurface"，可是我遇到了一个很奇怪的问题，我无法使用触摸屏，但是电源显示以及按钮都是正常的，所以我就想来咨询一下，希望能解决这个问题。

我已经尝试把"BigSurface"的版本从6.3降级到6.2，但还是毫无效果qwq，我将把我的设置截图放上来，如果还有什么我需要提供的截图，请尽管说，我将会一一放上来！

辛苦了！～

_（哦对，还有个问题，我怎么才能启用环境光？是只能自己编写驱动了吗？）_

[image]
[image]
[image]

Xiashangning, 2022-11-23:
环境光：是的，只能你自己写

你需要发给我终端`sudo dmesg | grep -E "SurfaceM|IntelP"` 的输出，同时记得把DebugEnhancer.kext启用
然后你双击IPTSDaemon那个可执行文件，把他的输出截图给我

Xiashangning, 2022-11-23:
还有，请永远使用最新版的驱动

123MEMZ, 2022-11-24:
这是终端截图

[image]

我很抱歉没有找到DebugEnhancer.kext和IPTSDaemon这个执行文件，还请大佬能够指点一下我～

Xiashangning, 2022-11-25:
额，那是因为还要下载IPTSDaemon才能用触摸屏。是我的另一个项目，那里有开启触摸屏的详细教程

closed 2022-11-25 by Xiashangning

123MEMZ, 2022-11-26:
这是IPTSDaemon的输出截图，请查收！
[image]

123MEMZ, 2022-11-26:
我看您这个项目的介绍说是只有macOS 12及以上的版本才需要安装DebugEnhancer.kext吧，我是macOS 11

Xiashangning, 2022-11-26:
请仔细看IPTSDaemon的安装说明，按要求来

123MEMZ, 2022-11-26:
我按照您的项目一步步走下去之后会提示这个

[image]

123MEMZ, 2022-11-26:
我不知是正常的还是出了点问题

123MEMZ, 2022-11-26:
然后运行install_daemon.sh后终端无任何输出，但重启依旧没有触控

123MEMZ, 2022-11-26:
[image]