Xiashangning/BigSurface #90: Surface Pro 5 (2017) frozen after boot

MauriceChocoSwiss, 2023-02-15:
Hi, I've a trouble with my SP5. After the install, when it start, it froze on language select page, sometime juste after.

I've seen on openCore page that i need to use CPUTsc but I don't know if i need to use it or no with BigSurface.

I've removed all SMC plugins too.

What did you need to help with this ?

SP5, I7-4660U, 46Gb, 1Tb.

Xiashangning, 2023-02-17:
I can't. BigSurface is a driver to improve the using experience of Surface devices on macOS. If you cannot even boot into the system, I am afraid nothing I can do. You should first try to boot into macOS without any problem (disable BigSurface, enable VoodooI2C, VirtualSMC etc.). Then, you can enable BigSurface for full support of the battery etc.

Xiashangning, 2023-02-17:
BTW BigSurface has no conflict with CPUTsc

closed 2023-02-17 by Xiashangning

MauriceChocoSwiss, 2023-02-17:
I've finished the installation, I've used antoher EFI i found, with your kext inside and i don't know really why, it work this time.

Thank's for the infomartions ;)