Xiashangning/BigSurface #93: Battery not recognised in System Preference in Ventura

Xiashangning, 2023-02-26:
In Ventura, without any changes made to the battery code, the battery failed to be recognised in System Preference.

However, macOS does seem to know that there is a battery present as it can show it correctly in the control center or menu bar.

Xiashangning, 2023-02-26:
Don't know if normal laptop has this issue in Ventura or not. The battery code is almost exactly the same from SMCBatteryManager.

he1833, 2023-03-11:
Ordinary notebooks also have this problem, although I don't have ordinary books, I saw the same case of oil and our problem in the Vision Forum, and the general solution seems to be compiled with the latest ECEnabler.kext to resolve

Xiashangning, 2023-03-25:
Thanks for your advice but ECEnabler won't help

he1833, 2023-03-26:
I succeeded and found a workaround by modifying the error identification in the FACP table in the form of an APCI patch so that he correctly identified the Surface as a laptop. Without patching, it is identified as the type of tablet, and it will not correctly match the battery (for laptops) or energy saving (for desktops) in the settings.

he1833, 2023-03-26:
[image]

Xiashangning, 2023-03-26:
Yes, this patch works for me!

Thanks a lot, I will add it into the readme

closed 2023-03-26 by Xiashangning

olekor, 2023-05-08:
Could you give me advice about the installation IPTSDaemon (Monterey 12.4, surface book 2 13" i5 7300u) When running install_daemon.sh  I have this picture
[image]

as I suppose to install a daemon.sh need to use Homebrew, but at the present time, I don't have a USB WIFI dongle to connect internet, as for Marvell AVASTAR wifi card doesn't have a kext.
I tried to install the daemon manually but also had no success. I would be very grateful for your advice.

LvxlongSir, 2024-05-05:
try MacBook Air. sb shared sp7 with this model and it's okay with battary on ventura.