Xiashangning/BigSurface #94: Best Compatible Surface

rkinlen, 2023-03-06:
I switch from a desktop Hackintosh to a laptop Hackintosh. Does anyone know the best surface model to be used with this?

plsh2me, 2023-03-06:
For me Microsoft Surface Laptop 3 15inch is the best one - https://www.ebay.co.uk/itm/394482658879?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=_vqbygs5sey&sssrc=2349624&ssuid=_vqbygs5sey&var=&widget_ver=artemis&media=WHATS_APP

rkinlen, 2023-03-07:
would this work? https://www.microsoft.com/en-gb/store/configure/Surface-Laptop-Go-2/8pglpv76mjhn?icid=PDP_LaptopGo2_CTA1

rkinlen, 2023-03-12:
> For me Microsoft Surface Laptop 3 15inch is the best one - [...]

would this work? https://www.microsoft.com/en-gb/store/configure/Surface-Laptop-Go-2/8pglpv76mjhn?icid=PDP_LaptopGo2_CTA1

closed 2023-03-20 by Xiashangning