Xiashangning/BigSurface #97: Surface Book 3 - Ventura - BigSurface not working.

MetalChris87, 2023-03-24:
Hi,

I have been using BigSurface on Monterey on Surface Book 3 with no issues, however I have recently tried to install Ventura and can't get it to work. To be honest it could be my EFI folder. I have tried a few changes but without success. Basically the mac boots but sometimes the mouse and keyboard work and sometimes they don't. Sometimes wifi works and sometimes it doesn't. And most times the system crashes and shuts off . 

If you or anyone has a working Ventura EFI for me to try? At least one I can tweak for the Surface Book 3.

Thanks

Xiashangning, 2023-03-25:
> Sometimes wifi works and sometimes it doesn't

This has nothing to do with BigSurface.

> sometimes the mouse and keyboard work and sometimes they don't

When they don't work, do you still have your battery? If no, then this means BigSurface fails to be loaded and thus log need to be checked. send me `sudo dmesg | grep -E  "VoodooUART|SurfaceS|SurfaceH|SurfaceB"`

If upgrading to Ventura makes your install unusable, I would say that this may not be an issue with BigSurface. But any way, please add `keepsyms=1` to your boot args and if your system crashed, reboot it and send me your kp detail, I can take a loot at that.

valasco, 2023-04-20:
蒙特利尔下，0.7.8鼠标键盘工作正常，任务栏有电池图标，充电有相应闪电，不充电显示外接充电器状态。但是在偏好设置里面无电池图标反而是💡的节能图标！

MetalChris87, 2023-05-06:
Hi. You can delete this issue. Everything is working perfectly. 
Thanks

closed 2023-05-06 by Xiashangning