Xiashangning/BigSurface #101: Catalina or Surface Laptop 1

MeisterLone, 2023-04-04:
Everything works, but only some of the time. 

30-50% of the time, either keyboard or mouse will not work. Sometimes both dont work. I reboot a couple of times and suddenly both work. Any ideas how I can debug this issue? Do you need some kind of log?

Xiashangning, 2023-04-04:
Load DebugEnhancer.kext, boot into macOS and attach the output of `sudo dmesg | grep -E "VoodooUART|SurfaceS|SurfaceB|SurfaceH"`

MeisterLone, 2023-04-04:
@Xiashangning I 
[Archive.zip](https://github.com/Xiashangning/BigSurface/files/11151566/Archive.zip)
retrieved logs as requested for all 3 scenrios. I only had to restart about 5 times to get these so its happening pretty frequent.

Scenarios:
1. Trackpad works but no keyboard
2. Keyboard works but no trackpad
3. Keyboard and track both work flawlessly

See attached archive, thanks. Let me know if there are additional debugging steps I can do.

MeisterLone, 2023-04-05:
I installed Venture to see if it would be any better. Similar issue. No the battery charge icon takes about 10 seconds before it shows in the top right corner. The board starts working only AFTER the battery icon appears in top right corner. (some times) What could explain this behaviour? Maybe load order of something is bad??

Xiashangning, 2023-04-05:
Please try this build
[BigSurface.kext.zip](https://github.com/Xiashangning/BigSurface/files/11162586/BigSurface.kext.zip)

MeisterLone, 2023-04-05:
Thanks for the explanation. So it seems like the initial communication is what it causes the problem? Can we somehow implement a delay or delay before devices go live? 

Your posted kext does not seem to make any difference. Although I have noticed since upgrading to ventura, trackpad now always works. I have not booted once with trackpad not working.

Now it is just the keyboard and battery indicator that work intermittently.

Something that seems to work is if I wait about 10 seconds or so after welcome screen show until the battery icon renders in the corner. (This takes about 10 seconds). When the battery icon loads, I can use the keyboard. If i try to use the keyboard before the icon shows, it is doomed and wont load in. 

Is it normal for battery icon to take 10 seconds + to show?

Xiashangning, 2023-04-07:
Can you attach the output after waiting battery icon to show up?
Copy the output directly from the terminal
`sudo dmesg | grep -E "SurfaceS|SurfaceH|SurfaceB"`

Xiashangning, 2023-04-12:
> Can we somehow implement a delay or delay before devices go live?

Yes, it is already implemented and the delay is 20 seconds.

> Is it normal for battery icon to take 10 seconds + to show?

Not normal. For example, SP6 which has the same hardware as SL1, with 20s delay, does not produce too many errors. And the battery can show up before the welcome screen shows. Maybe the initialization config for the hardware is different? If you would like to solve this with me, please install Ubuntu and this custom kernel, boot into this kernel, and get me the output `sudo dmesg | grep "debug"`

MeisterLone, 2023-04-12:
Hey @Xiashangning sorry for not responding quicker. I have given the laptop to my wife and she is actually using it now daily, so I have not had some time to spend with it. 

I will try today to assist in this. Do I need to install ubuntu internally or can it be booted from USB?

Xiashangning, 2023-04-15:
You can install it on the USB drive. But you do need to install it. After that you can proceed with the instructions in the link

closed 2023-05-12 by Xiashangning