Xiashangning/BigSurface #105: USB C charging stopped working?

ciuhc, 2023-05-23:
It used to work but now seems like there's issues with handshake. Anyone else experience this?

Xiashangning, 2023-05-23:
For me no, and this is also has nothing to do with BigSurface.kext.
I think the only possible problem is USB mapping issues. But anyway I will leave it here for a couple of days to see if others have the same problem

jlempen, 2023-05-25:
Well, on my i5 SP7 with BigSurface v6.5, if I boot with the USB-C power adapter plugged in, the battery icon shows the little lightning bolt as if it was charging, but the battery is actually discharging. If I unplug and replug the USB-C power adapter, it starts charging again.
BTW, I use CoconutBattery to monitor my battery.

Rainringgoaway, 2023-06-05:
Might be surface UEFIframework update.

ciuhc, 2023-06-14:
I was playing around a bit and concluded there is some handshake issue when charging starts - OSX keeps showing the icon change between charging and not charging every 5 seconds...

Rainringgoaway, 2023-09-13:
最近的surface驱动更新修复了c口充电问题，可以去win系统里更新下驱动再尝试充电。

jlempen, 2024-08-19:
> 最近的surface驱动更新修复了c口充电问题，可以去win系统里更新下驱动再尝试充电。

```
Translation: The latest Surface driver update has fixed the C-port charging problem. You can update the driver in the Windows system and try charging again.
```
I think we can close this now.

closed 2024-08-19 by jlempen