Xiashangning/BigSurface #109: Bluetooth Not turning on

JadenMishael, 2023-07-11:
On my Hackintosh Surafec pro 7 my bluetooth doesn't turn on, when i turn it on in settings the switch just turns off (macOS Ventura) btw already tryed fixing with yt videos and more still not working FIX PLEASE

Xiashangning, 2023-07-14:
I'm sorry but BigSurface contains no code about bluetooth & wifi. They are managed by `IntelBluetoothFirmware` (https://github.com/OpenIntelWireless/IntelBluetoothFirmware) and `IntelBTPatcher` (https://github.com/acidanthera/BrcmPatchRAM)
You should consult to their installation instructions for trouble shooting. (ref https://openintelwireless.github.io/IntelBluetoothFirmware/FAQ.html#what-additional-steps-should-i-do-to-make-bluetooth-work-on-macos-monterey-and-newer)
Also, something might be helpful: make sure your bluetooth device is recognized in macOS. (In hackintool's USB section, the bluetooth is shown as a USB device and marked as internal)

closed 2023-07-14 by Xiashangning

balopez83, 2024-02-06:
Hi @JadenMishael I have an up to date EFI  for the SP7 that has Bluetooth working out of the box. See below:
https://github.com/balopez83/Surface-Pro-7-Hackintosh
Xiashangning is right, Bluetooth is not related to this kext and is instead a combination of other kexts and settings.