Xiashangning/BigSurface #111: Ventura触控飘移/幽灵手势

SoSlowA, 2023-08-26:
(IDK how to express this in english, so plz read chinese)
在Ventura下安装的IPTSDaemon触控出现问题，具体体现如标题：在单指触控/使用Surface Pen时，可能出现光标不跟手指走的情况，并且在使用单指/双指时，可能出现多指手势，即存在幽灵判定
另：电脑触摸屏正常（Windows下）

PGBSean, 2023-09-22:
Translation:"There is a problem with the touch control of IPTSDaemon installed under Ventura, as shown in the title: When touching with one finger/using Surface Pen, the cursor may not follow the finger, and when using one finger/two fingers, multiple problems may occur. Pointing gesture means there is a ghost judgment
Another: The computer touch screen is normal (under Windows)"

Translated by Google Translate

PGBSean, 2023-09-22:
I can confirm this, but as macOS does not have multi-touch support, macOS can get insane while handling "2 mouses" (as supposed to 2 touches)

jlempen, 2024-08-20:
> I can confirm this, but as macOS does not have multi-touch support, macOS can get insane while handling "2 mouses" (as supposed to 2 touches)

Well, I strongly disagree! macOS has **_fantastic_** multitouch support on the trackpad to begin with. And once [BigSurface v6.5](https://github.com/Xiashangning/BigSurface/releases/download/v6.5/BigSurface.zip) and the [IPTSDaemon v1.4](https://github.com/Xiashangning/IPTSDaemon) are properly installed, you get the same fantastic multitouch support on your touchscreen. 

Tested on macOS Ventura and Sonoma on SB3 and SP7.

Please make sure [BigSurface v6.5](https://github.com/Xiashangning/BigSurface/releases/download/v6.5/BigSurface.zip) is installed and working correctly, then download and install the [IPTSDaemon v1.4](https://github.com/Xiashangning/IPTSDaemon) by following my [HOWTO](https://github.com/Xiashangning/IPTSDaemon/issues/20).