Xiashangning/BigSurface #112: surface laptop 3 触摸板断触

kevin-Curry, 2023-09-28:
使用最新6.5bigsurface和1.4 IPTSDaemon
surface laptop 3 Ventura Monterey断触

szsong, 2023-10-03:
Same here. 遇到一样的情况。
Downgraded to release version 6.3 and the issue resolved.
降级到 release 6.3 以后工作正常。

kevin-Curry, 2023-10-03:
> Same here. 遇到一样的情况。 Downgraded to release version 6.3 and the issue resolved. 降级到 release 6.3 以后工作正常。

如何使用命令安装fmt9.0？

szsong, 2023-10-20:
> > Same here. 遇到一样的情况。 Downgraded to release version 6.3 and the issue resolved. 降级到 release 6.3 以后工作正常。
> 
> 如何使用命令安装fmt9.0？

??? why this question?

jc-bao, 2024-02-09:
Downgrade to 6.3 doesn't resolve the issue for me 😢 I am using SL3 15inch. Does anybody have other solutions? Could this related to the firmware version?

jlempen, 2024-08-19:
Fixed by https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390