Xiashangning/BigSurface #114: Surface Laptop 3 Camera works after USB mapping with USBToolbox

alex32012, 2023-11-09:
Was trying to fix my USB port randomly stop working by mapping USB ports checked in Hackintool and Surface Front Camera showed up in USB ports. So opened camera and it worked. I'm not sure if this is known or not.

gonav8, 2023-12-28:
Hi,
Can you share screenshot on your Hackintool USB port setup?
I cannot see USB Camera works

jlempen, 2024-08-19:
The cameras on the `Surface Laptop 3` and `Surface Laptop Go` are connected to the USB bus and work out of the box once you have mapped your USB ports properly, for instance with the help of [USBToolBox](https://github.com/USBToolBox/tool).

The cameras on the `Surface Book 3`, `Surface Pro 7`, `Surface Laptop 1` and `Surface Laptop 2` are IPU4 and will probably never work on Linux and macOS.

On the `Surface Book 1`, `Surface Book 2`, `Surface Pro 4`, `Surface Pro 5` and `Surface Pro 6` they are IPU3. There's been some progress to make the cameras work on Linux, see https://github.com/linux-surface/linux-surface/wiki/Camera-Support. Some users have posted some instructions to try to make them work on macOS: https://github.com/Xiashangning/BigSurface/issues/65.

closed 2024-08-19 by jlempen