Xiashangning/BigSurface #117: loptop3 15寸 的设备ID：0x34ED 开机无法正常启动

xixi160323, 2024-01-13:
[image]
希望老师有时间能够更新对此设备的支持

xixi160323, 2024-01-13:
[image]

xixi160323, 2024-01-13:
[image]

jlempen, 2024-08-19:
@xixi160323 

The `Surface Laptop 3` works with `BigSurface` v6.3 and newer, even if the trackpad is a bit choppy. But there's a fix for this issue: https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390 

Regarding your boot issues, this has nothing to do with `BigSurface`. Please read the [Dortania OpenCore guide](https://dortania.github.io/OpenCore-Install-Guide/) to make your own EFI for your SL3.

closed 2024-08-19 by jlempen