Xiashangning/BigSurface #118: Erratic Ghost Touch on SP4 with SP5 Screen Replacement

balopez83, 2024-01-24:
@Xiashangning Thank you again for all the work you have done with this kext. 

I noticed an odd issue with Touch and Stylus on the Surface Pro 4 that has had the Surface Pro 5 screen replacement (due to the de-lamination issue impacting the Surface Pro 4). The cursor will move erratically all over the screen after the IPTSdaemon is enabled (touch doesn't work even with the latest kext without this on any Surface Pro 4). This condition will continue until I am able to disable the daemon. 

This condition does not impact a regular Surface Pro 4 with a Surface Pro 4 screen. I have both and validated this condition side by side. Also the Surface Pro 4 that had the screen replaced with the Surface Pro 5 screen used to work until the screen was replaced. 

It's important to note that all other OS's work fine with touch and stylus and do not experience this ghosting issue (e.g. chromeOS, Windows, Linux)

I was attempting to isolate what the issue is messing around with the configurations and even attempting to remove the stylus and/or touch drivers and building to see if one or the other was having the issue but have been so far unsuccessful. 

Any thoughts you have on this would be very helpful. Thank you in advance.

rongchuan17, 2025-01-22:
Hey, I want to ask if you solved this problem, I had a similar problem to you.

balopez83, 2025-01-22:
@rongchuan17 Hi, no unfortunately I never resolved this. I suspect that @Xiashangning will need to update the kext for any changes caused by using the SP4 screen controller, adapter cable, and the SP5 screen.